Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #7: *.ctarget Project Umbrella Proposal

ReinhardKeil, 2021-07-05:
This is a proposal for the umbrealla project that describes complex applications.  It integrates the concepts of CMSIS-Zone, but merges the __*.azone__ file into the __*.ctarget__ file. 

File Extension | Description
:---------------|:---------------
*.rzone           |  Describes the target processor and contains the system memory (and optionally peripheral) resources.
*.ctarget         | Describes the application and the assignments of of resources to sub-projects. (It could include the *.azone information to avoid additional files.)
*.fzone           | Data Model for template engine.  Used in combination with a (toolchain specific) linker script template.
*.cprj              | One or multiple project files.  Each file generates contains all information to generate an independent binary. 

[image]

# How it may work 
When a application is started a *.ctarget file is generated. To define the target four different options are possible.

  - When target board is selected, a *.rzone file is used as it describes all system resources of the board including memory that is off-chip from a processor.

  - When a device is selected, a *.rzone file is generated; this file contains only on-chip memory provided by the device.  Optionally (for v8M with TrustZone recommended) a
    *.rzone file could be part of the Device Family Pack (DFP) that contains the complete system topology including peripherals, alignments, interrupts, etc.
	
  - When a *.cprj is selected, the device or board selection is taken from the *.cprj file.  If no *.rzone file is provided in the DFP or Board Pack,     it is generated is described in the previous step.

  - When a *.clayer is selected that defines a target, a *.cprj is generated. If no *.rzone file is provided in the Device Family Pack or Board Pack,  it is generated is described in the previous step.
	
## Modify a *.ctarget file
	
An application described in a *.ctarget file can be modified as follows:
  - A different *.rzone file can be associated, which re-targets all *.cprj files to a new device.
  
  - An existing *.cprj can be added to a *.ctarget. If the device mismatches it can be overwritten (after user confirmation).
  
  - A *.cprj can be removed from a *.ctarget file.
  
  - The *.ctarget file should define a build order (in the first iteration without file dependencies).
  
  - Resource assignments can be modified using a CMSIS-Zone development flow (a command line variant may be defined).

## Build application  
A application build process generates:
  - The *.fzone file is generated that represents a data set for the script template engine.
     -- This *.fzone is used as input for the linker script template to generate a linker script that relate to each *.cprj project file.
     -- This *.fzone file is also the input for other script templates exists, for example to generate a SAU and MPC setup.

  - Then each *.cprj file is used to generate binaries. The build order is defined in the *.ctarget file.

## Validation for Consistency  
When a *.ctarget file is loaded, the tool should perform the following consistency checks:
  - Is the related *.rzone file there?  If not, issue user notification with the option to recreate the *.rzone file from the DFP.
  - Is the device in each *.cprj consistent with the *.rzone file.  If not, issue user notification with the option to update the device in the *.cprj file.

__NOTE:__ This enables to retarget an application by replacing the *.rzone file.  Since the *.rzone file describes the resource of a system, it is therefore easy to change the target hardware, for example from evaluation board to production hardware.

LMESTM, 2021-07-06:
A first question to start with: why choosing *.ctarget as a name for an umbrella project file for complex applications.

ReinhardKeil, 2021-07-06:
I'm open to any (meaningful) file extension.  I just have started with something and 'c' stands for CMSIS and 'target' for the fact that it describes the complete target application.  Any other suggestions are welcome.

LMESTM, 2021-07-06:
Maybe I should have started by saying that yes it makes sense to have an umbrella project.

On the name, as this is an umbrella _project_ , by similarity to the *.cprj file extension *.uprj or *.ucprj may be used for instance ... but I haven't checked if this extension has already been used.

ReinhardKeil, 2021-07-06:
Here is how the *.ctarget extension could relate to an IDE.

[image]

LMESTM, 2021-07-06:
ok so that can be ok with "Target Hardware" menu replaced by "target application", or "application", and user could enter free text that may or may not include the target name : "aws_project_stm32l5" or "myApp" ...

ReinhardKeil, 2021-07-07:
Personally, I suggest to replace "Target Hardware" with just "Target".  The name displayed here should be the *.ctarget file name without extensions (perhaps with some build options).  The name is complete free-form, but a wizard in the IDE might suggest names. Examples could be:
  - "STM32L562"  if this relates just to a device.
  - "STM32L5-Discovery" if this relates to a board.
  - "GoPro Model1"  to indicate the target system (the default build) - this name is choose by the user.
  - "GoPro Model1:Debug" the debug build of the *.ctarget file
  - "GoPro ComTest" a test platform for communication test (reuses *.cprj or *.clayer components from "GoPro Model1").

I hope that this makes the overall concept a bit clearer. Please let me know if there are further questions.

I know that I have indirectly introduced the motion of build variants, but I just wanted to exemplify how this could be shown in an IDE.

cartu38, 2021-07-07:
Means no way way here to get an umbrella project targeting multiple hardware ?
"NOTE: This enables to retarget an application by replacing the *.rzone file. Since the *.rzone file describes the resource of a system, it is therefore easy to change the target hardware, for example from evaluation board to production hardware." is equal to me to single target per umbrella project.

As part of .ctarget we may have a mix of hardware agnostic .cprj (midllewares, ...)  & hardware related one(s). Coupling .ctarget to a specific hardware is possibly cutting some flexibility. What's your view here ?
I'm referring to https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack/issues/10 discussion stating: "Gap: Consider a workflow where the user starts from application and middleware and in the end can choose from HW that supports this."

I'm thinking too to application "variants" support as part of an umbrella project.

ReinhardKeil, 2021-07-12:
_Means no way way here to get an umbrella project targeting multiple hardware ?_
A: you are correct, this limits flexibility somehow. But we have to start somewhere and should see if we can improve this aspect.  Supporting two hardware targets that are similar should be something that we should consider with build variants, as indicated here https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack/issues/7#issuecomment-875325631. 

_Gap: Consider a workflow where the user starts from application and middleware and in the end can choose from HW that supports this."_
A: This is definitely something that we should re-visit.  I will note that.

ReinhardKeil, 2023-01-09:
This is outdated

closed 2023-01-09 by ReinhardKeil