Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #10: CMSIS-Pack Gap Analysis

jkrech, 2021-07-06:
The following items are collected from the discussions of the Shadowfax Componentization WG:

### Component Discovery
- search for components within all available packs 
  - today's tools search within installed packs
- filter components by Vendor, Device, Board, Middleware, API, Cclass 
  - today's tools unconditionally filter by device, processor, tool-chain, enabled processor feature
- separate component identification from classification
  - Cclass/Cbundle/Cgroup/...
- extend component classification using predefined feature enumeration for semantic sorting/filtering

### Component Configuration
Currently limited to single dimensional "Cvariant" selection
- add component configuration parameters that impact:
  - file and include path selection
  - the condition (dependency)
- add component characteristics (e.g. Heap / Stack / Clock requirements)

### Component Licenses
- one or more licenses for one or multiple components 
- click through optional
- spdx license reference
- display title

### Change Log
- associated with one or multiple components outside of releases section

### Description Structure
- specify component condition within the scope of a component
- separate filter condition from dependencies (undecided)
- separate <api> from <component> ID (undecided)
- taxonomy and grouping levels need more flexibility 

### Description Syntax
- condition syntax is hard to read and write (use boolean expressions)
- add description of peripherals to a device to be used by conditions

### Description Scope
- add the ability to describe quality metrics, test results and coverage
- add the ability to describe tests for automated testing
- add the ability to describe and access "data files" required by "other tools" / e.g. generators/configuration tools
- add a description for specifying a list of pack versions forming a consistent set of component = Software Development Kit (SDK)

### Pack description development and validation
PackChk does not:
- validate that a component with resolved dependencies builds without missing modules and include paths
- have an option to check that each file of the zip archive is referenced at least once in the description
- provide a cookbook for developing a component and package description
- generate component dependencies from source code analysis
Pack generation does not add "digital signature" / Pack installation does not validate "digital signature"
We are lacking tools for easily adding components to pack descriptions

### End User Workflows
Today's implementations work under the assumption that at each time the user has selected a device.
Investigation focuses on migrating a project across different boards/devices easily.
Gap: Consider a workflow where the user starts from application and middleware and in the end can choose from HW that supports this.

iomint, 2021-07-12:
Joachim, thanks for this good recap of the Shadowfax componentization WG discussion. 
I come back to one point of the ST feedback shared during last week meeting: the spec should allow vendors to add their differentiators to a common corpus of  spec adding hooks possibly from every nodes and accepted as such I.e extension by a pack-checker tool. 
I think this point is underlying a few points you have captured but it may be interesting to add it explicitly possibly in the "pack description development and validation.

jkrech, 2021-07-13:
@iomint I have added the "extension" topic explicitly above.

iomint, 2021-07-20:
Reinhard, 
The <environment> extension point is definitively something interesting and also noticed by maxime in the discussion #5 , one question is whether it is applicable to other elements than <device>
I do agree with your comment on <taxonomy> and I would like to copy here a short slides presented in June by ST in the shadowdax context showing  some needs around the components taxonomy depth.
[ShadowFax ST inputs Components depth.pdf](https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack/files/6848853/ShadowFax.ST.inputs.Components.depth.pdf)

DavidJurajdaNXP, 2021-07-27:
I am missing differentiation between various include path types in pdsc. I am recommending following types: 

- asm
- c
- cpp

jkrech, 2021-08-02:
> I am missing differentiation between various include path types in pdsc. I am recommending following types:
>     * asm
>     * c
>     * cpp

Added section **Component Description** above.

ReinhardKeil, 2021-08-02:
> I am missing differentiation between various include path types in pdsc. I am recommending following types:
> 
> * asm
> * c
> * cpp

C/CPP header files are somewhat inter-changeable (extern "C") and to my knowledge, assembler source files are pre-processed and translated with the C/C++ compiler.  Is it really a requirement to specify different include path types given the fact that include files can be generic?  Why is this a requirement?

DavidJurajdaNXP, 2021-08-02:
Hi @ReinhardKeil,

Yes, this is requirement from NXP side.

Reasons are:
1. IDEs are typically offering separated forms for C/C++ and asm include paths (MDK included). We are creating data for data driven tools, if tools offering differentiation of include paths, meta data should also offer same capability.
2. Include files can be generic, but once there is dedicated form in GUI for particular set of include paths (C/C++, asm), it should contain only necessary, related paths and not others.
3. We are differentiating such types internally in MCUXpresso SDK.

LMESTM, 2021-09-06:
About :
> - specify further "extension points" to the description format for "vendor's customization" (further discussion needed)
>   - currently: device subsection <environment name="MyVendor"> ... </environment>

The question was asked where exactly we think it could help to add extension points (goal is to avoid to many add-ons in the corresponding schema ) 

We think that below are placements that would / could be useful:
-	at the root (same level as <keywords> or <requirements> so that environment specific information can be added if needed)
-	under component and components, in case of specific components where special "work-around" might be needed
-	under condition and conditions (depending on the outcome of discussions around <produces> <consumes> or "features" discussions)
-	under generators and generator as this may be very toolset specific and extra specific info might be useful
-	under files and file, in case specific file format of information, or even environment specific files that wouldn't be meaningful for other environments

Note: the use of environment extension points should be as limited as possible and any GAP that cannot be implemented with the standard should be discussed in open-cmsis-pack forum to find a standard way to address it. Nevertheless adding this possibility may provide vendors and pack providers enough freedom to temporarily implement work-arounds and achieve project deadlines while the standard evolutions are discussed.

jkrech, 2021-09-07:
Please review a draft version of the PACK.xsd attached.
[PACK.xsd.txt](https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack/files/7121027/PACK.xsd.txt)
I did not add extension to `<file>` as this should be handled on the files level if this is environment specific files/filetypes.

LMESTM, 2021-09-17:
Hi. Here is another topic I'd like to propose for discussion as a GAP.

In ST deliveries we have template files that need to be adapted by users for their project needs, like stm32u5xx_hal_conf_template.h in 
https://github.com/STMicroelectronics/STM32CubeU5/tree/main/Drivers/STM32U5xx_HAL_Driver/Inc
This file is named as stm32u5xx_hal_conf_template.h and it is documented that user should renamed it stm32u5xx_hal_conf.h in his project.

With the template attribute of CMSIS Pack:
https://www.keil.com/pack/doc/CMSIS/Pack/html/pdsc_components_pg.html#FileAttributeEnum
In our example, the stm32u5xx_hal_conf_template.h template file can be listed but it would not be renamed as desired.
That would be useful to be able to optionally specify the target name different from the original name. 
Also that would be appreciated that the user file can be moved to another target folder path than the default RTE folder, like the project root for instance, or relative to the project root.

jkrech, 2021-09-21:
@LMESTM I would like to better understand why you consider the stm32u5xx_hal_conf_template.h file a template rather than a configuration file. 
see: https://arm-software.github.io/CMSIS_5/latest/Pack/html/pdsc_components_pg.html#FileAttributeEnum

As far as I understand this file is mandatory to be present in a project rather than optional. As such it is a configuration file to me rather than a template. If you specify it as such it will be copied into the project for you and the version being tracked. I don't see a reason for needing different names in the pack than in the project you just ensure that it is not located in a directory that is used as include path.

LMESTM, 2021-09-21:
@jkrech - Actually you're probably right, it can be considered a configuration file. The same requirement actually applies for a configuration file and a template:
- we'd like the file to be renamed to stm32u5xx_hal_conf.h during the process of copy
- we'd like that the file can be copied somewhere else than RTE folder

jkrech, 2021-09-28:
@LMESTM, I hoped we would discuss further during yesterday's project meeting but since you were not able to attend I suggest we continue here. 
@DavidJurajdaNXP  has agreed to describe NXPs view on template/config files

The only reason I can see for needing to rename a 'config' is backward compatibility with today's repository directory structure, but I would rather think twice before using this as an argument for extending the pack specification.
I also believe that it is not a good idea to place a config file into a general include folder (Drivers/STM32U5xx_HAL_Driver/Inc) as it would be picked up if the include folder would be set for the compiler command line.
I am not sure I understand why the location of a configuration file matters to

LMESTM, 2021-09-28:
> @LMESTM, I hoped we would discuss further during yesterday's project meeting but since you were not able to attend I suggest we continue here. @DavidJurajdaNXP has agreed to describe NXPs view on template/config files

I guess you mean last week ? yes sorry that I could not make it.

> 
> The only reason I can see for needing to rename a 'config' is backward compatibility with today's repository directory structure, but I would rather think twice before using this as an argument for extending the pack specification. I also believe that it is not a good idea to place a config file into a general include folder (Drivers/STM32U5xx_HAL_Driver/Inc) as it would be picked up if the include folder would be set for the compiler command line. I am not sure I understand why the location of a configuration file matters or makes a difference.

First I think CMSIS PACK should allow to distribute existing software, and that it should be flexible enough for that. If distribution through CMSIS packs requires rework of the software structure, then it will lower adoption level.  

In the discussion, I am considering both cases of files of attribute config and templates. In our solution, we're providing the config file as a template: cubeMX will generate it by default, if user does not use cubeMX, he will have to create the configuration file based on the template that is provided with _template suffix.  This file is named with _template suffix to make it clear this is a user file . Because of suffix (or thanks to it) , it is not an issue that the file is placed inside the include folder because other files will not include it ...  but it requires renaming. 

We are thinking about other use cases as well. 
- for instance, our cubeMX generator can "generate" files from templates, and we would like to be able to offer configurable names and locations for different SW layers to avoid having all the generated files in a single hard coded place. Typically it allows to generate i2c.c file in a $Project/User/Core folder and we would like to be able to cope with this kind of case.
- One other possible use case is multiple instances of a single template (.c or .h files) - different target folders and/or different target names would open for this possibility.  In the case above, some users may prefer to generate i2c1.c for instance 1 and i2c3.c for instance 3 of I2C. 
-

closed 2025-02-11 by jkrech