Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #8: Review CMSIS-Zone

ReinhardKeil, 2021-07-06:
This issue is indented to discuss pros/cons of CMSIS-Zone.

The overall concept of CMSIS-Zone is explained here https://arm-software.github.io/CMSIS_5/Zone/html/index.html.  The overview page contains a video that explains the operation of CMSIS-Zone.

__Strenghts of CMSIS-Zone__
  - Awareness of hardware limitations (alignment, number of regions, etc).
  - Interactive mode for resource allocation with instant user feedback.
  - [Flexible template engine via FreeMarker](https://arm-software.github.io/CMSIS_5/Zone/html/GenDataModel.html) allows to generate source files and tool control files.
   
__Current Implementation Issues__
 - Eclipse based: CMSIS-Zone is currently an Eclipse plug-in - obviously not the technology we favor going forward.  
 - [Command line mode](https://arm-software.github.io/CMSIS_5/Zone/html/zTCLI.html) that is basic and does not allow to manage resources.  Perhaps the next-generation implementation should be split into a command-line and UX part.
 - Not widely adopted.

__Potential Problem with JAVA__
[FreeMarker](https://freemarker.apache.org/) is a open-source project managed by the Apache Software Foundation. The current implementation is JAVA based which adds complexity for desktop installations. Also Oracle has the trend to change license conditions - any legal or technical concerns here?

cartu38, 2021-07-07:
About Java / Freemarker I fully share concern. Why not adopting GPDSC https://www.keil.com/pack/doc/CMSIS/Pack/html/pdsc_generators_pg.html same way of working ?

If Eclipse based I may imagine we are discussing here about UI contribution to help .azone creation. According to me such is IDE responsibility to promote its own. But a CLI is required to manage then CMSISzone data -> project files.

ReinhardKeil, 2021-07-13:
I take your point to review GDPSC as an option, but the issue here is that we would like standardized ways to generate tool-specific files.  The GDPSC approach does not have a template language behind.

Ultimately we should support flows like described here https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack/issues/12.
FreeMarker is a JAVA implementation, but today the dependency to Oracle is gone, as there are open source alternatives for the run-time system.  Given the flexiblity, I think FreeMarker templates are an option that we should consider.

jkrech, 2025-02-11:
Currently out of scope.

closed 2025-02-11 by jkrech