Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #11: CMSIS-Build Gap Analysis

jkrech, 2021-07-12:
The following items are collected from the discussion of the Shadowfax Componentization WG exploration team investigating requirements for a project format and a build management CLI tool.

Discussion was based on CMSIS-Build 0.10.0 https://github.com/ARM-software/CMSIS_5/releases/download/5.7.0/cbuild_install.0.10.0.sh
note: a number of limiting items raised in the discussion have been fixed in version 0.10.2:

### Gaps:
- pre- and post-build steps must be specified as part of the build project (item 2)
  - cross platform support
  - specify runtime environment
- define generator operation as part of the build
- define multiple build configurations variants within single project (e.g. Debug vs. Release)
  - need to carefully decide what the supported variations shall be 
  - do not allow different board, device, module? 
  - only allow difference in component variants?
- define an additional file that orchestrates multiple dependent projects ("Umbrella") (item 12)
  -  multi-core, secure / non-secure, boot-loader + application, etc.
  - build order, cross-project references (item 5)
- define and maintain a dictionary of tool-chain flags that allows to specify command line options for different toolchains in a common way (e.g. optimization levels, debug vs release flags) (item 10)
- extend the files section to allow specifying a physical directory rather than individual files to be included in a build (item 7) (Note: this feature was not agreed, as some people found it "convenient" and others found it "dangerous" and "fuzzy" (e.g. for reproducible builds) - convenience could be achieve through tools)

### Limitations
- support nesting of logic file groups in files section (item 3) (fixed)
- support file specification in files section without group (item 4) (fixed)
- extend target section with toolchain independent specification of include path and define (item 9) (fixed)
- multiple toolchain support requires toolchain specific component configuration files to be named uniquely as they otherwise overwrite each other (design requirement ?)

### Conclusions from the discussion:
- the cprj file must allow to support multiple compiler tool-chains configuration options concurrently (item 8)
- the cprj file must not store IDE/UI settings. These settings shall be stored in separate files (item 1)
- in order to create self contained copies of projects, all project file references are relative to the project file (item 6)

tarek-bochkati, 2021-07-13:
thanks @jkrech for summing up CBuild/CPRJ related notes.

I'm just missing the flags override attribute like the 'add' and 'delete' attributes, as far as I remember we were partially OK on this point.

is it possible to add this to the GAPs list ?

jkrech, 2021-07-14:
@tarek-bochkati good catch. I did miss it above. I have added it to the list of gaps, however I also added the conclusion from the discussion, that in the rare case of an override, this should be achieved by specifying the "remove" with the complete flags and specifying the complete set of new flags using the "add attribute.

LMESTM, 2021-07-15:
Another item that would be useful, probably in cprj, is how to handle the source code of project components. 
As of now, AFAIK, in Keil components source code is kept in CMSIS Pack root. But we think that users may prefer to have a copy of the selected source at specific location in their project tree. So cprj should offer an option to specify for each component to either use the cmsis pack root reference, or to make a copy (and where to copy).

cartu38, 2021-07-15:
Any initiative to specify static libraries (.a) dependencies order (maybe reading order is way to do today?) ? Kind of corner case but sounds to me required thinking reproducible builds.

cartu38, 2021-07-16:
'/cprj/files/.../file' 'name' attribute : do you consider support of some variable describing path to get resource ? Today relative (or absolute path if supported ?) are not helping project structure revisit (resource(s) repository(ies) renaming, move, ...).

jkrech, 2021-07-19:
> '/cprj/files/.../file' 'name' attribute : do you consider support of some variable describing path to get resource ? Today relative (or absolute path if supported ?) are not helping project structure revisit (resource(s) repository(ies) renaming, move, ...).

Today files are specified relative to cprj file only, since absolute paths do not allow to relocate the project (projects shall be copied out of packs using a recursive copy of a base folder). Renaming and moving of files is probably a feature that a tool shall provide rather than burdening the project format with?

jkrech, 2021-07-19:
> Another item that would be useful, probably in cprj, is how to handle the source code of project components.
> As of now, AFAIK, in Keil components source code is kept in CMSIS Pack root. But we think [...]

Added to the description above.

jkrech, 2021-07-19:
> 
> 
> Any initiative to specify static libraries (.a) dependencies order (maybe reading order is way to do today?) ? Kind of corner case but sounds to me required thinking reproducible builds.

Added to the description above.

slhultgren, 2021-10-06:
Another topic is "lockfile" concept.

Since the .cprj file allows working with version ranges, and since the condition resolving of components are non-obvious to users it would be very helpful to record the exact used packs and components in a ".cprj.lock" file.

This will allow for a similar workflow as with "package.json/yarn.lock" from NodeJS ecosystem, where the ".cprj" describes what CAN be used, and the ".cprj.lock" describes what IS used.

This way, there is no fear when moving a project from one computer to another computer with different CMSIS_PACK_ROOT contents, since the .cprj.lock file will guarantee that we still build with the same contents.
The users will then commit the .cprj.lock file, and if there was no changes in the .cprj.lock file they can feel confident that they are still using the same sources in the build.

brondani, 2021-10-08:
> Another topic is "lockfile" concept.

It seems the "lockfile" would have precisely the content of the file generated with the [cbuildgen](https://arm-software.github.io/CMSIS_5/Build/html/cbuildgen.html) `--update=<CprjFile>` option:
```
Generate <CprjFile> with fixed versions for reproducing the current build.
```
@slhultgren can you please confirm it?

brondani, 2021-10-11:
> An issue with this approach to locking (--update) is that it cannot support both reproducible builds in a team and still keeping the project definition fairly loose.

@slhultgren I believe you misunderstood the `--update=<CprjFile>` usage. In your workflow when building a project, to create/update a `lock.cprj` without touching the `loose.cprj` you could simply call:
```
cbuildgen cmake loose.cprj --update=lock.cprj
```

LMESTM, 2021-10-12:
> > Fetch all dependencies that are in the lockfile and store it in the project folder.
> 
> @0xc0170 Isn't it just a matter of setting the CMSIS_PACK_ROOT to a project subfolder?

I don't know if this this @0xc0170 idea, but my understanding would rather be to get all the dependencies (most probably only the applicable subset of CMSIS_PACK_ROOT content for a given project) into the project folder / file system tree so that user can zip it. That process may require to configure the project to point to those local copies thanks to a project-level local_repository.pidx. We think this is something users will want to do.

jkrech, 2025-02-11:
CMSIS Solution specification and tools are covered in devtools and cmsis-toolbox repository now. Feature requests are tracked there.

closed 2025-02-11 by jkrech