Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #20: Review CMSIS-Pack generator concept

jkrech, 2021-08-03:
The currently specified scope of a generator associated with a component is defined here:
[https://arm-software.github.io/CMSIS_5/latest/Pack/html/pdsc_generators_pg.html](https://arm-software.github.io/CMSIS_5/latest/Pack/html/pdsc_generators_pg.html)

Initial gap analysis:
- assumption: generators are explicitly triggered to allow explicit user input (add option for non-interactive mode).
- the generator does not have access to project information (e.g. currently selected components or resources).
- the generator does not have access to project information of "dependent projects".
- a generator can only be associated with a component but not with a board or device.
- there is no way to express dependencies between generators - only component dependencies are updated and re-evaluated.

ReinhardKeil, 2021-11-12:
I assume this is for generation of configuration information.  I have a hard time to read the diagrams as they are animated.  Is there a static view to this diagrams.  

Should we consider it during the "Project Manager" work that we started?

ReinhardKeil, 2022-03-01:
# Proposal for Revised Generator Support

## Workflow assumptions

The composition of a solution of a solution should have the following steps:

- Create `cproject.yml` files and the `csolution.yml` container that refers the projects.
- Select `device:` or `board:` (optionally by using `target-types:`)
- Add `components:` or `layers:` to the `cproject.yml` file
- For components that have configuration, run the generator in configuration mode
   - change pinout, clock, resources, etc.
   - reflect configuration in *.gpdsc file (and related settings files)

## Steps for component selection and configuration

The following explains the generator workflow of CSolution / CBuild for configuration of components:

1. User selects components in `*.cproject.yml` under `components:`
    - When these components require generation, user is notified to run a generator.
    - "CSolution Run GenID” is invoked for a list of components.
    - CSolution generates `*.cgenerator.yml` file that provides project context and a list of user-selected components.
  
2. Running the Generator (for Component Configuration, i.e. pin selection)
     - Generator reads `*.cgenerator.yml` file
     - User performs the configuration is done.
         - Interactive mode (where a settings file is generated)
         - Remote mode (where a settings file is an input) **IS THIS REALLY REQUIRED**
     - Generator creates a `*.gpdsc` file that informs the CSolution tool about
         - (a) the fact that a component is configured and has generated code, 
         - (b) additional components that are the result of some user configuration.

   Discussions: 
     - is a component list or a dependency list
     - Generator might be VS Code plugin or web based

3. User creates CBuild output with CSolution Convert command
      - Both `*.cproject.yml` and `*.gpdsc` are read by Csolution and create the complete list of selected components.
      - If `*.gpdsc` does not contain component information about a component that has `genId` and selected in `*.cproject.yml` the generator configuration is incomplete. This can happen when a component is added at a later step.
      - Likewise the Generator can detect with the `<component User="1">` attribute that a component is longer required. In this case the user is notified to run a generator.

## Enhance Usability

Add Run Generator buttons to Cclass descriptions.

[image]

## Workflow

1. For `*.cproject.yml` files that contain selected `<components>` with a `generator` or `genid` attribute the `csolution` manager checks if a file with the name `./<project>/RTE+<target>/genid.cgen.yml` exists.
    - When this file is missing, it is required to use the command `csolution run genid` to start the generator.
    - When this file exists, the `csolution` manager checks if the list of components with `genid` has changed. If this is the case it is required to use the command `csolution run genid` to reconfigure generated components.

2. The command `csolution run genid` creates the file `./<project>/RTE+<target>/genid.cgen.yml` and starts the generator.  The generator creates a *.GDPSC file along with other source files that are required [as specified](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_generators_pg.html#element_generators).

**Example Content of \*.cgen.yml (in this case STM32CubeMX.cgen.yml)**

The `*.cgen.yml` file is passed to generator as argument.

```yml
cgenerator:
  device: STM32F407IGHx
  board: NucleoF407
  solution: C:/tmp/MySolution/MySolution.csolution.yml 
  project: C:/tmp//MySolution/Blinky/Blinky.cproject.yml 
  context: .Debug+Nucelo    # build-type and target-type of the current generated context
  destination: C:/tmp/MySolution/Blinky/RTE+Nucelo/

  packs:                     # packs that are used for the project
    - pack: Keil::STM32F4xx_DFP@2.16.1
      path: C:/CMSIS-PACKS/Keil/STM32F4xx_DFP/2.16.1/
[... 139 lines cut ...]
  </conditions>
  <components>
    <bundle Cbundle="STM32CubeMX" Cclass="Device" Cversion="1.4.0">
      <component generator="STM32CubeMX" Cvendor="Keil" Cgroup="STM32Cube Framework" Csub="STM32CubeMX" condition="STCubeMX">
        <description>Configuration via STM32CubeMX</description>
        <RTE_Components_h>
          #define RTE_DEVICE_FRAMEWORK_CUBE_MX
        </RTE_Components_h>
        <files>
          <file category="header" name="MX_Device.h"/>
          <file category="header" name="STCubeGenerated/Inc/stm32g4xx_hal_conf.h"/>
          <file category="source" name="STCubeGenerated/Src/stm32g4xx_hal_msp.c"/>
        </files>
      </component>
      <component  Cgroup="Startup" User="1">  <!-- user selected component in *.cproject.yml -->
        <description>System Startup for STMicroelectronics</description>
        <files>
          <file category="source" name="STCubeGenerated/MDK-ARM/startup_stm32g474xx.s" />
          <file category="source" name="STCubeGenerated/Src/system_stm32g4xx.c" />
        </files>
      </component>
    </bundle>
  </components>
</package>
```

jkrech, 2022-03-08:
We need to agree on using either *.cgenerator.yml or *.cgen.yml above.
In #78 there is three updates that need to be reviewed and reflected above:
a) Revised Additional FileTypes: 
- genParams
- genSettings
- genInput
b) Context specific Generator file format YML vs. JSON (configurable): *.cgenerator.yml 
c) *.cgenerator.yml to provide a list of all files for each selected component including path information

ReinhardKeil, 2022-03-08:
I believe `genSettings` are not part of the pack itself as this are the settings that are used to configure the parameters.  These settings are project specific (and should be therefore supplied during creation of the project) or this is incorrect.

`genInput` would replace genHeader and genSource as from the perspective of the generator it is not important to know which type of file is created.  Instead, the file extension identifies the filetype (header, source, scatter file, ect.).  Is this correct.

ReinhardKeil, 2022-03-08:
@fred-r when I understand it correct, the data input format to the handlebars engine JSON. This would be true for:
  - cgenerator - the file that defines the input to the generator
  - fzone data file - the file that contains the resource information of the system

Is it correct that both of these files should have JSON format?

ReinhardKeil, 2022-03-08:
This is an useful example in this context

CMSIS-Zone example for STM32L5 reworked with Handlebars:
https://github.com/fred-r/CMSIS-Zone/tree/main/Examples/STM32L5

fred-r, 2022-03-08:
> @fred-r when I understand it correct, the data input format to the handlebars engine JSON. This would be true for:
>
> * cgenerator - the file that defines the input to the generator
> * fzone data file [...]

Yes, in fact the rational behind this proposal is that if a file is not intended to be used by a human-being then JSON is more direct for the tools (serialization) and also the JSO support is built-in in some languages (JS, pyhton).

fred-r, 2022-03-08:
> I believe `genSettings` are not part of the pack itself as this are the settings that are used to configure the parameters. These settings are project specific (and should be therefore supplied [...]

Yes,

genSettings is more for the project (and the layer).
genInput is probbaly not a great name but after discussing with my colleagues we concluded that what is important is to identify the inputs for the codegen, not the exact file type (naming rules or extensions can be used for that). This gives the flexibility to support different technologies or to add additional subdivisions without reworking the spec each time.

ReinhardKeil, 2022-03-08:
Then I would propose the following new file categories for `components`
  - genParms template parameters
  - genInput source templates for the generator and other related input files for the generator

This reduces the number of extensions to what we see is needed today.

ReinhardKeil, 2022-03-08:
I suggest to complete this discussion under #104

fred-r, 2022-08-26:
Dear all,

on STM side, we are currently progressing on our flow and usage of the generators.
From a user flow perspective, we are aligned with the proposal.

We confirm that we envision projects with:
- several components invoking the same generator
- 1 component invoking 1 generator

So:

__Components description__
1. The spec must allow us to associate the very same generator to several components.
2. The spec must allow us to provide a global unique identifier per generator: we do not need this to be a direct way to invoke an executable. It can be a unique identifier and then it is up to the project manager (csolution tool) to invoke it properly.

Then:

__Generator Invocation__
The project manager (csolution) must be able to invoke each __unique__ generator: no need to call N times the same generator if several components using it are in the project

__Generator Definition__
Now, regarding the way to define this generator: we can keep the existing arguments for backward compatibility reasons.
But, we also need one more argument to provide the generator input file (cgen.yml/json).

Our current thinking is to introduce a $G argument:
```
    <generator id="myUniqueId">
      <description>An example of Open CMSIS Generator</description>
      <exe>
        <command host="win">generator.bat</command>
        <!-- $S = Device Family Pack base folder -->
        <argument>$S</argument>
        <!-- D = Device (Dname/Dvariant as configured by environment) -->
        <argument>$D</argument>
        <!-- Project path and project name (as configured by environment) -->
        <argument>$P</argument>
        <argument>#P</argument>
        <!-- #name of the gpdsc template contained in this PACK and aligned with GPDSC name-->
        <argument>my_template.gpdsc</argument>
        <!-- #path of the generator input file -->
        <argument>$G</argument>
      </exe>
      <!-- workingDir></workingDir -->
      <!-- path is specified either absolute or relative to PDSC or GPDSC file. If not specified it is the project directory configured by the environment -->
      <gpdsc name="$P\my_gpdsc.gpdsc" />
    </generator>
```

For the format of this input file, we are in favor of "unifying it" with the lock file proposal from @slhultgren.
This file must describe the project "at file level" so that there is no need to go back to the pdsc to understand it.

__Executive Summary__
We need:
1. a spec update to allow multiple components to reference the same generator (more a clarification than a real update)
2. a spec update to introduce the unique generator id concept
3. a spec update to define the generator input file format : hopefully, this can be the same as the lock file format
4. a spec update to introduce the new $G argument in the \<generator\> element
5. a spec update to have new files categories
6. a csolution tool update to invoke only each __unique__ generator and not all generators
7. a csolution tool update to generate the generator input file and pass it with $G

Point to be clarified:
The concept of user selected component passed to the generator is unclear to us.
I guess that if the project composition evolves, then the project manager (csolution) should reinvoke the unique generators "blindly". Then, up to each generator to determine if something needs to be done or not depending on what it does ?
The complexity is more when a file used by a generator gets updated.
If this file is not related to project composition, then the project manager (csolution) has no way to detect this change and determine that a generator must be run again.
Probably here we should not try to deal with it at project manager level.
The "entity" updating the file should be able to invoke the concerned generator, ideally via csolution so that csolution can decide if it runs only the requested generator or all of them.
This might get us to the point identified by Joachim : expressing dependencies between generators.
I am not sure we should shoot for this complexity from day one.
The dependencies could be kept at components level, then I am not sure we should try to have the project manager dealing with all dependencies coming from generators.
Open thoughts : if really we want to go this way, then probably a generator should also define his dependencies (file level) in its definition so that the project manager (csolution) can evaluate it. But I am afraid we will enter a too high complexity level here.
Probably the project manager (csolution) should care only for components.

jkrech, 2022-08-30:
Continued here #104

closed 2022-08-30 by jkrech