Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #21: Support of Component Features in the *.PDSC format

ReinhardKeil, 2021-08-31:
Background: Generic software components may have optional features that can depend on configuration settings or availability of hardware features. As software may have fixed requirements on such optional features, it is proposed to extend the CMSIS-Pack specification as described below.

NOTE: This is an early proposal that would require further review. In the moment it is for discussion and not a commitment to implement this type of functionality.

Extend the element [component](https://arm-software.github.io/CMSIS_5/Pack/html/pdsc_components_pg.html#element_component) and [Condition](https://arm-software.github.io/CMSIS_5/Pack/html/pdsc_conditions_pg.html#element_condition) with the attribut Cfeature="value". "value" allows to specify a bit-mask as hex value or a character sequence.

The usage could be:
```
<component Cclass="Device" Cgroup="SDK Driver" Csub="clock" Cvariant="Std" Cfeature="abc"> ....
<component Cclass="Device" Cgroup="SDK Driver" Csub="clock" Cvariant="Min" Cfeature="ac"> ....
<component Cclass="Device" Cgroup="SDK Driver" Csub="dma" Cfeature="0x2F"> ....
```
This condition would match only the first component with Cfeature="abc"
```
<condition id="x1"> <require Cclass="Device" Cgroup="SDK Driver" Csub="clock" Cfeature="ab" />
```

This condition would match the Csub="dma" component as the mask is fulfilled:
```
<condition id="x1"> <require Cclass="Device" Cgroup="SDK Driver" Csub="dma" Cfeature="0x21" />
```

This condition would not match the Csub="dma" component as the mask is not fulfilled:
```
<condition id="x1"> <require Cclass="Device" Cgroup="SDK Driver" Csub="dma" Cfeature="0x11" />
```

To describe the available features, the [taxonomy](https://arm-software.github.io/CMSIS_5/Pack/html/element_taxonomy.html) element could be extended as shown below (this needs feasibility check as description has currently no child elements). The addition of 
the attribute Dvendor and Ddevice would allow to make the taxonomy specific to a device vendor or a device.
```
<taxonomy Dvendor="STMicroelectronics" Ddevice="STM32U*">
  <description Cclass="Device" Cgroup="SDK Driver" Csub="clock" Cvariant="Std" name="Standard Clock Driver">
    <Cfeature value="a"    name="ASCII mode"         doc="http://www.st.com/cmsis/Cfeature/SDK/clock/a.html"/>
    <Cfeature value="b"    name="Binary mode"        doc="http://www.st.com/cmsis/Cfeature/SDK/clock/b.html"/>
    <Cfeature value="0xF"  name="Clock Source Input" doc="http://www.st.com/cmsis/Cfeature/SDK/clock/inputmode.html"/>
    <Cfeature value="0x30" name="Clock PLL mode"     doc="http://www.st.com/cmsis/Cfeature/SDK/clock/dllmode.html"/>
  </description>
</taxonomy>
```
**Open questions:**
  - Does this match the requirement?
  - Is this sufficient to extend the Cvariant capabilities (perhaps we could make features configurable)?  Can we therefore drop the extension of Cvariant?
  - Can we merge this somehow with the Layer concept of provide and consumed <interfaces>?
  - Can we simplify it further?
  - Is there any conflict for extending the CMSIS standard in this way.
  - Do older implementations of the RTE management would ignore the attribute Cfeature so that it is upward compatible?

mdortel-stm, 2021-09-01:
Hi @ReinhardKeil ,
Thank you for describing this proposal.
Personally, I do not find in your background explanation the mindset behind the discussions we started some months ago.

The features mentioned in discussion #5 were already discussed many months ago to solve what ST saw as an issue, what is the fact that in the current way of component description identification and classification are mixed in the same attributes. The main example is the Cclass: it is part of the unique component ID but also allows to filter component on their "focus". Beyond this, the fact that bundles is defined as the variant of class highlights the fact that the Cclass values granularity brings limitation. When you define "Wireless", you may mean "NFC" or "WiFi" for example. So, if you need to bring in one project two bundles, on for NFC and the other WiFi, it won't be possible as they are mutually exclusive, due to the class variant.

After many exchanges, an idea came: associating a list of other attributes, what we call here "features", to a component, with the following formalism:

`<component ...>
  <features>
     <feature key="x" value="..."/>
     <feature key="y" value="..."/>
  </features>
</component>`

This brings a kind of flexibility:
- You can have some feature keys (and even values) normalized by the definition provided by Open-CMSIS. You can also allow vendor specific keys.
- This mechanism allows to set several values for the same key: depending on the software component size / granularity, this can have an importance. The larger the component is, the more likely it is to bring several values for the same key.

After taking some time to look at you proposal, I understand you built it from the specific issue we describes in discussion #5, focusing, if I correctly understood, on the relationship between a software component and a hardware feature. But I think that the mechanism above allows to cover many more situations, for example dependency of software features.

I could imagine that this answer is not able to clearly sum up what I had in mind when writing the discussion, so I can understand it would be worth a live discussion.

Regards,

Maxime

jkrech, 2021-09-14:
@mdortel-stm could you draft a scheme based on ST's requirements to play it through?
e.g.
key="OCP-Grouping"
name="value" type="OCP-GroupingType"

```
  <xs:simpleType name="OCP-GroupingType">
    <xs:restriction base="xs:token">
      <xs:enumeration value="RTOS" />
      <xs:enumeration value="Wireless" />
      <xs:enumeration value="USB" />
      <xs:enumeration value="FileSystem" />
      <xs:enumeration value="Sensors" />
      ...
    </xs:restriction>
  </xs:simpleType>

```
As well as drafting what you would expect to see in a "component view/list"

ReinhardKeil, 2023-01-09:
@mdortel-stm is this still relevant or are there alternative ways to solve it?

mdortel-stm, 2023-01-10:
@ReinhardKeil the situation has not changed on my side but the topic has not been rediscussed for a long time.

ReinhardKeil, 2023-10-30:
@mdortel-stm does "Parts" now solve it?  If so, could you please close this issue.

mdortel-stm, 2023-10-30:
This has been addressed more through #23. I guess we can close this one.

closed 2024-01-25 by ReinhardKeil