Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #22: Taxonomic rank shift

DavidJurajdaNXP, 2021-09-08:
API definition in ARM.CMSIS pack:
`{'Cclass': 'CMSIS', 'Cgroup': 'RTOS', 'Capiversion': '1.0.0', 'exclusive': '1'}`
Bundle definition in ARM.CMSIS-FreeRTOS pack:
`{'Cbundle': 'FreeRTOS', 'Cclass': 'RTOS', 'Cversion': '10.2.0'}`

RTOS category appears in two different taxonomic ranks (Cclass vs Cgroup). This discrepancy is confusing.
Solution might be:
- introduction of different classification systems for API and SW components (APIClass/SWComponentClass)
- introduction of new taxonomic rank ('framework': 'CMSIS')

Alternative solution could be separation of component identification from classification. This modification has strong impact on backward compatibility, but introduce higher flexibility for vendors. Anyone could introduce custom classification system for SW components according to individual needs. Classification become just additional element of SW component. Single SW component could be part of multiple classification systems.

jkrech, 2021-09-09:
I would just like to add to the above picture that there is also components implementing the specified API e.g.
```{'Cclass': 'CMSIS', 'Cgroup': 'RTOS', 'Csub': 'FreeRTOS', 'Capiversion': '1.0.0'}```
-> this component has a dependency on the FreeRTOS components under RTOS Class.

So a balanced view would be to have both RTOS with a "standard api" vs. a "proprietary api" in the Cclass
```
RTOS:
    CMSIS (common api):
                          FreeRTOS
                          Keil RTX
    FreeRTOS (proprietary api):
                          FreeRTOS
```

DavidJurajdaNXP, 2021-09-14:
I would like to show analogy with https://en.wikipedia.org/wiki/Taxonomic_rank
There are several levels of standard hierarchy ranks, but not all of them are mandatory and only last two are used for identification.

Image bellow is pseudo classification hierarchy based on fixed ranks.
[image]

Alternative attempt:
[image]

Key property is fact that certain unit/taxon may appear only on specific rank.

It would be useful to consider variable depth of ranks in some way. Dot convention in name level might be solution (freertos.allocator.heap).

ReinhardKeil, 2021-09-14:
Maybe we should work a bit more on Component Features (https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack/issues/21) as I think this adds another dimension to the component selection.

As for the taxonomy: we can of course discuss re-grouping, but what about a 'find' feature in the tools itself, and perhaps adding a "keyword" list to each of the components.  The 'find' feature could use Cvendor, Cclass, Cgroup, Csub, Cvariant, Cfeatuers, description, and keyword list.

ReinhardKeil, 2023-01-09:
@DavidJurajdaNXP is this something that you would like to drive forward?

closed 2024-04-24 by ReinhardKeil