Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #23: Component Discovery Gaps

jkrech, 2021-09-14:
Split out from: https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack/issues/10
links to: https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack/issues/21

Requires a tool that can:
- search for components within all available packs
  - today's tools search within installed packs
- filter components by Vendor, Device, Board, Middleware, API, Cclass
  - today's tools 'unconditionally' filter by device, processor, tool-chain, enabled processor features

Requires data model extension:
Extend component identification with independent classification(s)
      Identification: Cclass/Cbundle/Cgroup/...
      Classification: 
      key = classification scheme
      value = predefined / standardized / agreed classification id

associating a list of other attributes, what we call here "features", to a component, with the following formalism:
<component ...> <features> <feature key="x" value="..."/> <feature key="y" value="..."/> </features> </component>

This brings a kind of flexibility:
You can have some feature keys (and even values) normalized by the definition provided by Open-CMSIS. You can also allow vendor specific keys.
This mechanism allows to set several values for the same key: depending on the software component size / granularity, this can have an importance. The larger the component is, the more likely it is to bring several values for the same key.

jkrech, 2021-09-14:
@mdortel-stm could you draft a scheme based on ST's requirements to play it through?

mdortel-stm, 2021-09-14:
I just understand that I maybe missed the point and chose the wrong naming :) Sorry
I'll try to provide the right information in my next try.

mdortel-stm, 2021-09-14:
First proposal:

key="Layer"
name="value" type="LayerClassificationType"
```
  <xs:simpleType name="LayerClassificationType">
    <xs:restriction base="xs:token">
      <xs:enumeration value="RTOS" />
      <xs:enumeration value="Middleware" />
      <xs:enumeration value="Application" />
      <xs:enumeration value="Driver" />
      <xs:enumeration value="BSP" />
      ...
    </xs:restriction>
  </xs:simpleType>
```

mdortel-stm, 2022-03-08:
Hi all,

Such a long time I did not talk about this topic. So I come back, with the hope getting closer to a conclusion.

1. The need to associate the component a list of key/value pair to achieve its classification seems to be clear and understood.
2. The wording is somehow anecdotic. Features / Keywords (too close to pack key word: not fond of it) / Attributes (?): I don't really know how to conclude.
3. The support of list allows to provide several values for the same key: it is an improvement compared to the current definition
4. Nevertheless we need to build bounds and common ground for some information. For that purpose, I suggest to consider a node to list "schema" files and not define it in the schema of the PDSC.
  The pack checking should be able to verify as many things as possible using that input, because some of the definitions will be provided by OpenCMSIS, some by a vendor.
   I was thinking about something like:

```
<keywordSpecs>
  <spec url="OpenCMSIS.1.0.0.spec"/>
  <spec url="STM.1.2.1.spec"/>
</keywordSpecs>
```

6. In each specification file will be found a "schema" for some keys:

```
<specification version="1.0.0">​
  <keyword key="layer" maxOccurrence="1" type="string">​
    <value>driver</value>​
    <value>middleware</value>​
    <value>rtos</value>​
    <value>application</value>​
    ...​
  </keyword>​
</specification>
```
Which is the equivalent of my previous message content. I just imagine that XSD cannot support that kind of specification. (I might be wrong?)

And a customer specific spec could be:

```
<specification version="1.2.1">​
  <keyword key="st-driver" type="string">​
    <value>HAL</value>​
    <value>LL</value>​
  </keyword>​
</specification>
```

Whatever the format (XSD or something else), the schema should allow to declare the keys, their value type and, if applicable:
- The allowed values.
- The minimum occurrence count: by default 0, else key becomes mandatory.
- The maximum occurrence count: by default not defined; could allow the "singleton" check if set to 1, for example.

mdortel-stm, 2022-03-11:
I remarked this piece of in another issue (10):

```
<component Cclass="Board Support" Cgroup="SDK Project Template" Csub="project_template" Cvariant="MK64F12" Cversion="1.0.0" type="config" condition="device.MK64F12_AND___component.serial_manager_AND_utility.debug_console__OR_utility.debug_console_lite__AND_component.uart_adapter_AND_device.MK64F12_startup_AND_driver.adc16_AND_driver.clock_AND_driver.common_AND_driver.dspi_AND_driver.gpio_AND_driver.i2c_AND_driver.port_AND_driver.rtc_AND_driver.smc_AND_driver.uart" isDefaultVariant="1">
      <description>Devices_project_template MK64F12; {for-development:SDK-Manifest-ID: project_template.MK64F12.MK64F12}</description>
```

 I wonder if the current proposal could support what is encoded between braces in description node

mark-edgeworth, 2022-03-30:
This seems to be a way of adding vendor-specific extensions to the standard. By their very nature these would be non-portable between vendors and hence against the concept of an open standard. If there are concepts and data that are generally applicable and useful then these should form part of the standard. If they are for use only in a specific environment then these should be separate from the standard.

fred-r, 2022-04-04:
To me, we can specify in a standard a "standard field" by specifying:
- its format
- its purpose
- but not necessarily specifying the list of possible values.

So to me the proposal is to add a standard field, whose pick-list can be customized.

If I make a parallel with some past experience, it is like 3GPP specs where you have USSD: Unstructured Supplementary Service Data. This is customized per PLMN/mobile vendor for the features but the "pipe" to convey it is specified.

So, here, it is the same thing, we specify the "pipe" (the standard field) but it can convey some vendor specific information because silicon vendors can have specificities:
- specific features
- specific naming for a feature

So, to me, a customizable field is not against the standard.

DavidJurajdaNXP, 2022-06-21:
> I remarked this piece of code in another issue (98):
> 
> ```
> <component Cclass="Board Support" Cgroup="SDK Project Template" Csub="project_template" Cvariant="MK64F12" Cversion="1.0.0" type="config" condition="device.MK64F12_AND___component.serial_manager_AND_utility.debug_console__OR_utility.debug_console_lite__AND_component.uart_adapter_AND_device.MK64F12_startup_AND_driver.adc16_AND_driver.clock_AND_driver.common_AND_driver.dspi_AND_driver.gpio_AND_driver.i2c_AND_driver.port_AND_driver.rtc_AND_driver.smc_AND_driver.uart" isDefaultVariant="1">
>       <description>Devices_project_template MK64F12; {for-development:SDK-Manifest-ID: project_template.MK64F12.MK64F12}</description>
> ```
> 
> I wonder if the current proposal could support what is encoded between braces in description node

We used "description" element for notes during development. I am not recommending to embed any data in this way in production. There should be dedicated element for any extension data.

closed 2022-06-21 by JonatanAntoni