Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #26: Component Level Configuration Gap

jkrech, 2021-09-14:
Today's CMSIS-Pack description only allows a single dimensional "feature - selection" using Cvariant. Multiple features can only become selectable if all combinations are specified (requires duplication of component description for each case).

Extend CMSIS-Pack component description with component configuration parameters which can be used for impacting:
- component dependency
- active files (source and header files as well as include paths)
- macros RTE_Components.h
- configuration to be passed to generator

tcsunhao, 2022-04-05:
Any progress/update on this topic?
In NXP cases, all component level configurations are **target specific**, like target specific Marcos(DEBUG, NDEBUG). 
A real case is driver.xip component, today we have followings:
```xml
<component Cclass="Device" Cgroup="SDK Drivers" Cvariant="flash target" Csub="xip_device" Cversion="2.0.2" condition="device.MIMXRT1064_AND_driver.common">  
    <RTE_Components_h>  
    #ifndef XIP_EXTERNAL_FLASH  
    #define XIP_EXTERNAL_FLASH 1  
    #endif  
    #ifndef XIP_BOOT_HEADER_ENABLE  
    #define XIP_BOOT_HEADER_ENABLE 1  
    #endif  
    </RTE_Components_h>  
    <files>  
        <file category="header" name="xip/fsl_flexspi_nor_boot.h"/>  
        <file category="sourceC" name="xip/fsl_flexspi_nor_boot.c"/>  
    </files>  
</component>  
<component Cclass="Device" Cgroup="SDK Drivers" Cvariant="ram target" Csub="xip_device" Cversion="2.0.2" condition="device.MIMXRT1064_AND_driver.common">  
    <RTE_Components_h>  
    #ifndef XIP_EXTERNAL_FLASH  
    #define XIP_EXTERNAL_FLASH 0  
    #endif  
    #ifndef XIP_BOOT_HEADER_ENABLE  
    #define XIP_BOOT_HEADER_ENABLE 0  
    #endif  
    </RTE_Components_h>   
    <files>  
        <file category="header" name="xip/fsl_flexspi_nor_boot.h"/>  
        <file category="sourceC" name="xip/fsl_flexspi_nor_boot.c"/>  
    </files>
</component>  

```

One alternative way is like
```xml
<component Cclass="Device" Cgroup="SDK Drivers" Csub="xip_device" Cversion="2.0.2" condition="device.MIMXRT1064_AND_driver.common">
    <Cvariant name="flash target">
        <RTE_Components_h Cvariant="flash target">
            #ifndef XIP_EXTERNAL_FLASH
            #define XIP_EXTERNAL_FLASH 1
            #endif
            #ifndef XIP_BOOT_HEADER_ENABLE
            #define XIP_BOOT_HEADER_ENABLE 1
            #endif
        </RTE_Components_h>
    </Cvariant>
    <Cvariant name="flash target">
        <RTE_Components_h>
        #ifndef XIP_EXTERNAL_FLASH
        #define XIP_EXTERNAL_FLASH 0
        #endif
        #ifndef XIP_BOOT_HEADER_ENABLE
        #define XIP_BOOT_HEADER_ENABLE 0
        #endif
        </RTE_Components_h>
    </Cvariant>
    <files>
        <file category="header" name="xip/fsl_flexspi_nor_boot.h"/>
        <file category="sourceC" name="xip/fsl_flexspi_nor_boot.c"/>
    </files>
</component>
```

jkrech, 2022-04-05:
@tcsunhao, sorry we did not investigated into this further. Thanks for sharing your proposal, above. It manages the variants under a component but the foremost restriction still applies that it only give you a single dimension. If you had to parameters with two states this would mean you need to create 4 variants to cover all combinations in a single variant.
I am wondering whether your example above take a similar direction as component instances here #76 ?

tcsunhao, 2022-04-06:
@jkrech, our case is not in the similar direction as component instances. 
It is about how to organize configuration data in component level. In project level, configuration is organized somehow in target because configuration serves target directly.
We are expecting progress in this topic because we have issues related to component configuration.

ReinhardKeil, 2022-04-06:
@tcsunhao I believe this relates somewhat with the "Generator" concept that is proposed and this is the next topic that I will address - so your request comes timely.

The concept proposal is described here https://github.com/Open-CMSIS-Pack/devtools/blob/main/tools/projmgr/docs/Manual/YML-Format.md#workflow

An element of it are configuration parameters (currently under instances) which looks very similar to your request.
```
 component: Device:HAL:UART
      instances:
        - WiFi: 0
          setup: wifi-config.json
          baudrate: 19200 
        - Debug: 2
          baudrate: 57600
```
In your case you create variants to actually provide configuration parameters to the very same set of files.   Initially a variant was intended to provide the same functionality, but with a different set of files, for example:  
  - "Library" where a library is included in the project
  - "Source" where source code is used instead
  - "Debug" where the component adds additional source files to perform debug output

In your case it is a pure (#define) option to the component.  The concept of ST assumes that there will be a template generator that could digest the different options and create, for example a header file that reflects it.  

If you are OK, we should prepare something for next week here to review in the Technical meeting on Tuesday.  I understand that the "Generator" concept might be overwhelming and maybe we can reduces it or provide different levels of implementations.

tcsunhao, 2022-04-08:
Thank you very much for the detailed explanation, @ReinhardKeil.

I believe it will be very welcome if we can get something related to this topic to review in next technical meeting.

I cannot agree more that "Generator" concept might be overwhelming, not all entities have the Generators. We need a more easy-access or fixed way to deal with such cases.

tcsunhao, 2022-04-08:
In pdsc we already have a way to define macros: RTE_Components_h. Of course pdsc can provide a more dedicated way.  
But what we need is not the macro, it is about the conditional macro, or more general, conditional configuration. 

Generally, open CMSIS could provide certain way to make developers/users to be able to set different conditions to different parts (mainly configuration, like defines)of components. In the project which uses this component, if certain condition inside the component is met, then that part configuration is included into the project(maybe a specific target), other not met condition part will not be included.

For NXP case, our original request is that, if users use driver xip inside projmgr yml(or keil project definition file) to construct a project with different targets. Certain target can get corresponding defines, like XIP_EXTERNAL_FLASH: 1 for flash target, XIP_EXTERNAL_FLASH: 0 for ram target.

It could be like
```xml
<component Cclass="Device" Cgroup="SDK Drivers" Csub="xip_device" Cversion="2.0.2" condition="device.MIMXRT1064_AND_driver.common">
    <condition id="flash target">
        <RTE_Components_h Cvariant="flash target">
            #ifndef XIP_EXTERNAL_FLASH
            #define XIP_EXTERNAL_FLASH 1
            #endif
            #ifndef XIP_BOOT_HEADER_ENABLE
            #define XIP_BOOT_HEADER_ENABLE 1
            #endif
        </RTE_Components_h>
    </condition>
    <condition id="ram target">
        <RTE_Components_h>
        #ifndef XIP_EXTERNAL_FLASH
        #define XIP_EXTERNAL_FLASH 0
        #endif
        #ifndef XIP_BOOT_HEADER_ENABLE
        #define XIP_BOOT_HEADER_ENABLE 0
        #endif
        </RTE_Components_h>
    </condition>
    <files>
        <file category="header" name="xip/fsl_flexspi_nor_boot.h"/>
        <file category="sourceC" name="xip/fsl_flexspi_nor_boot.c"/>
    </files>
</component>
```
In the project which uses this component, in the csolution yml, there could be some attribute like "condition" under certain build-types to say that under this build-type, we provide which condition to work, like
```yaml
  build-types:
    - type: flash debug
      compiler: AC6
      misc:
        - C*:
            - -O1
            - -g
      condition:
        - flash_target
```

There are 2 difficulties in this way:
1. How to define the condition to make it identical
2. How does the outside users know about the conditions, so that they can use it in ymls. It is not proper to let them open pdsc to search. 

I used "Cvariant" firstly to try to workaroud such difficulties.  Not sure this is a proper direction to solve the issue.

(For the following, I think it might be not proper to set defines again under components because these defines are already inside the component definition pdsc. What should be here is actually the "condition" in certain build-type. 
```yaml
   - component: Device:SDK Drivers:xip_device
    defines:
      - XIP_BOOT_HEADER_ENABLE: 1
      - XIP_EXTERNAL_FLASH: 1
```
)

fred-r, 2022-04-08:
I think we must be careful not to mix composition and configuration.

The PDSC describes the components, what they offer.

The YML defines the components we assemble.
Configuring the components is another step.

To me if you want to say:  I have a component A with a configuration for FLASH and another one from RAM then it is 2 variants of the same component.
So, either you can accept to create 2 variants and then "at composition stage" the user decides which variant he wants (FLASH or RAM), or you make it a configuration choice.

In this case, the user selects the same component then configures it.
The configuration method can be many things.
You may very well provide a header file with attribute "configuration" (see https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_components_pg.html#FileAttributeEnum) and the end-user may change his defines based on how he wants the component to behave.

So, to me you have2 choices:
- either the user selects a variant of acomponent based on what he wants to do ==> composition choice expressed in YML to decide which varaint of a PDSC component he takes
- or you select a component that the end-user can configure afterwards, maybe via attr=config or any other method of your choice. But I do not think it is the role of the YML to handle the configuration. But, the YML can reflect the configuration.

tcsunhao, 2022-04-21:
@ReinhardKeil I rethink your following way, I think it is really pragmatic(Sorry for previous comment).
```yaml
  - component: Device:SDK Drivers:xip_device
    defines:
      - XIP_BOOT_HEADER_ENABLE: 1
      - XIP_EXTERNAL_FLASH: 1
```
I think just provide Marco definition is not enough. It is pragmatic if we can directly provide "config" file to replace the existing one in SW pack, so I raise [devtool#296](https://github.com/Open-CMSIS-Pack/devtools/issues/296) for component configuration after component selection in project file(no matter yml/cprj/uvprojx).

tcsunhao, 2022-06-10:
When review existing NXP component data, just find that there is no way to specify toolchain/compiler specific macro definitions in component pdsc definition.
All macro definitions are defined in RTE_Components_h which are shared by alll toolchains, not specific one.
(A easy backward compatible way to add an optional attribute "condition" for it.)

jkrech, 2022-06-10:
@tcsunhao , alternatively you should be able to make use of the pre-processing capabilities of the toolchains already now.
If you take a look at the ARM CMSIS pack: CMSIS/Core/Include/cmsis_compiler.h where the different compilers are identified 
```
#if defined  ( __ARMCC_VERSION)

#elif defined ( __GNUC__)

#elif defined ( __ICCARM__ )
...
#endif
```
I would be interested to better understand the use case of such toolchain/compiler specific macros specific to components. Could you please provide some examples?

tcsunhao, 2022-06-13:
@jkrech , for the time being, we don't have real example for it. We also use following to support it.
```
#if defined  ( __ARMCC_VERSION)

#elif defined ( __GNUC__)

#elif defined ( __ICCARM__ )
...
#endif
```

I just it might be an improvement point.

ReinhardKeil, 2023-10-30:
@tcsunhao is this still relevant?  If not, could you please close the issue.

ReinhardKeil, 2024-01-30:
closing as there is no further activity

closed 2024-01-30 by ReinhardKeil