Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #27: CMSIS-Pack provisions for 'temporary'/'vendor specific' extensions (Gap)

jkrech, 2021-09-14:
Currently: device subsection <environment` name="MyVendor" ... environment>

Where exactly we think it could help to add extension points (goal is to avoid to many add-ons in the corresponding schema )

We think that below are placements that would / could be useful:

    at the root (same level as <keywords> or <requirements> so that environment specific information can be added if needed)
    under component and components, in case of specific components where special "work-around" might be needed
    under condition and conditions (depending on the outcome of discussions around <produces> <consumes> or "features" discussions)
    under generators and generator as this may be very toolset specific and extra specific info might be useful
    under files and file, in case specific file format of information, or even environment specific files that wouldn't be meaningful for other environments

Note: the use of environment extension points should be as limited as possible and any GAP that cannot be implemented with the standard should be discussed in open-cmsis-pack forum to find a standard way to address it. Nevertheless adding this possibility may provide vendors and pack providers enough freedom to temporarily implement work-arounds and achieve project deadlines while the standard evolutions are discussed.

jkrech, 2022-06-02:
@fred-r could you please add a comment to this issue. I think your name need to be listed as "participant" in the side bar in order to assign the issue to you.

fred-r, 2022-06-02:
@fred-r 
I am  volunteering for being the assignee

fred-r, 2022-06-21:
@jkrech and @ReinhardKeil : usually you are in favor of "step by step" approach.
So here I would propose to get started with the minimal change we propose (only components) and see the adoption.
If more needs arise we may extend it.
If it becomes too messy, at least it is limited to components...

What do you think ?