Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #28: Component FileCategoryType GAP

jkrech, 2021-09-14:
### add support for source type/language specific include paths specification by extending FileCategoryType
- for all sourceAsm new enumeration values _includeAsm_ and _headerAsm_ paths shall be applied
- for all sourceC new enumeration values _includeC_ and _headerC_ paths shall be applied
- for all sourceCpp new enumeration values _includeCpp_ and _headerCpp_ paths shall be applied

Note: for backward compatibility the existing include and header categories are applied to all file categories Asm, C and CPP for backward compatibility. Tools shall ensure that they don't rely on the split by source language.

tcsunhao, 2022-03-18:
Agree with this proposal.

I have questions about what is the way KEIL get "include" from component. 
Some component pdsc provides explicit include with <file category="include" name="xxxxxxx"/>
Some component pdsc doesn't provide it.

Are the following steps used by KEIL to get the include path for component?
1. If a component doesn't provide explicit include, then get all category="header" file paths as include paths
2. If a component provide explicit includes, then directly use the include path.(won't analyze category="header" anymore)

If above steps are not the one used, then how KEIL decides include paths for a component?

jkrech, 2022-03-18:
You can think of it like this:
a) get all category="header" file paths as include paths
b) add all category="include" file paths
c) remove duplicates

Note that the reason for specifying a file as category="header" explicitly, means that this file forms part of the 'contract' with other source module "using" this component.
a) The filename of this header file must not change throughout the lifetime of the component.
b) Semantic Version of the component reflects the compatibility of all "header" files of the component.

category="header" shall be used for locations that only contain "internal headers".
There is discussion to allow to explicitly list all header files. Therefore e.g. a category="headerInternal" could be a useful extension.

tcsunhao, 2022-03-18:
> You can think of it like this: a) get all category="header" file paths as include paths b) add all category="include" file paths c) remove duplicates

There seems to an issue with this approach: If a header file is included in a source as a relative way like (include "xxx/xxx.h"), then the head file path itself should not be added into project include, because it is useless. There might be warning like redundant include path in the project build. 

> category="header" shall be used for locations that only contain "internal headers". There is discussion to allow to explicitly list all header files. Therefore e.g. a category="headerInternal" could be a useful extension.

I totally agree with list all headers.  "a category="headerInternal" could be a useful extension." do you mean headerinternal is direct the include path?

jkrech, 2022-03-18:
Point 1:
For file category="header" there is an optional attribute path 
see: https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_components_pg.html#element_file
This separates the include path from the specification of a public header.
Point 2:
If each and every component internal header file (=must not be explicitly included by other code) shall be explicitly listed for a component, then you would use category="headerInternal". The processing would be the same. The include path would be extracted and duplicate include paths removed.

Am I missing something?

tcsunhao, 2022-03-18:
Thanks for the explanation @jkrech , the "path" attribute should work for include "sub_dir/includeFile.h". I have no more issues about this topic.

closed 2022-07-04 by JonatanAntoni