Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #29: What is the taxonomy to be used for a component which is a library ?

fred-r, 2021-09-23:
Hi,

I'd like to identify some components which are libraries (say a .a file).
How can i indicate this with the taxonomy ?

Is it via the CgroupType ?
https://www.keil.com/pack/doc/CMSIS/Pack/html/element_taxonomy.html#CgroupType

It is not clear to me if there is a standard Cgroup for this kind of components ?

Thanks & Regards,
Fred

jkrech, 2021-09-23:
Hi Fred,

the [taxonomy](https://arm-software.github.io/CMSIS_5/latest/Pack/html/element_taxonomy.html) is there to provide "Semantic" to the component, however whether a component content is a library or a set of source modules and header files is captured by the [file](https://arm-software.github.io/CMSIS_5/latest/Pack/html/pdsc_components_pg.html#element_file)'s element attribute [category](https://arm-software.github.io/CMSIS_5/latest/Pack/html/pdsc_components_pg.html#FileCategoryEnum).

E.g.: in case of the Keil RTX5 component (Cclass="CMSIS", Cgroup="RTOS2", Csub="Keil RTX5") we initially provided a library. Later we have decided to offer the component as source code. Therefore we introduced a Cvariant="Source" and a Cvariant="Library" but this did not impact the rest of the component identification.

Does that answer your question?

fred-r, 2021-09-23:
Hi Joachim,

so the idea is not to indicate a component as a library but more a file into a component right ?

https://www.keil.com/pack/doc/CMSIS/Pack/html/pdsc_components_pg.html#FileCategoryEnum

So to identify if a component is a library I go down to the file level ?

Is that the approach you would recommend ?

Thanks & Regards,
Fred

jkrech, 2021-09-23:
Hi Fred,

correct. You need to specify the type of file(s) within a component and identify a library there in any case. (The build tools need to know whether the file needs to be passed to the linker or the compiler). 
I would strongly discourage e.g. inventing a Cclass="Library" to tell the user that the component contains a library. This should be irrelevant to the user.

ReinhardKeil, 2021-12-14:
Can we close this item or does someone feel that it needs more discussion?

fred-r, 2022-01-25:
As the submitter, I agree to close this point.

closed 2022-01-25 by fred-r