Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #30: Extend configuration file description to specify destination folder and filename

jkrech, 2021-09-30:
This request for extension got raised in [Issue #10](https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack/issues/10) Gap Analysis:

Add the ability to specify the destination directory and filename of a configuration file explicitly.

```
  <file name="CMSIS/RTOS2/RTX/Config/RTX_Config.h"        category="header" attr="config" version="5.5.2"/>
  <file name="CMSIS/RTOS2/RTX/Config/RTX_Config.template" category="header" attr="config" version="5.5.2"
        destination="RTX/RTX_Config.h />
```
Today CMSIS-Pack specification mandates that the destination folder is RTE/<Cclass>/[<Dname>/]<name.file()>

Note: in order for projects to remain "self-contained" destination directories are specified relative to project file and must be located in the project directory of a subdirectory ('../' not allowed)

jkrech, 2021-09-30:
Background provided by @LMESTM:
CMSIS PACK should allow to distribute existing software, and therefore needs some flexibility to map to project layouts. If distribution through CMSIS packs requires rework of the project software structure, then it may lower the adoption level.

In the discussion, I am considering both cases of files of attribute config and templates. In our solution, we're providing the config file as a template: cubeMX will generate it by default, if user does not use cubeMX, he will have to create the configuration file based on the template that is provided with _template suffix. This file is named with _template suffix to make it clear this is a user file . Because of suffix (or thanks to it) , it is not an issue that the file is placed inside the include folder because other files will not include it ... but it requires renaming.

We are thinking about other use cases as well.

    for instance, our cubeMX generator can "generate" files from templates, and we would like to be able to offer configurable names and locations for different SW layers to avoid having all the generated files in a single hard coded place. Typically it allows to generate i2c.c file in a $Project/User/Core folder and we would like to be able to cope with this kind of case.
    One other possible use case is multiple instances of a single template (.c or .h files) - different target folders and/or different target names would open for this possibility. In the case above, some users may prefer to generate i2c1.c for instance 1 and i2c3.c for instance 3 of I2C.

slhultgren, 2021-10-06:
From a user point of view it would be good to have more control over the files that are placed into my project.

The .cprj spec could be extended to allow specifying a general new "BASE_FOLDER" for these tool-created files.
It should be possible to specify (and split) attr="config" files from attr="template" files (and gpdsc files) so that the end user can have complete control over the final relative placement of the files.

Perhaps introducing $CONFIG_BASE_FOLDER and $TEMPLATE_BASE_FOLDER into the .cprj spec and then using these when populating files in the project.

LMESTM, 2022-04-20:
@ReinhardKeil  as mentioned in #116 and #30 the file with attribute "template" also needs to be addressed, not only config files. I think this is not part of the current proposal.
Also I think we should be able to mention if a template file is expected to be copied into $RTE folder (or sub-folder) or in $project folder

fred-r, 2022-04-20:
What I would prefer is really to be able to specify on a per file basis to have full flexibility.

If this is not possible, then we should at least be able to determine the folder for:
- components section of the project
- user section of the project

closed 2023-05-30 by jkrech