Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #32: Discuss selection of Template engine used by Open-CMSIS-Pack tools

jkrech, 2021-09-30:
In the context of [#12](https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack/issues/12) the requirement for a template engine was raised.

Identify available open-source template engines which are suitable for generating
- source modules (C/C++/asm, ...)
- header files (C/C++, ...)
- linker scripts 
- other ?

- [Freemarker](https://freemarker.apache.org/)
- [handlebars](https://handlebarsjs.com/) 
- other?

Create a list of Pros and Cons, discuss and agree and create ADR

mdortel-stm, 2021-10-05:
You can find here a comparison matrix of a log of engines: https://en.wikipedia.org/wiki/Comparison_of_web_template_engines
Even if not strictly true, we can take the first assumption that handlebars is like mustache.

fred-r, 2021-10-05:
Maybe we should clarify what the templating engine will address ?
I guess we are talking here about the equivalent of CMSIS-Zone in term of features ?
Or is it more than this ?

0xc0170, 2021-10-06:
>Maybe we should clarify what the templating engine will address ?

As I read the first comment, source files for a project (not project files like cproject or similar) -  a tool would generate these files based on project configuration. Looking at the templates in CMSIS-Zone, confirms this, source files generated - basically config files, linker file or any other source file that can be generated based on project configuration.

What languages are OK to be used - no requirement there? Looking at freemaker it's written in Java. Handlebars in javascript.

fred-r, 2021-10-06:
> What languages are OK to be used - no requirement there? Looking at freemaker it's written in Java. Handlebars in javascript.

Good point indeed. What is the technology to be used in this project ?

LMESTM, 2021-10-12:
@ReinhardKeil  do you mean: "the language that it is written in is _not_ important"  ?

jkrech, 2022-03-08:
I am closing this issue, as we have concluded to focus on using Handlebars for now.

closed 2022-03-08 by jkrech