Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #43: Feedback on Pack Manager Overview.md

ReinhardKeil, 2021-11-09:
An Overview of the Pack Manager is here:
https://github.com/brondani/devtools/blob/mvp-prototype/tools/projmgr/docs/Overview.md

For now, please post your feedback using this issue.

LMESTM, 2021-11-10:
Here are a few questions and feedbacks concerning this proposal.

1- is csolution a workspace-like environment where users can list all the projects they are working with or is it meant to support a specific application and offers specific added value for serving this application ?  
In previous proposals we were considering .ctarget as what we called an "application project", that would provide the full picture of all the projects required for building a possibly complex application.  
With current proposal, it looks like a user may use csolution to list completely different "applications" like "secure lock on target A", then "connected fan on target B" maybe because they share a few components together. Is that part of what we want to support ?

2- Clayer and Cproject allow to projects or sub-projects 
If the csolution would apply to one target type at a time, clayer and cproject would anyway allow sharing common parts between 2 (or more) projects. Each csolution project could also include the target specific part without impacting the other project. 

3- supporting different targets in csolution  raises many questions : 
- as depicted in example diagram, the various projects may contain very different SW layers and fitting all of them inside a common structure may bring more noise (exclude / include management) than just sharing sub-projects or layers between 2 different projects https://github.com/brondani/devtools/blob/mvp-prototype/tools/projmgr/docs/images/TargetBuild-Types.png
- the main diagram explains that ctarget.yml (BTW: typo, name wasn't updated) should include azone.info. I guess this can have added value as the csolution can provide a consolidated view of the memory mapping and resource assignments (secure / non secure memory map, peripherals assignments, bootloader flash area vs. application flash area, etc....) but how will that work when  multiple HW are supported ?
- same for compiler options, what if virtual platform requires different options than real mcu, do we need also  include and exclude mecanisms for compiler options ? same for defines ...

4- worskspace with multiple csolution projects ?
if you get a workspace with several csolution application projects, your workspace can provide you a list of solutions and target-types you support in your workspace, and that could possibly offer the user the possibility to switch between different target-types for the related solution, e.g. switch between iot-product~virtual.csolution.yml and iot-product~production-HW.yml .

ReinhardKeil, 2021-11-11:
1. The csolution.yml file is not related to a workspace.  We defined it right now as: A solution is the software view of the complete system. It combines projects that can be generated independently and therefore manages related projects.

2. You are right, it might be sufficient to have csolution to support just one target at a time.  However, we got feedback that this might be not provide the flexibility required. If we need at the end the 'target-type' approach or not is something that should be evaluated (with several test projects). It is possible to exchange layers based on target/build type selections - so yes the concept you describe is there.

3. You are raising a valid risk here, and this approach will need clear guidance to users what makes sense and what should be avoided. And clearly we will need to do work on the rzone assignment - this is ahead of us and I plan to get the concept done before XMAS.  There are already provisions for different compiler options (specific to a target), so that problem is already addressed. I can provide an example if needed.

4. As said, we should explore the best way to work with build/target-types.  Let's not come to early to conclusions that it is too complex.  I believe managing projects with build/target-types simplifies, but I might be wrong.  We should get examples before we judge.  It is always possible to decide that target-types do not get exposed to the users.

fred-r, 2021-11-11:
Hi,

as you may expect I agree with LMESTM's point of view.

One more element I'd like to highlight again is the fact that I am not convinced bu the "solution" wording : see my points [here](https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack/issues/33#issuecomment-957963415)

ReinhardKeil, 2021-11-12:
Fred, 

thanks for your good contribution. I like your user stories and agree we should evaluate it. As "full stack" example I believe we can use a project like this: https://github.com/arm-software/ATS-Keyword.  And we could map it to:
  - Nucleo board viewed as "production hardware"
  - Disco board (with similar device, but a superset with different pin mapping) viewed as "test board"
  - Virtual Hardware (simulation) for testing.

I did not like "solution" initially. But Visual Studio introduced it ~20 years ago and many programmers know already what it is. We can go around our virtual "Open-CMSIS table" to see if there are better ideas.  IMHO a single word works better, so "product" or "system" might be feasible too.  (Multi-project Workspace - the MDK term - is clumsy).  In context of all the complexity, I feel this is a detail.

Reinhard

"azone/rzone" needs more work - your picture with the different "context/resource" boxes is good.  We should find a way that users can understand and manage such systems.

fred-r, 2021-11-12:
Well, to me as you said solution has a strong "track record" in the dev community and I immediately associate it to an sln file so Visual Studio solution.
I would prefer something showing more clearly we are dealing with embedded SW for a target HW. But I agree it is tricky and probably not critical.

LMESTM, 2021-11-15:
> thanks for your good contribution. I like your user stories and agree we should evaluate it. As "full stack" example I believe we can use a project like this: [...]
 
@ReinhardKeil  In your above proposal, all the boards / HW types seem to refer to the same device. in the user stories suggested by @fred-r there are different devices, typically one is trust-zone enabled, the other one is not.

ReinhardKeil, 2021-11-18:
Laurent, when looking at the Nucleo/Disco STM32L5 offering the devices on the boards are different and also the pinout differs.  The idea is to mimic production hardware (on a device with less pins/memory). If the example ATS-Keyword is suitable or not needs to be evaluated.

LMESTM, 2021-11-19:
Yes I understand that it _could_ make more sense for HW types that are closely related one to the other, but if you open this door, do you plan to restrict users to select only closely related HW ? how can we ensure this ?

0Grit, 2021-11-19:
@ReinhardKeil  It sounds like the discussion above is trying to determine what to call the 'application' now that managing multiple execution contexts, and processors on a single board is desired?

As someone who has done a decent amount of schematic & layout but specializes in firmware & IoT;

* I would want to be able to describe the hardware unit (PCB) and any logic specific to bringing up or managing that hardware to exist within a BSP pack. 
* That pack would contain tests and a reference to a sample external to said pack. 
* I would not want any samples for that board in the pack because that breaks my mental model for the "full stack".
* That BSP should be 'optionally' dependent packs that add processor support and drivers for other IC's on board.

I think I would have application level 'projects' for each processor, and possibly a project per zone needed by my application on a particular MCU.

I could live with the top level collection of applications/projects being called a "Solution" or "System".

0Grit, 2021-11-21:
Also. Do not forget about Rust. It is quickly gaining momentum. New engineers feel no sense of attachment to C and will drop it like a rock.

Please 🥺🙏 consider looking at how Cargo and crates can be either used for inspiration or used directly to manage packs. Extend it if you have to.

iomint, 2021-11-22:
The order of processing of the project files induces the overlay order of the elements that can be present at various levels like the  general properties keywords.  In the proposal, the .csettings.yml is processed first (even before the .csolutions.yml) but when I look at the list of the general properties keywords I think it is sufficient to apply them at a lower level  .project.yml except  compiler: and defines: (for symbol and path settings) that sounds general and could correspond to a "workspace" like environment (concept introduced by LMESTM in the first point of his first post of this thread https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack/issues/43#issuecomment-965463956 ) where users can list all the projects they are working with.

iomint, 2021-11-22:
> Also. Do not forget about Rust. It is quickly gaining momentum. New engineers feel no sense of attachment to C and will drop it like a rock.
> 

@0Grit : I do not see an issue in what is proposed here to handle a project from packs containing sw components developped in C language or in Rust. Do you ?

0Grit, 2021-11-22:
Build system is abstracted from packaging? If so I trust your assertion.

 I'm strapped for time and am unable to do much more than skimming of the materials. Just wanted to make sure these things are at least mentioned in the conversation. I've seen people like @ilg-ul who have a wealth of knowledge and work put into projects like this, get ignored in the past.

fred-r, 2021-11-22:
Well, I would think that:

- the proposal is based on CMSIS-Pack components, so we can have a component providing RUST code
- when it comes to the build chain, the spec lets it open I guess. Today it is proposing GCC, IAR and AC6 as a compiler but I assume we may support a toolchain with bot C and RUST support ?
- or maybe more realistic would be to mix C project and RUST projects compiled with their own toolchain each ?

I assume the major point you highlight is how to use both cargo and CMSIS packs ?

ilg-ul, 2021-11-22:
After embracing the CMSIS Packs idea and trying to use them  for my projects (long time ago), I finally realised that their design has different goals, that do not match my criteria, especially in providing layer separation and reduced complexity.

Thus I designed **xPacks**, which are a more generic version of the highly successful, industry standard, **npm packs**.

In terms of the first criteria, layer separation, source xPacks are completely neutral to build processes, languages, platforms, etc. 

In practical terms, source xPacks are collections of files with static dependencies to other xPacks, the same as npm packs are nodejs modules with dependencies of other packages.

Satisfying dependencies is a separate step that is executed before the build, to bring in all dependent packages, sources or binaries.

As for the second criteria, reduced complexity, CMSIS Packs favour very large packages (I saw some weighting hundreds of MB), with many, many source files, grouped in lots of components, each with its own version.

On the opposite end, xPacks (and npm packs), favour many small packages, each package having its own version, which applies to all files in the package.

Another major difference is that the main CMSIS Packs repository is centralised and not automated (or at least it was not when I used it, it required someone to accept the packages, and to update the index, which is, most of the time, outdated with vendor packages).

xPacks, being in essence npm packs, can be stored on any npm compatible server, like npmjs.com, GitHub, etc.

My 2 cents...

fred-r, 2021-11-22:
"As for the second criteria, reduced complexity, CMSIS Packs favour very large packages (I saw some weighting hundreds of MB),with many, many source files, grouped in lots of components, each with its own version."

==> to me this is the way CMSIS packs have been used so far.
The spec does not force you to do so.
I agree that it would be a much better practice to design smaller packs not mixing very different things.
The pack is only the way to convey the CMSIS components : this is the important point, designing consistent components.

0Grit, 2021-11-22:
I highly recommend looking at @ilg-ul's x-pack project. I've used it and it's a great experience. He has packs to install toolchains, cmsis, etc. A lot of it sitting on top of NPM's infrastructure too.

The only thing preventing it from widespread usage is that he is basically the only person working on the project.
You would think he had a team considering how extensive the project is.

It would bring a tear to my eye to see some of his work be used by open-cmsis-packs somehow. Even if only by inspiration; though just using xpacks by itself and adapting the open-cmsis-packs spec to it would save everybody some time.
Not sure about how or if xpacks handles the "zones" concept though.

ilg-ul, 2021-11-22:
If allowed, multi-component packages may present additional challenges. Imagine you have multiple versions of a package which defines multiple components, each with its own version, all different in each package. 

For whatever reason, a project may decide that it needs one component from one package version and another component from another package version. I did not check the recent specs, so I apologise if this is nonsense, but can the proposed package manager handle cases like this?

ilg-ul, 2021-11-22:
> Not sure about how or if xPacks handles the "zones" concept though.

I don't know what the zones are, but I did a lot of thinking and experimenting with packages, and my current understanding of the subject is that any design that requires the packages to be identified by anything more than a (scoped) name and a semver, is a design too complicated.

As for what a package is, according to the **npm** documentation, _"a package is any of the following"_:

1. A folder containing a `package.json` file.
2. A gzipped tarball containing (1).
3. A URL that resolves to (2).
4. A `<name>@<version>` that is published on the registry with (3).
5. A `<name>@<tag>` that points to (4).
6. A `<name>` that has a latest tag satisfying (5).
7. A git url that, when cloned, results in (1).

I personally could not find a more generic definition.

0Grit, 2021-11-22:
@ilg-ul see the discussion earlier up. https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack/issues/43#issuecomment-966479145

ilg-ul, 2021-11-22:
Yes, I saw it, but for me such a project is too complicated to be a single package.

Small simple packages are also much easier to test, and in environments like GitHub, you can run the unit tests automatically on push.

fred-r, 2021-11-23:
The example provided above (https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack/issues/43#issuecomment-966479145)  and referenced by @0Grit  is clearly not a single package.
It is an application project made of several software projects.
Each software project is the composition of several CMSIS components (via several CMSIS packs).
I agree there is complexity but it only reflects the complexity of the systems we are dealing with.

I think we should not mix all topics:
1.  I understand that you challenge the CMSIS-pack technology => to me x-pack covers this based on what you explained, even if I think that a "proper" usage of CMSIS-Pack should do the job...to be discussed by @jkrech and @ReinhardKeil 
2. There is the project management aspect => here nobody said an application project should be a single pack

So I would recommend focusing this discussion thread on subject 2 and maybe @jkrech can redirect the 1st topic to the appropriate issue ?

iomint, 2021-11-23:
We are indeed deviating from the subject of this thread but these are interesting discussions for which we can open new issues thread or redirect to existing relevant ones.
During early discussion to set-up this open-CMSIS initiative, we have agreed to start from CMSIS-Pack (It has a certain footprint in the embedded with many MCU vendors publishing CMSIS packs) and see how to improve the spec and push an open reference implementation;   @ilg-ul I invite you to look at the others issues - in particular #10 and #11 - to get an overview of the improvements we are looking for. It is great if there are good things pointing in the same directions from your xPack work that we can pick-up. It can helpful if you could have a look to #10 and #11 and flag to us the relevant elements of xPack.  
About too large/complex (CMSIS) pack I fully agree with @ilg-ul concern and the feedback from @fred-r. How to favor (enforce ?) design practices to end up with "small" packs is  a very good topic and will deserve a dedicated issue thread. We can provide examples/patterns and/or do something with a Pack checker but where to set the limit is a tricky question.

ilg-ul, 2021-11-23:
> if you could have a look to #10 and #11 and flag to us the relevant elements of xPack.

I took a quick look, but as long as you do not have a clearly layered design, and throw everything (dependencies, build, components, configuration, etc) into the same bucket, it is hard to flag any relevant differences.

jkrech, 2021-11-23:
I have now created issue #47 and hopefully captured the comments correctly there. I am hiding the comments in this issue as they are off-topic, sorry in case the initial issue title was misleading into this direction.

fred-r, 2021-11-23:
> Fred,
>
> thanks for your good contribution. I like your user stories and agree we should evaluate it. As "full stack" example I believe we can use a project like this: [...]

@ReinhardKeil : I am not sure ATS is the best choice for a first shot.
It requires an AWS account and, it deals with Audio as well as ML ?
Besides, I am not surea bout the adherence with the HW : I do not see many peripherals into the project.

Why not getting started with a CMSIS-Zone project ?
[https://github.com/ARM-software/CMSIS-Zone/tree/main/Examples/STM32L5](https://github.com/ARM-software/CMSIS-Zone/tree/main/Examples/STM32L5)

I would like to see how to deal with this application:
- when we have the STM32L552E-EVAL
- when we want the same application project on STM32L476RG-NUCLEO : the secure software being implemented with the Firewall instead of TrustZone
- on a simulator : how to deal with the virtualization ?
This would be user-story 4.

For user-story 1 and user-story 2, I would recommend this project:
https://www.st.com/en/embedded-software/x-cube-sbsfu.html

This provides a bootloader and a user-application.
I would like to see how we can handle just a few situations:
1. Supporting Nucleo-L476RG and Nucleo-L432KC: one has a user-button, the other one not
2. Supporting DISCO-L496 which supports 2 crypto libraries
3. Supporting STM32F769I-Discovery which is another series, has no Firewall and supports 2 crypto libraries too
4. Supporting STM32H7B3I-DK which is yet another series and has external FLASH
5. Why not a virtual device ?

For user-story 5, X-CUBE-SBSFU can be nice two as it has a dual core flavor.

For user-story 6, probably we could consider the mbedTLS library ?

fred-r, 2021-11-23:
> ects' for each processor, and possibly a project per zone needed by my application on a particular MCU.

@0Grit : I agree that the board pack should not contain examples.
Probably a discussion can be initiated about what a good pack is as suggested by @iomint .

> I think I would have application level 'projects' for each processor, and possibly a project per zone needed by my application > on a particular MCU.

My intention here is to promote one single application project per board.
This allows dealing with shared resources between components of the board.
Typically, if you have one SoC with multiple cores then we use execution contexts to associate each SW project to an HW execution unit.
If you split it in several application projects (one per core) then you are in trouble to handle shared resources in my opinion and also to do the partitioning of the system.
That's my opinion and I'll be happy to get your view on this.

0Grit, 2021-11-23:
I think the only alternative would be to have some new protocol and software developed to manage and share resources in a distributed fashion, without a central entity to manage the resource. That sounds like a nightmare so I think I'm aligned with your intent.

Your first sentence throws me off though. A board can have multiple use cases. Someone who buys an off the shelf solution will develop a different "whole board application" than another person who buys the same board for another purpose.

The board support package is a dependency of the application project and enables it to function. If I ever wanted to port the "whole board application" to another board I should be able to as long as that other boards BSP provides the execution contexts and resources needed by my application.

fred-r, 2021-11-23:
The application project contains all the SW projects running on the target board (making the application).
Now, these SW projects have:
- applicative layers => board-agnistic
- system software : RTOS, board support package (including MCU drivers)

Then yes when you move to another board you can re-use your applicative layer (the CMSIS component).
But you need a brand new project as the system software will be different at least for the BSP part (not the same MCU drivers typically, not the same pins, not the same memories). 

That's why I am not convinced one application project can support multiple HW targets easily but this is what we are going to experiment.

But the intention with the "application project" is to have a cradle allowing you to handle your application (both HW + SW).

0Grit, 2021-11-23:
What "I think I would want" are build configurations for each supported board within one top-level project.

Agree that this needs experimentation.

fred-r, 2021-11-23:
> What "I think I would want" are build configurations for each supported board within one top-level project.
> 
> Agree that this needs experimentation.

Ok, you are aligned with the proposal from our ARM colleagues.
Let's see if this is feasible without ending up with a nuclear plant system... :-)

fred-r, 2021-11-25:
What would help me also is a legend to understand the arrows in this diagram:

[image]

I mean:
- the blue arrows going from top to bottom probably mean : "references..."
- the blue arrows going from left to right have two meaning I think
     * "is an input of"
     * "generates ..."

Also the color code for the boxes is not obvious to me:
- green is clearly an enity
- light blue an light yellow must be groups of "similar elements"
- but .csolution.yml is blue wjile .cproject.yml is white : why ?

And .cproject.yml appears twice in the diagram:
- one time as aninput of the PM entity
- one time as an output of the PM entity
This puzzles me.

iomint, 2022-01-12:
Referring to the updated spec proposal shared during the week (January 11th, open CMSIS meeting): https://github.com/brondani/devtools/blob/mvp-prototype/tools/projmgr/docs/Manual/Overview.md
The project description format to use to export or import a project seems now the .cproject.yml while the .cprj file becomes a intermediate file for the compilation. Is it the way you see it ?

ReinhardKeil, 2022-01-12:
The CPRJ file allows reproduceable builds as it contains all information to work stand-alone.  We will use it in CI systems for test automation where CMSIS-Build downloads packs in case that they are missing.  Also it fixes the pack versions used during the build process.

Today the CPRJ file is also the import mechanism for the MDK IDE to support backwards compatibility. Currently *.cproject.yml is not used as import/export file for third-party IDEs as the features might not map to an 'old' IDE.  However the *.yml files are clearly structured in a way to enable IDE integration (for example with VS Code).

ReinhardKeil, 2022-02-16:
I would suggest to close this item - or did we miss anything?

closed 2022-03-02 by ReinhardKeil