Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #51: Make software and licence information associated with packs easier to find

EdmundPlayer, 2021-11-23:
Some users of Arm devices are interested in knowing what open source software and software licences are associated with device family packs.  In some ST packs, for example, we typically ship a nice license.md (Markdown) file that lists all of the software in a table and what type of licence is associated with it, e.g., Keil.STM32F4xx_DFP.2.15.0.pack from https://www.keil.com/dd2/.  However, for other packs, we don't seem to have such info available.  According to the pack tutorial page (https://arm-software.github.io/CMSIS_5/Pack/html/cp_PackTutorial.html) there's an option to create a license file that will be shown during the installation of the Pack, but it's not mandatory.  I'm opening this issue to discuss whether providing such information should be 'recommended' or 'mandatory' instead of 'optional' before a device vendor publishes a pack.  This could help end users (working with a device) to more easily find information about the software involved, without having to potentially have to scan through all the sources manually themselves. Is this something anyone has considered looking into before?

EdmundPlayer, 2021-11-23:
Good idea, I'd not realised this was an option: https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack/discussions/53. I guess we can continue with this issue or not depending on the results of any discussions.

ReinhardKeil, 2024-01-30:
This is now addressed by `licenses:` in the build information files.  However missing is:
  - documentation in CMSIS-Toolbox 
  - packchk that should issue a warning when no license information is present.

jkrech, 2024-01-30:
click-through license display and acceptance needs to be added to `cpackget` see: https://github.com/Open-CMSIS-Pack/cpackget/issues/199

closed 2024-02-06 by jkrech