Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #54: Lost with the CMSIS-pack spec URLs

fred-r, 2021-11-25:
As of now, I have referenced 3 URLs providing CMSIS-Pack version 1.7.2 information:

[https://www.keil.com/pack/doc/CMSIS/Pack/html/index.html](https://www.keil.com/pack/doc/CMSIS/Pack/html/index.html)

[https://open-cmsis-pack.github.io/Open-CMSIS-Pack/main/html/index.html](https://open-cmsis-pack.github.io/Open-CMSIS-Pack/main/html/index.html)

[https://arm-software.github.io/CMSIS_5/Pack/html/index.html](https://arm-software.github.io/CMSIS_5/Pack/html/index.html)

Hence I am a bit lost when it comes to know which one is the source of trust.
The content seems to be the same but would it be possible to have only one URL with the up to date information and the other URLs being redirected ?

Thanks & Regards,
Fred

ReinhardKeil, 2021-11-25:
Fred,

The CMSIS-Pack specification is in the process to transition to
https://open-cmsis-pack.github.io/Open-CMSIS-Pack/main/html/index.html
This will be the source of trust going forward.

Each document has semantic versioning which helps to understand the latest spec (in case that there are duplicates during the transition period).

Reinhard

https://www.keil.com/pack/doc/CMSIS/Pack/html/index.html

https://open-cmsis-pack.github.io/Open-CMSIS-Pack/main/html/index.html

https://arm-software.github.io/CMSIS_5/Pack/html/index.html

IMPORTANT NOTICE: The contents of this email and any attachments are confidential and may also be privileged. If you are not the intended recipient, please notify the sender immediately and do not disclose the contents to any other person, use it for any purpose, or store or copy the information in any medium. Thank you.

fred-r, 2021-11-26:
Ok, clear, thanks.

closed 2021-11-26 by fred-r