Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #56: CMSIS-pack conditions : is the spec clear enough ?

fred-r, 2021-11-26:
In the spec I can see:
https://www.keil.com/pack/doc/CMSIS/Pack/html/pdsc_conditions_pg.html#element_condition

So:
<html>
<body>

Cclass* | Specifies a component class name. Predefined values can be used as listed in the table Component Classes. | CclassType | optional
-- | -- | -- | --
Cgroup* | Specifies a software Component Group. Predefined values can be used as listed in the table Component Groups. | CgroupType | optional

</body>
</html>

But if I write conditions like:
        <condition id="Porting">
            <description>Port SBSFU on a STM32</description>
            <require Cgroup="Porting Layer*"/>
            <require Cgroup="Entry*"/>
            <require Cgroup="Startup*"/>
            <require Cgroup="Memory Partitioning"/>
        </condition>

Keil microvision complains:
fred-r.STM32_Bootloader.0.0.1-PoC0: condition 'Porting': error #502: error(s) in condition definition:
 'require Porting Layer*': Cclass or Cgroup attibute is missing in expression
 'require Entry*': Cclass or Cgroup attibute is missing in expression
 'require Startup*': Cclass or Cgroup attibute is missing in expression
 'require Memory Partitioning': Cclass or Cgroup attibute is missing in expression

To fix it:
        <condition id="Porting">
            <description>Port SBSFU on a STM32</description>
            <require Cclass="Security" Cgroup="Porting Layer*"/>
            <require Cclass="Security" Cgroup="Entry*"/>
            <require Cclass="Security" Cgroup="Startup*"/>
            <require Cclass="Security" Cgroup="Memory Partitioning"/>
        </condition>

Is there a point in the spec saying that Cclass is mandatory in addition to Cgroup ?
I understand that it probably makes no sense to provide only a Cgroup but I thought this was allowed.

jkrech, 2021-11-29:
You are correct, the specification describes the individual attributes but does not describe any limitations nor good use case examples. At the same time I think that this is an MDK uVision implementation specific error condition trying to give some user guidance. The fact that a user does not specify a class of a component increases the likelihood for the system to suggest a component which does not actually satisfy the requirement of the component.

jkrech, 2024-03-12:
Documentation updated. See version 1.7.32 (or higher)
https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_conditions_pg.html#element_accept

closed 2024-03-12 by jkrech