Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #57: CMSIS-Pack conditions : only 1 condition per component ?

fred-r, 2021-11-27:
Reading the spec:
[https://arm-software.github.io/CMSIS_5/Pack/html/pdsc_components_pg.html#element_component](https://arm-software.github.io/CMSIS_5/Pack/html/pdsc_components_pg.html#element_component)

It looks like a component can have only 1 condition ?

This means that a condition might become a "big set of non-related topics".
Would it be possible to associate several conditions to a component ?

jkrech, 2021-11-29:
No, there is only one reference to a condition per component. For more complex conditions you can combine conditions in a new condition which should be effectively the same.

```
    <condition id="ARMCC6">
      <accept Tcompiler="ARMCC" Toptions="AC6"/>
      <accept Tcompiler="ARMCC" Toptions="AC6LTO"/>
    </condition>

    <condition id="CM0">
      <description>Cortex-M0 or Cortex-M0+ or SC000 processor based device</description>
      <accept Dcore="Cortex-M0"/>
      <accept Dcore="Cortex-M0+"/>
      <accept Dcore="SC000"/>
    </condition>

   <condition id="CM0_ARMCC6">
      <description>Cortex-M0 or Cortex-M0+ or SC000 processor based device for the Arm Compiler 6</description>
      <require condition="CM0"/>
      <require condition="ARMCC6"/>
    </condition>
```

fred-r, 2021-11-29:
Thanks Joachim

DavidJurajdaNXP, 2021-12-01:
I would actually like "big set of non-related topics". I think it is more readable if all conditions are fully described inside the component itself, not hidden in multiple levels of remote condition aliases.

I would prefer to express conditions in form of single boolean expression per component like
(c1 AND c2) OR (c2 AND c3 AND c5 AND c6)
or
OR(AND(c1 c2) AND(c2 c3 c5 c6))

This would also simplify generation of components from alternative data sources (elimination of named condition sub parts generation).
Such feature could be probably implemented as backward compatible  extension.

This is also part of CMSIS-pack gap analysis #10 -> "specify component condition within the scope of a component"

ReinhardKeil, 2023-10-30:
@fred-r @DavidJurajdaNXP I accept that there are other and perhaps better ways to specify conditions, but how important is it to address this issue?

jkrech, 2024-02-12:
I close this issue.
The most pragmatic way to have multiple topic conditions for a component is to define a condition for a component and on the first level require multiple conditions where each represents another (independent?) topic:

```
  <condition id="myComponentID">
    <require condition="topicA" />
    <require condition="topicB" />
    ...
  </condition>

```

closed 2024-02-12 by jkrech