Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #58: CMSIS-Pack file attribute : copy not documented

fred-r, 2021-12-02:
Joachim told me I can use attr="copy" whenever I need an element in my local project without considering it as a configuration file.

This is not documented today in version 1.7.2:
<html>
<body>
<p style="font-family: &quot;Lucida Grande&quot;, Verdana, Geneva, Arial, sans-serif; font-size: 13px; line-height: 1.3; color: rgb(0, 0, 0); font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><b>Table: File Attributes</b></p><p style="font-family: &quot;Lucida Grande&quot;, Verdana, Geneva, Arial, sans-serif; font-size: 13px; line-height: 1.3; color: rgb(0, 0, 0); font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">The file attribute defines the special handling in the project when being used as configuration, template, or interface file. The table lists the values available as a file attribute.</p>

attr= | Description
-- | --
config | The file is a configuration file of the component. It is expected that only configuration options are modified. The file is managed as part of the component, as a project-specific file typically copied into the component section of the project.s
template | The file is used as a source code template file. It is expected to be edited and extended by the software developer. The file can be copied into a user section of the project.

</body>
</html>

jkrech, 2021-12-07:
We should review this feature. It is supported in MDK but the concept needs to be reviewed. 
Just copying files into the project directory without tracking e.g. versions seems premature.
Maybe it is just the lack of accessing files (e.g. scripts, utilities) in the pack directory without the actual need for copying it.

fred-r, 2021-12-07:
> Maybe it is just the lack of accessing files (e.g. scripts, utilities) in the pack directory without the actual need for copying it.

Fully agree.

ReinhardKeil, 2021-12-14:
I suggest we review the Access Sequences here https://github.com/brondani/devtools/blob/mvp-prototype/tools/projmgr/docs/Overview.md#access-sequences

It may overcome the shortcomings that we tried to address using attr="copy".  I suggest that attr="copy" (which is undocumented) should get documented as "deprecated".  Not sure if this is still in use; if not I recommend to remove it even from source code.

ReinhardKeil, 2022-01-12:
Since the attr="copy" was never published, I suggest that the PackChk utility flags it as warning deprecated.  Not sure if it is somewhere in use, but IMHO we should not go this way.   

Instead we should define a clean method to import Layers from Packs.  This is ToDo.

jkrech, 2022-01-25:
I have checked and it is today used in the Keil.MDK-Middleware pack to overcome the limitation that script/executable could not be referenced from a pack (Net_Security.bat, GUIBuilder.exe)

jkrech, 2024-03-12:
packchk issue captured here: https://github.com/Open-CMSIS-Pack/devtools/issues/1371

closed 2024-03-12 by jkrech