Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #61: CMSIS-Pack spec : more hyperlinks ?

fred-r, 2021-12-07:
Being picky as usual...but would it be possible to systematically use hyperlinks when describing the types ?

<html>
<body>

Cclass | Defines the component class to which the component belongs. This is a mandatory part of the component ID. Predefined values can be used as listed in the table Component Classes.Must not be specified for a component within a bundle. | CclassType | required
-- | -- | -- | --
Cgroup | Defines the component group to whoch the component belongs. Is a mandatory part of the component ID. Predefined values can be used as listed in the table Component Groups. | CgroupType | required
Csub | Defines the component subgroup. Is an optional part of the component ID. The type is described in Component Subgroups. | CsubType | optional
Cvariant | Defines a variant of a component. Is an optional part of the component ID. The variant specifier is a brief string (for example: release, debug). | xs:string [3..32] | optional

</body>
</html>

I would like to have hyperlink to find the definition of CclassType, CgroupType, Csubtype.

Thanks & Regards,
Fred

jkrech, 2021-12-07:
Good point. The nice thing is that you have full access to the doxygen source to help us with such tasks, as they do not require expert knowledge. I am looking forward to a PR.

fred-r, 2021-12-08:
Maybe it would be worth adding a contributing.md file explaining the tools we need and the way to go to contribute ?

For instance:
frq09468@LMECWL0888 MINGW64 /c/Open-CMSIS-Pack/doxygen (hyperlinks)
$ ./check_links.sh
./check_links.sh: line 3: linkchecker: command not found
grep: linkchecker-out.csv: No such file or directory

frq09468@LMECWL0888 MINGW64 /c/Open-CMSIS-Pack/doxygen (hyperlinks)
$ ./gen_doc.sh
which: no doxygen

So I guess I need to install:
LinkChecker-9.3.exe
doxygen-1.9.2.-setup.exe ?

fred-r, 2021-12-08:
P.R. : https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack/pull/62

fred-r, 2021-12-10:
> Good point. The nice thing is that you have full access to the doxygen source to help us with such tasks, as they do not require expert knowledge. I am looking forward to a PR.

I am looking forward to a merge :-)

fred-r, 2022-01-25:
Merged so can be closed.

closed 2022-01-25 by fred-r