Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #65: "RTE" directory configurable

ReinhardKeil, 2021-12-14:
Consider to allow renaming of the "RTE" directory.   We should list the changes that are required to enable this feature as it impacts to the version control of the "config" files stored in this directory.

It relates to this pull request https://github.com/Open-CMSIS-Pack/devtools/pull/65

jkrech, 2022-02-01:
Assumptions:
- "RTE" is default name for the directory
- if the directory name is different, it is required to store this information in the project/layer
- any project conversion tool requires modification for the import

Compatibility of Change:
- any previous tool version will not be able know where the config files are stored and the expected 'RTE' directory does not exist
- the project deems unusable unless the named RTE directory is manually renamed to 'RTE'

ReinhardKeil, 2023-01-09:
I believe this is implemented

closed 2023-01-09 by ReinhardKeil