Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #68: Broken link in https://github.com/Open-CMSIS-Pack

fred-r, 2021-12-22:
https://open-cmsis-pack.github.io/open-cmsis-pack-spec/   seems to be broken
[image]

fred-r, 2021-12-22:
This one is working:
https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/index.html

But this is not the top link:
[image]

ReinhardKeil, 2021-12-22:
Works for me, but seems to be case sensitive. Using this link:
https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec

[broken_link]<https://user-images.githubusercontent.com/8255133/147119353-a002f097-8262-48c5-a697-dec15262bea6.PNG>

IMPORTANT NOTICE: The contents of this email and any attachments are confidential and may also be privileged. If you are not the intended recipient, please notify the sender immediately and do not disclose the contents to any other person, use it for any purpose, or store or copy the information in any medium. Thank you.

fred-r, 2022-01-25:
The link is still broken if you click it from here:
https://github.com/Open-CMSIS-Pack

It is still referenced as:
https://open-cmsis-pack.github.io/open-cmsis-pack-spec/

instead of:
https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/

fred-r, 2022-01-25:
Ok, now.

closed 2022-01-25 by fred-r