Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #69: Use a cross-platform language for the tools

ilg-ul, 2022-01-12:
This is a feedback to https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack-Spec/issues/43, but might also apply to other tools, so I entered it as a separate issue:

The decision to use C++ for the tools is unfortunate, since you'll have to distribute binaries for all platforms (Windows, Linux Intel, Linux Arm, Linux Aarch64, macOS Intel, macOS Apple Silicon, etc).

A more convenient cross-platform solution is to use Node.js CLI applications, which run directly on all modern platforms.

VGRSTM, 2022-01-14:
Would add my vote to, especially if no risk to face performance issue looking at foreseen processing

chaws, 2022-01-14:
What about Golang?

Although Node.js runs on all platforms, it's like a JVM for JavaScript. Golang outputs compiled binaries with no other dependency to run them.

Here's a tool I've been working on for Open-CMSIS-Pack project: https://github.com/Open-CMSIS-Pack/cpackget

There's probably much more available packages in JavaScript than Go, but for what CLI tools need I think it's OK.

ilg-ul, 2022-01-14:
> Golang outputs compiled binaries with no other dependency to run them

I did not check Go recently, but when I evaluated it for the xPack project, it required to generate separate binaries, one for each platform; thus for xPack I went with Node.js, which uses a single package that runs on all platforms.

Can the Go compiler generate the entire set of binaries for all platforms on the same build machine (for example in a GitHub Actions job?)? What if someone will want to use the tools on a less common platform? 

node & npm are available virtually everywhere, and this greatly simplifies the cross-platform maintenance.

chaws, 2022-01-17:
> Can the Go compiler generate the entire set of binaries for all platforms on the same build machine (for example in a GitHub Actions job?)? What if someone will want to use the tools on a less common platform?

Yes! Go has its own compiler and it's really easy to cross-compile binaries: `GOOS=$OS GOARCH=$ARCH; go build`, the list of supported OS'es and Archs is defined in [Golang's source code](https://github.com/golang/go/blob/master/src/go/build/syslist.go), it's quite extensive. In cpackget's repository I set up a [Go-Releaser github action](https://github.com/goreleaser/goreleaser-action). It reads my custom [config file](https://github.com/Open-CMSIS-Pack/cpackget/blob/main/.goreleaser.yml) that specifies which OS/Archs to build against to and generates artifacts from the pipeline straight to the [releases page of cpackget](https://github.com/Open-CMSIS-Pack/cpackget/releases/tag/v0.2.0).

> node & npm are available virtually everywhere, and this greatly simplifies the cross-platform maintenance.

Can't argue with that!

It has been long since I last touch Node.js. I remember it was a nightmare to keep up the `package-lock.json` file.

ilg-ul, 2022-01-17:
> ... it was a nightmare to keep up the package-lock.json file.

What do you mean by 'keep up'?

I never had to bother with it, it is maintained automatically by npm, all I had to do was to store it in the repository. And I did quite a lot of releases, for example <https://www.npmjs.com/package/xpm?activeTab=versions>.

chaws, 2022-01-17:
> What do you mean by 'keep up'?

Iirc it kept changing too often, and other team mates often send patches with changes in this file as well. Maybe we didn't use it correctly.

madchutney, 2022-01-31:
I've been using Go, running in GitHub Actions and generating binaries for Windows, macOS and Linux for Intel and Arm architectures using GoReleaser, in a similar way to cpackget. 

We had experience with JS, Python and C++ and by far the easiest deployment for a CLI was with Go. The downside we've found with using any language that requires an ecosystem is that if the consumer isn't in that ecosystem, then they are reluctant to use the tools.

This becomes especially noticeable if the tools are going to be used in the cloud, someone running a Python based web-service doesn't want to install Node alongside Python, likewise for someone running a JS app doesn't want to install Python.

Something we do to support cloud consumers, who want to use the tools directly, is to provide a [gRPC](https://grpc.io/) interface and generate client libraries from the [protobuf](https://developers.google.com/protocol-buffers) definitions. The CLI can be used to start a gRPC server and then the clients can call the tool via the client library via gRPC, without the need to create subprocesses or parse stdout. This isn't to say that you need to write in Go to do this, but it does provide a nice interface for JS, Python etc.

ReinhardKeil, 2022-03-28:
For now (as we derived the tools from existing implementations), it is decided to use C++ and Go to implement the tools.  Future implementations may use different languages.

closed 2022-05-03 by ReinhardKeil

ilg-ul, 2022-05-03:
Just be sure you provide binaries for all platforms, including macOS and Raspberry Pi OS.