Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #73: Device, pack, component and board vendors are defined in multiple places using different data types

willlordarm, 2022-01-26:
There are (at least) five places where a pack creator can specify a vendor name within a pdsc file:

1. As a *Dvendor* attribute of the [/package element](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_package_pg.html#element_package)
2. As a *vendor* child element of the [/package element](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_package_pg.html#element_package)
3. As a *Dvendor* attribute of a [device /family element](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_family_pg.html#element_family) 
4. As a *vendor* attribute of the [/boards element](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_boards_pg.html#element_board)
5. As a *Cvendor* attribute of the [component /bundle element](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_components_pg.html#element_bundle)

Of these vendor definitions only Dvendor provides a consistent identifier for a unique vendor by setting the type to `DeviceVendorEnum` ([docs](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_family_pg.html#DeviceVendorEnum)). This enumerated list is maintained as part of the schema: 

> The table lists predefined values representing device vendors. The list is extended from time to time (on request by new vendors). Contact  [cmsis@arm.com](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_family_pg.html#)  to ask for an extension. These values can be used in the elements:
>
> [/package/boards/board/mountedDevice](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_boards_pg.html#element_board_mountedDevice)
> 
> [/package/conditions/condition/accept](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_conditions_pg.html#element_accept) 
> 
> [/package/conditions/condition/require](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_conditions_pg.html#element_require) 
>
> [/package/conditions/condition/deny](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_conditions_pg.html#element_deny) 
>
>  [/package/examples/example/board](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_examples_pg.html#element_example_board)
> 
>  [/package/boards/board/compatibleDevice](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_boards_pg.html#element_board_compatibleDevice) 
>
>  [/package/generators/generator/select](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_generators_pg.html#element_gen_select) 

All the other vendor definitions (board `vendor`, `Cvendor`, pack `vendor` use the type `xs:string` meaning that the pack creator can include an arbitrary string definition of the vendor name in any of these places. 

These vendor names are used to cross-reference elements between and within packs. The `<examples>` element can link a vendor supplied software example with a given board using the board `vendor` and board `name` or a `Cvendor` can be specified as part of a reference to a software component.  

## Why does this matter?

1. The same concept (vendor) is modelled in multiple ways: as a vendor of boards, packs, devices and software. This is conceptually confusing but also fails to reflect back the reality that many vendors will provide _all_ of these things.
2. Data (in the form of vendor names and ids) are managed as part of the schema. Data management should not be the role of a schema.
3. It is not possible to reliably link components, boards and devices provided by the same vendor to that single vendor entity. This matters because the linking (where it takes place) is controlled by the client/consuming application rather than being consistently represented to any consumer of CMSIS ecosystem services. For example, I may choose to map ‘LordBoards’ and ‘Lord Boards’ to a single vendor identifier in my client but another client may not. This is not a hypothetical, the situation exists today. 
4. There is no protection for a pack creator who includes a typographical error in the vendor string, meaning that a software vendor who wants to associate their example with board ‘WL64’ from vendor ‘LordBoards’ but accidentally wrote ‘Lord Boards’ would not successfully create this association. This is a problem for any system (e.g. keil.arm.com) that wants to make available all examples associated with a given board regardless of where the example was specified.
5. The manual nature of adding Device Vendors to the enumerated list is a barrier to scale and gives Arm effective control over which vendors can submit device definitions. This is almost certainly by design but may need to be reconsidered in Open-CMSIS-Pack.
6. If a vendor name changes (possibly through acquisition) there is no simple way to change the way this vendor is represented in consuming applications. Changing my vendor name from ‘Lord Boards’ to ‘Wills Boards’ requires a change to every pack that references the previous name or use of the `<dominate>` element (which works to a degree but leaves vendor names unchanged in the packs and, therefore, on disk). It is also likely to necessitate changes at an application level. Separate vendor name mappings currently exist in RTE, the MDK Eclipse plugin and on keil.com, all of which might need to be updated in such a scenario. 

## Proposed solution
A centrally administered web service that:

* allows vendors to create an account and generate a vendor id (assuming some degree of vetting/control over *who* can create these)
* maintains the list of generated IDs along with a minimal set of data about the vendor (e.g. name, website, logo)
* allows vendors to update their vendor data (but not the id)
* makes the list of vendors and their ids available for use by pack creators
	* For example a pack validation or submission flow could check whether a specified vendor id referenced in a pack exists and throw an error if it does not. 

## Open questions
* How might we retroactively apply IDs to existing packs? 
* How would we keep user management simple? It’s possible to imagine a scenario in which vendors need multiple user accounts associated with the vendor account.
* How would we ensure that vetting of new applicants (probably required to avoid spamming) is fair and rapid?
* Arguably devices and boards should /also/ have unique identifiers. Should we consider extending the discussion to consider this?
* How would ids need to exposed so that they are consumable by a range of clients (including offline?)
* Who build this? Such a system could potentially be built by Arm (who have the skills and capacity to deliver a web-based system such as this) and contributed in to the Open-CMSIS-Pack project. 
* Who maintains this? Linaro would be the obvious choice as a neutral party.
* How would we ideally handle the situation where a vendor is acquired by another? Is it enough to simply overwrite the old name with the new one or would we need a way of maintaining these relationships (e.g. 'formerly know as', 'is part of') within a vendor management system?

ReinhardKeil, 2023-01-09:
@willlordarm  how important is that issue? And what should Open-CMSIS-Pack should do here?

willlordarm, 2023-01-13:
To take real world example the creation of this vendor page on keil.arm.com - https://www.keil.arm.com/vendors/st/ - involves a service that maps multiple vendor definitions to a single vendor. This is what allows 'st' as a device vendor and 'st' as a board vendor to be treated as a single entity. When the page is extended to include packs then 'st' as a pack vendor will also need to be mapped. The mapping service also handles misspellings and data inconsistencies ('ST-Microelectronics' = 'STMicroelectronics' = 'St-Microelectronics') but only through manual intervention.

Since the service now exists the mapping is no longer a critical issue. As long as everyone is happy that the only way to get a consistent list of vendors and the contributions they make to the ecosystem is via this API then I don't think we have an issue for Open-CMSIS-Pack group.

I think the proposed solution above is still sensible and wouldn't require any schema changes. It's really just an automation of the existing vendor creation and pack submission process. Assuming everyone is happy with that then this is work that Arm can continue with, with any practical changes to submission processes reported back to this group.

ReinhardKeil, 2024-01-30:
@willlordarm I believe you did solve this problem using the web infrastructure.  If there is anything open, please be specify on what we should address.  Unfortunately we cannot change history or deployed packs.

closed 2024-01-30 by ReinhardKeil