Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #78: Extend spec to support templating languages and use it with generators

fred-r, 2022-02-01:
The proposal is to say that:
- a component may have templating language artefacts
- a generator may serve several client components, maybe through software generation via templating engines

We propose extending the specification to allow it, without specifying any mechanism in particular.

Nevertheless, the first application we may foresee is CMSIS-Zone.

fred-r, 2022-03-04:
Hi Reinhard and all,

regarding the proposal shared during the OpenCMSIS meeting : is it possible to add a 5th file category ?

  - `genParms` template parameters
  - `genHeader` header templates for the generator
  - `genSource` source templates for the generator
  - `genHelper` additional files for the generator

genPartial : partial templates for the generator

And maybe a 6th category, more for layers:

- genSettings : template values selected by the end-user

fred-r, 2022-03-04:
Another point also in the proposal:

```yml
cgen:
  device: STM32F407IGHx
  board: NucleoF407
  solution: C:/tmp/MySolution/MySolution.csolution.yml 
  project: C:/tmp//MySolution/Blinky/Blinky.cproject.yml 
  context: .Debug+Nucelo    # build-type and target-type of the current generated context
  destination: C:/tmp/MySolution/Blinky/RTE+Nucelo/

  packs:                     # packs that are used for the project
    - pack: Keil::STM32F4xx_DFP@2.16.1
      path: C:/CMSIS-PACKS/Keil/STM32F4xx_DFP/2.16.1/
    - pack: Keil::STM32F4xx_BSP@2.1.1
      path: C:/CMSIS-PACKS/Keil/STM32F4xx_BSP/2.1.1/

  components:               # components that have a genid and are specified in *.cproject.yml
    - component: Device:STM32Cube HAL:Common
    - component: Board:LED
```

For each component, may we have the list of files with the path ?
This way we do not need to evaluate conditions or whatever and we are sure about the files to be provided to the generator.

ReinhardKeil, 2022-03-04:
*For each component, may we have the list of files with the path ?*
Yep, we could do that if it helps to create the "generator" tool.  Let's discuss next week in the Open-CMSIS-Pack TM.

ReinhardKeil, 2022-03-04:
> Hi Reinhard and all,
>
> regarding the proposal shared during the OpenCMSIS meeting : is it possible to add a 5th file category ?
>
> * `genParms` template parameters
> * `genHeader` header templates for [...]

Yep, all this can be added. But we should perhaps better explain how it is used. Could you help with that?

fred-r, 2022-03-04:
Sure

- genParms provides the configuration parameters to the configuration tool triggered by the generator (if any)
- genHeader/Source are template files used by the generator to generate code 
- genHelper is a set of facilities that a template can use
- genPartials is an incomplete template file that can be "included" by a genSource/genHeader
- genSettings is a set of predefined values that can be used by the generator
 
Of course, I have handlebars in mind :-)
But I think it is generic.

Let's consider CMSIS-Zone with FTL:
https://github.com/fred-r/CMSIS-Zone/blob/main/Examples/STM32L5/Zone/ftl/SystemIsolation_Config.c.ftl

- genParms can be the rzone file
- genSettings can be the fzone file
- genHelper can collect the FTL functions like <#function mpcRegs mpc>
- genPartials can be <#include "helper.ftlinc"/>
- genHeader/Source is partition_gen.h.ftl / SystemIsolation_Config.c.ftl

closed 2022-03-18 by ReinhardKeil