Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #80: How to deliver layers ?

fred-r, 2022-02-01:
Hi,

I think that with ProjectManager we have the possibility to leverage layers (https://github.com/MDK-Packs/CB_Lab4Layer) to deliver preconfigured elements.

How can we deliver such layers ?
i think they should not be coupled with a project but delivered as independent artefacts, just like a CMSIS component.

Will it be part of a CMSIS Pack ?
I do not find layer in the Open-CMSIS-Pack-Spec.

Do we need to add it here ?

ReinhardKeil, 2022-02-01:
Yep that is already proposed, see here https://github.com/Open-CMSIS-Pack/devtools/blob/main/tools/projmgr/docs/Manual/Overview.md#cmsis-pack-extensions

fred-r, 2022-02-01:
Ok, so next step is a PR on the schema ?

ReinhardKeil, 2022-04-05:
There is a proposal for this now here: https://github.com/ReinhardKeil/IoT_App/blob/main/ProposedStructure/Proposals.md#changes-to-the-pdsc-file

fred-r, 2022-04-06:
I think the tag <csolution> is misleading, because the artefacts we list here will not end-up in csolution.yml but cproject.yml.
Basically, we provide a list of "macro-components" pre-assembled, preconfigured or not, precompiled or not.
So why not calling it <macroblock> ?

------

"Board Layer from Pack to Project"
I am not sure we should restrict this part to boards.

To me, there are 2 aspects:
- either the macroblock is only a set of components preassembled ==> nothing to copy in the project, only the cproject.yml is updated
- or the macroblock is also providing some configuration ==> we need to copy these configuration elements

------

The csolution tool does not copy the layer when:

there is already a Board.clayer.txt in the target directory
a existing file in the RTE directory would be overwritten

==> To me, the layer is a way to deliver preconfigured components.
But, we should not prevent the end-user from reconfiguring these elements.
So, we cannot say that because a copy has been made initially it cannot be updated.
Maybe not by a blunt copy from the pdsc content, but by reconfiguration.

We also need to think about an update of the pack with new layers settings.
I guess we should consider it as configuration element in the RTE folder and apply the same strategy as today ?
But then, this would mean that each element is also copied with its version like in RTE today ? Dup with .txt ?

[image]

ReinhardKeil, 2024-01-30:
Layers are distributed in a pack as described here https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_clayers_pg.html

I believe we simply forgot to close this issue.  Hence I'm closing it now.

ReinhardKeil, 2024-01-30:
Thinking twice: we need to document a good example that shows usage of layers. We will do this together with the upcoming MDK middleware.

closed 2024-04-24 by jkrech