Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #82: What is the best way to restrict a Pack components to a specific device ?

LMESTM, 2022-02-04:
During some prototype use of a STM32 HAL Drivers pack , we ended up in a case where MDK-ARM selects HAL L5 components in place of HAL U5 components. 

The components of those L5 and U5 HAL Drivers packs are all supposed to be device specific using conditions based on below statements:
```
      <require Dvendor="STMicroelectronics:13" Dname="STM32U5*"/>
      <require Dvendor="STMicroelectronics:13" Dname="STM32L5*"/>
```

Any idea what could happen ?

jkrech, 2022-02-08:
When walking backwards through the conditions from the one that is specified by the component, you need to ensure for a device specific component that somewhere in the chain of conditions is a requirement for Dname (as shown above).
It is always a good exercise to create a project for a different device (that does not have the same component IDs), e.g. ARMCM3.  Now your components must not show, or something is wrong with your conditions.

You may also decide to create a specific condition for each component. In this case simply place the filter condition at the very top of your condition, so it becomes easier for you to check.

jkrech, 2022-02-08:
Are you happy to close this specific issue?

LMESTM, 2022-02-14:
@jkrech 
Yes you're right.
I would just add as according to specification Dname can also be specified for an entire pack as part of the /package definition as defined here: https://arm-software.github.io/CMSIS_5/Pack/html/pdsc_package_pg.html