Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #83: PDSC conditions: circular dependencies not detected by PackChk

LMESTM, 2022-02-04:
During packs prototyping (again) we've inserted unintended circular dependencies in our pdsc files.

PackChk does not seem to detect them or maybe this is not considered as an error ?

Nevertheless this does not seem to be a good practice and makes Conditions understanding quite hard to read and understand.
Any feedback about it ?

LMESTM, 2022-02-04:
A basic example of such a circular dependency

```
    <condition id="STM32L5 HAL Common">
      <description>STMicroelectronics STM32L5 Device and HAL Common</description>
      <require condition="STM32L5 device"/>
      <require Cclass="CMSIS"  Cgroup="CORE"/>
      <require Cclass="Device" Cgroup="STM32 HAL" Csub="Common"/>
      <require Cclass="Device" Cgroup="STM32 HAL" Csub="Cortex"/>
      <require Cclass="Device" Cgroup="STM32 HAL" Csub="RCC"/>
      <require Cclass="Device" Cgroup="STM32 HAL" Csub="PWR"/>
    </condition>

    <component Cclass="Device" Cgroup="STM32 HAL" Csub="Common"    Cversion="0.0.1" condition="STM32U5 HAL Common">
      <description>Common HAL driver</description>
      <files>
        <file category="include" name="Inc/"/>
        <file category="include" name="Inc/Legacy/"/>
        <file category="header"  name="Inc/stm32u5xx_hal.h"/>
        <file category="source"  name="Src/stm32u5xx_hal.c"/>
      </files>
    </component>
```

jkrech, 2022-02-04:
I do appreciate that a component having a condition with a dependency on itself is not very useful, but in this case the effect is like an ```if (true)``` because the dependency is only evaluated if the component is already selected.
I think this becomes more problematic in case of <deny>, as you will never be able to resolve the dependency then ```if (false)```. If you describe a choice using <accept> the user will never see the choice he should make because the dependency is immediately resolved.
Do you think the specification should spell out that this is an error, or do you think that PackChk should flag this as warning or even error? In the latter case, please raise an issue against PackChk in the devtools repository.

fred-r, 2022-02-04:
Well, on one hand I agree that a component requiring itself sees its dependency being satisfied "automatically" as the component is selected, this is true.

But, with the pdsc we describe "build" dependencies so at the end of the day inclusion rules I think.
So circular dependencies should not be allowed.
I think PackChk should detect it with a warning.

What is unclear to me is what the tool (like MDK-ARM) is going to do when the component is selected:
- select the component A
- MDK-ARM checks the condition and sees component A as a dependency : A is in the project so this is satisfied
- But MDK-ARM is not going to follow the dependencies of A again ?

Something like:
-  I check A ==> it needs A and B
    -I check what B needs : nothing. STOP.
   - I check what A needs => A needs A and B
            - B needed => I check what B needs : nothing. STOP.
            - A needed => I check what A needs => A needs A and B
...and so on...

Obviously MDK-ARM does not enter this kind of cyclic check.
So I guess you do something like:
- I check A ==> it needs A and B
- A needed => A is present : OK, do not check what A needs
- B needed => B is present : OK, do not check what B needs
but I do not follow the dependencies now.
So when moving on:
- I do not check A again.
- I check B : OK

Is it how you avoid an infinite loop of checks ?

edriouk, 2022-02-04:
PackChk detects condition recursion and issues corresponding message: 
`{ "M384", { MsgLevel::LEVEL_ERROR,    CRLF_B, "Condition '%NAME%': Direct or indirect recursion detected. Skipping condition while searching for '%NAME2%'." } },`
However the check is performed only on used conditions, so it starts with conditions referenced by components and files.

edriouk, 2022-02-04:
Recursion detection is performed inside RTE Model, therefore MDK-ARM is able to handle that situation as well.
In particular the following code is performed:
`
RteItem::ConditionResult RteCondition::Evaluate(RteConditionContext* context)
{
  if (IsEvaluating(context)) {
    return R_ERROR; // recursion error
  }
  SetEvaluating(context, true);
  RteItem::ConditionResult result = EvaluateCondition(context);
  SetEvaluating(context, false);
  return result;
}
` 
Each RteCondition keeps track if it is being evaluated or not  => can detect recursion

jkrech, 2022-03-08:
I did not see any further questions in this thread. Therefore I am closing this.

closed 2022-03-08 by jkrech

LMESTM, 2022-03-08:
indeed thanks for your support !