Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #85: Handling of packs (Filter, Repositories, Local packs, etc.)

ReinhardKeil, 2022-02-08:
Introduce a section `packs:` in `*.csolution.yml` and `*.cdefault.yml` to:
  - reduce the scope of packs used in a project
  - specify pack versions
  - specify local paths to packs

```yml
packs:                                  # start section that specifics software packs
  - filter: AWS::                       # use packs from AWS
  - filter: NXP::*K32L*                 # use packs from NXP that relate to K32L series (would match K32L3A60_DFP + FRDM-K32L3A6_BSP)
  - filter: ARM::                       # use packs from Arm

  - pack: Keil::Arm_Compiler            # add always Keil::Arm_Compiler pack (bypasses filter)
  - pack: Keil::MDK-Middleware@7.13.0   # add Keil::MDK-Middleware pack at version 7.13.0
  - pack: NXP::K32L3A60_DFP             # add pack
    path: .\local\NXP\K32L3A60_DFP      # path to the pack (local copy, repo, etc.)
```

LMESTM, 2022-02-15:
Thanks for the proposal 
One question:  which entity is expected to copy the PACK to the specified path ? 
Basically, if a csolution (and related project files) is shared from a user to another, can he expect that the file system project will be re-created in the same way ?

ReinhardKeil, 2022-02-15:
_One question: which entity is expected to copy the PACK to the specified path ?_
In the current specification the `path:` has no restrictions; if it is relative to the workspace, I assume it would be part of a repo; if it is an absolute path to some random directory, I would assume that the organization that uses it has some guidelines when such projects are shared.

slhultgren, 2022-03-04:
Sounds good to have this flexibility of specifying a path to where a pack is installed.

Would be good to have this on the `.cproject.yml` level instead/as well.

ReinhardKeil, 2022-03-04:
Agreed.  Let's see if it is possible/easy to implement.  Might be challenging as you may change the packs based on what project is processed.

ReinhardKeil, 2024-01-30:
closing as I believe this is outdated.