Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #86: Active target-type and build-type ?

fred-r, 2022-02-15:
In the current csolution I do not see a way to indicate an active target-type and build-type.

It could have 2 usages:
- refocus on the last active setup
- if we cannot understand multi-targets then we know which one to select

fred-r, 2022-02-15:
Unclear to me the point about mutliple people using the same csolution.yml at the same time.
Isn't csolution.yml the entry point of your desktop tools (like IDE) ?
So it will be in configuration management to be shared but you will work on your local copy until you publish it?

Or do you envision collaborative work via cloud accounts ?

jkrech, 2022-02-16:
@fred-r it does not need to be concurrently. The problem already manifests when a user commits changes to the csolution.yml file in a repository. At this point in time the "active" target-type + build-type will be stored as well.
If this type of user specific information is stored in a separate file, it could be added to an "ignore file".

ReinhardKeil, 2022-02-16:
The assumption is that csolution.yml (and cproject.yml, clayer.yml) are part of a repository and version managed.  Multiple people and test automation systems use this repository (as in this picture https://github.com/ARM-software/VHT-GetStarted).  If the "active" target-type + build-type would be stored in csolution.yml, everyone would work with the same configuration.  However there might be test engineers that work with a test/debug build, system integrators that work with production/release build, etc. Therefore each user must have the option to use a different configuration and csolution.yml should only be modified if you add/remove files/components or change tool settings.

jkrech, 2022-02-21:
Maybe the default build and target type is something to consider adding to the defaults?
```[defaults].csettings.yml```

ReinhardKeil, 2022-02-21:
`cdefault.yml` should apply to many `csolution.yml` files. It cannot specify the defaults for a specific `csolution.yml` file as every `csolution.yml` may need different settings.

I suggest to be a bit more pragmatic for now. My suggestion:

The first `target-type` and first `build-type` would be the default for tools that do not support target/build-type selection.

jkrech, 2022-03-08:
Concluding to follow the proposal from Reinhard to use the first target-type combined with the first build-type specified by default for now.