Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #87: Resource handling

ReinhardKeil, 2022-02-15:
We started to work on resource handling.  A very first idea is here: https://github.com/Open-CMSIS-Pack/devtools/blob/main/tools/projmgr/docs/Manual/Overview.md#cmsis-zone-integration

fred-r, 2022-02-15:
Great, rather than doing something specific for CMSIS-Zone can we say that we open the door for a solution (rather than a project because to me we are talking here about system configuration) to have several system configuration files ?

- resources partitioning => CMSIS-Zone
- system clock => vendor dependent : no format defined
- pinout => vendor dependent : no format defined

But at least we have a placeholder to attach these settings to a solution.

ReinhardKeil, 2022-02-22:
The documentation is now reworked and the resource handling is here https://github.com/Open-CMSIS-Pack/devtools/blob/main/tools/projmgr/docs/Manual/YML-Format.md#resource-management

It is not completely spelled out yet, but closer to what we need finally.  When time permits we can discuss it today.

fred-r, 2022-02-22:
* rzone file format: no strong opinion about YML or JSON but we have some missing items to suggest
* Handlebars : this is a "YES" for me
* Feeding Handlebars with JSON is easier in my opinion because you "see" the data object processed by the templating engine directly

ReinhardKeil, 2024-01-30:
Even when the topic `RZone` has not been addressed yet, I'm closing this issue.

I believe we need to review resource management later this year in context with enhancements to linker script handling and the MPU support that is now available with CMSIS version 6 and the RTX implementation.

closed 2024-01-30 by ReinhardKeil