Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #88: Agree on terms

ReinhardKeil, 2022-02-15:
I believe we should align on terms that we are using.

In CMSIS-Zone we have today:
 
- Project Zone that defines effectively the resources (memory/peripherals) used by a cproject.yml

- Execution Zone that further spilts access in a cproject.yml - these are dynamic MPU protection settings that change during run-time in practice. For initial implementation in csolution/cproject.yml I suggest we ignore that in the moment. 

What’s missing is life span of resource allocation in cproject.yml. For example a memory might be only used during Boot phase. I’m leaning towards the term ‘phase’ here. 

This relates to https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack-Spec/pull/63

fred-r, 2022-02-15:
We can talk about a phase if you prefer.
But a phase should be a collection of 1..N projects.
So:
- projects operating in the same phase have to partition the resources between them.
- projects operating in different phases can reuse each other's resources unless there are constraints

ReinhardKeil, 2022-03-02:
Somewhat a duplicate with #33.  Can we consolidate this here and close #33

ReinhardKeil, 2022-12-09:
@fred-r help from your side would be appreciated.

ReinhardKeil, 2024-01-30:
somewhat outdated - hence closing it

closed 2024-01-30 by ReinhardKeil