Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #100: Component should has attribute "invisible"

tcsunhao, 2022-03-02:
Component should has attribute "invisible". Invisible component can be intentionally ignored by IDE/tools.

There are some cases that developers especially middleware owners want to intentionally hide certain intermediate component. For example, USB stack can be implemented into usb.controller required by usb.stack.common required by usb.stack.device/usb.stack.host/usb.stack.otg. Owner may only want to expose usb.stack.device/usb.stack.host/usb.stack.otg such end point components for users to select. There is not too much meaning to show intermediate components like usb.controller and usb.stack.common.

ReinhardKeil, 2022-03-02:
Thanks for the suggestion, we will look into it.

ReinhardKeil, 2022-03-08:
I suggest to name this attribute `inherent`.  

Behavior: 
  - components with `inherent` attribute are not selectable by the user, but selected when a condition requires this component. 
  - components with `inherent` attribute are de-selected when a **no** condition requires this component. 
  - IDE's may choose to hide such components in the RTE selection (my personal preference is however to show them).

This attribute replaces also the `genOwned` attribute proposed under *Changes to the *.PDSC file* in
https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack-Spec/issues/20#issuecomment-1055425549

jkrech, 2022-09-26:
#112 view attribute has been added to the specification of components.

closed 2022-09-26 by jkrech