Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #98: Component should has its own type

tcsunhao, 2022-03-02:
Every component should have its own type. Common types can be driver, middleware, OS, config.

IDE/tools can has specific operations on certain component type.

tcsunhao, 2022-03-03:
I think the main purpose of Cclass, Cgroup is to classify component into different categories. It doesn't perfectly specify the component attribute(although it can to some degree).  
Take above project template component for example, if there is a way to tell that this component is "config" which means all files under this component is "config", the we don't need to specify for each file then.

ReinhardKeil, 2022-03-03:
Sorry, I have not understood it the first time. Let me put it in my own words. 

The proposal is to have the file attribute ('attr') at component level, because all files of this component should be copied to the project for configuration. It saves that the 'attr' is specified multiple times for each file of the component.

Does it have other benefits too?

tcsunhao, 2022-03-03:
I don't think of other. 
NXP IDE/tool will process component for certain "type". The only one is "config" component(NXP call it template component), we anyway copy the sources under it into workspace.

ReinhardKeil, 2022-04-05:
Do we really need this extension?  In general it is possible to duplicate the "attr" field.  I would not rule out that we extend the format in future, but it need to be deployed to many different implementations and keeping them consistent is tricky.

How important is that for you?

tcsunhao, 2022-04-05:
It is not a must have attr. We are ok living without it.

ReinhardKeil, 2022-04-05:
Suggest to close it then.

closed 2022-04-05 by tcsunhao