Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #103: Numbers/star meaning in the XSD not explained

slhultgren, 2022-03-08:
In the XSD it is allowed to specify "1" or "0" for DfpuEnum
https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack-Spec/blob/024ba1153437f0295799b1efd4349f789a3d1a1b/schema/PACK.xsd#L993

But in the human-readable section it is not explained what this means:
https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack-Spec/blob/024ba1153437f0295799b1efd4349f789a3d1a1b/doxygen/src/devices_schema.txt#L2485

For DsecureEnum it also allows "2" which is harder to guess what it means
https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack-Spec/blob/024ba1153437f0295799b1efd4349f789a3d1a1b/schema/PACK.xsd#L1036

vs 
https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack-Spec/blob/024ba1153437f0295799b1efd4349f789a3d1a1b/doxygen/src/devices_schema.txt#L2571

Or on the human-readable rendering as well:
https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_family_pg.html#DfpuEnum

ReinhardKeil, 2022-03-08:
I would actually qualify this as bug in schema file.  Are there reasons, why numbers are used to enumerate FPU and TrustZone.  Is this legacy?  I propose to remove it - otherwise all tools would need to cope with current specification and the legacy (perhaps deprecated features).

jkrech, 2023-01-24:
Yes,  you can consider it legacy. There is many pack descriptions that use '0' and '1' instead of the predefined string values. We should label them deprecated in the PACK.xsd.
The documentation has the '0', '1' and '*' values already removed.

jkrech, 2024-01-30:
The numbers and start got removed from the documentation but remain in the PACK.xsd in order not to break existing descriptions. I suggest to close this and not document numbers and star.

closed 2025-02-10 by jkrech