Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #104: CMSIS-Pack generator component concept - Proposal

ReinhardKeil, 2022-03-08:
# Proposal for Revised Generator Support

## Workflow assumptions

The composition of a solution of a solution should have the following steps:

- Create `cproject.yml` files and the `csolution.yml` container that refers the projects.
- Select `device:` or `board:` (optionally by using `target-types:`)
- Add `components:` or `layers:` to the `cproject.yml` file
- For components that have configuration, run the generator in configuration mode
   - change pinout, clock, resources, etc.
   - reflect configuration in *.gpdsc file (and related settings files)

## Steps for component selection and configuration

The following explains the generator workflow of CSolution / CBuild for configuration of components:

1. User selects components in `*.cproject.yml` under `components:`
    - When these components require generation, user is notified to run a generator.
    - "CSolution Run GenID” is invoked for a list of components.
    - CSolution generates `*.cgenerator.yml` file that provides project context and a list of user-selected components.
  
2. Running the Generator (for Component Configuration, i.e. pin selection)
     - Generator reads `*.cgenerator.yml` file
     - User performs the configuration is done.
         - Interactive mode (where a settings file is generated)
         - Remote mode (where a settings file is an input) **IS THIS REALLY REQUIRED**
     - Generator creates a `*.gpdsc` file that informs the CSolution tool about
         - (a) the fact that a component is configured and has generated code, 
         - (b) additional components that are the result of some user configuration.

   Discussions: 
     - is a component list or a dependency list
     - Generator might be VS Code plugin or web based

3. User creates CBuild output with CSolution Convert command
      - Both `*.cproject.yml` and `*.gpdsc` are read by Csolution and create the complete list of selected components.
      - If `*.gpdsc` does not contain component information about a component that has `genId` and selected in `*.cproject.yml` the generator configuration is incomplete. This can happen when a component is added at a later step.
      - Likewise the Generator can detect with the `<component User="1">` attribute that a component is longer required. In this case the user is notified to run a generator.

## Enhance Usability

Add Run Generator buttons to Cclass descriptions.

[image]

## Workflow

1. For `*.cproject.yml` files that contain selected `<components>` with a `generator` or `genid` attribute the `csolution` manager checks if a file with the name `./<project>/RTE+<target>/genid.cgen.yml` exists.
    - When this file is missing, it is required to use the command `csolution run genid` to start the generator.
    - When this file exists, the `csolution` manager checks if the list of components with `genid` has changed. If this is the case it is required to use the command `csolution run genid` to reconfigure generated components.

2. The command `csolution run genid` creates the file `./<project>/RTE+<target>/genid.cgen.yml` and starts the generator.  The generator creates a *.GDPSC file along with other source files that are required [as specified](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_generators_pg.html#element_generators).

**Example Content of \*.cgen.yml (in this case STM32CubeMX.cgen.json)**

The `*.cgen.json` file is passed to generator as argument. 

**NOTE:** Shown is still a YML file, but the equivalent data would be formatted in *.JSON format

```yml
cgenerator:
  device: STM32F407IGHx
  board: NucleoF407
  solution: C:/tmp/MySolution/MySolution.csolution.yml 
  project: C:/tmp//MySolution/Blinky/Blinky.cproject.yml 
  context: .Debug+Nucelo    # build-type and target-type of the current generated context
  destination: C:/tmp/MySolution/Blinky/RTE+Nucelo/

  packs:                     # packs that are used for the project
[... 143 lines cut ...]
  </conditions>
  <components>
    <bundle Cbundle="STM32CubeMX" Cclass="Device" Cversion="1.4.0">
      <component generator="STM32CubeMX" Cvendor="Keil" Cgroup="STM32Cube Framework" Csub="STM32CubeMX" condition="STCubeMX">
        <description>Configuration via STM32CubeMX</description>
        <RTE_Components_h>
          #define RTE_DEVICE_FRAMEWORK_CUBE_MX
        </RTE_Components_h>
        <files>
          <file category="header" name="MX_Device.h"/>
          <file category="header" name="STCubeGenerated/Inc/stm32g4xx_hal_conf.h"/>
          <file category="source" name="STCubeGenerated/Src/stm32g4xx_hal_msp.c"/>
        </files>
      </component>
      <component  Cgroup="Startup" User="1">  <!-- user selected component in *.cproject.yml -->
        <description>System Startup for STMicroelectronics</description>
        <files>
          <file category="source" name="STCubeGenerated/MDK-ARM/startup_stm32g474xx.s" />
          <file category="source" name="STCubeGenerated/Src/system_stm32g4xx.c" />
        </files>
      </component>
    </bundle>
  </components>
</package>
```

mark-edgeworth, 2022-03-09:
Re the proposed `invisible` attribute. This is a UI concept and should not therefore be part of the spec (domain pollution). The UI (or any other client) should be able to decide whether to show or hide something based on the object type. 

I'm not really keen on the `inherent` term (it doesn't really mean anything to me!). How about `generated`, `derived`, `template` or even `virtual` perhaps? I'm not really familiar enough with the whole generator concept to see what problem it is trying to solve; it seems to be quite a complex thing...

ReinhardKeil, 2022-03-09:
An alternative term might be `indirect` as the component is *indirectly* selected when other components are *directly* selected by a software developer. 

I agree with you that the UI concept should not be part of the underlying specification and we leave this implementation defined.

edriouk, 2022-03-11:
In fact, RTE Model already assigns "generated" attribute to a component read from gpdsc. Why should we add another attribute (User) ?

ReinhardKeil, 2022-03-11:
"User=1" addresses the following scenario:

1. User adds a component that has a generator element
     - generator is started and provides the setup for that component

2. User removes the component
     - *.gpdsc still has a reference to that component, therefore it would be included
     - with "user=1" the tool can detect that it was removed by the user and can inform that it is no longer required

edriouk, 2022-03-11:
Both scenarios already work in RTE Model:
1. User adds the component that has a generator
 - that means the corresponding gpdsc file is required
 - if there is no gpdsc file yet associated with that generator, generator need s to be started
2. User removes the component 
- requirement to gpdsc file is removed from dependencies 

What should be different with the "user" attribute?

ReinhardKeil, 2022-03-11:
OK, if it works then there is no need to add it.    Not sure how it works today, but let's not debate it further.

ReinhardKeil, 2022-03-18:
The [generator proposal](https://github.com/Open-CMSIS-Pack/devtools/blob/main/tools/projmgr/docs/Manual/YML-Format.md#generator-proposal) is now incorporated.  Next week, I'm in vacation.

@fred-r it would be good to discuss the invocation of CubeMX on the various operating systems.  My initial proposed solution on Windows would not require that the EXE file name is part of the *.PDSC - but I'm not sure about other operating systems.

@edriouk thanks for all your feedback.  We will discuss it once I'm back and then integrate it.

edriouk, 2022-03-22:
The following PR has added registry read operation: https://github.com/Open-CMSIS-Pack/devtools/pull/227
The following is implemented:
- registry is only used on Windows (host="win"). on other platforms environment variable is used
- it will be possible to specify different keys for different hosts
The logic for registry key evaluation is: 
- first "HKEY_CURRENT_USER" then "HKEY_LOCAL_MACHINE" domain is evaluated unless required domain prefix is explicitly specified.
- the program first try to use the entire string as a key. If fails, the last segment is treated as value name

However there is a problem is in evaluating the returned value: 
- is it a path to executable specified in `<command>` element?
- or it include the executable name as  well?
- or it is something else?
That need to be clarified and standardized.

fred-r, 2022-03-23:
Hi,

on my side I think we should decorrelate the spec from the implementation method.
Probably the word "key" is misleading here.

What I would prefer:
- On OpenCMSIS-Pack-spec side, we define a way for a generator to "reference" its underlying tool (because I guess very often the generator will be a script calling an executable, or it could be a VSCOde extension invocation or whatever depending on the tool technology and OS)
- Then, it is up to the tooling to associate this "reference" to an actual tool (executable, VSCode extension, script, remote service whatever...)

When it comes to the "vanilla tooling" provided by OpenCMSIS, then I think we can very well handle it based on user inputs (just like the compiler path in cmsis-toolbox).

edriouk, 2022-03-24:
We need to move ```<key>``` element again
It should be an attribute of GeneratorCommandType to move it closer to the semantic command element and avoid confusion:
````
 <xs:complexType name="GeneratorCommandType">
    <xs:simpleContent>
      <xs:extension base="xs:string">
        <xs:attribute name="host" type="HostEnumType" use="optional" default="all" />
        <xs:attribute name="key" type="xs:string" use="optional"/> 
      </xs:extension>
    </xs:simpleContent>
  </xs:complexType>
````
This "key" attribute can represent on Windows ("host" attribute == "win" ): 
 1.  explicitly specified registry key in the current user domain, e.g.:
   ````<command host="win" key="HKEY_CURRENT_USER/Software/MyVendor/MyGen">MyGen.exe</command>````  
 2.  explicitly specified registry key in the local machine  domain, e.g.:
   ````<command host="win" key="HKEY_LOCAL_MACHINE/Software/MyVendor/MyGen">MyGen.exe</command>````
 3.  registry key that can be relative either to user or local machine depending on installtion: 
 ````<command host="win" key="Software/MyVendor/MyGen">MyGen.exe</command>````
 in this case current user domain is looked up first, then local machine
 4.  environment variable (for all platforms): 
 ````<command key="MYGEN_HOME">bin/MyGen.exe</command>````
 
The value of the registry key or environment variable can contain: 

- only path to the generator executable: ````C:\Program Files\MyVendor\bin\````
- or the entire absolute filename: ````C:\Program Files\MyVendor\bin\MyGen.exe````
in this case the name of executable should exactly match text specified in ```<command>``` element

fred-r, 2022-03-24:
Changes to the *.GPDSC file
To indicate that a component was generated due to a user selection in *.cproject.yml, the component element is extended with User attribute. When set to 1 it indicates that a component is included due to the selection in *.cproject.yml.

When a user removes this component in the *.cproject.yml, the CSolution could detect that a Run command should be executed.

==> this part is still a bit confusing for me. If the "initiator" component is removed (the one selected by the user and triggering the generator) then probably the components added as dependencies during the generation should be removed as well. I do not really see the role of the User attribute here.

edriouk, 2022-03-24:
@fred-r, as discussed above the 'user' attribute is redundant and won't be implemented. Current RTE code is already capable to to handle that situation.

fred-r, 2022-03-24:
About the "hidden" or "inherent" attribute : that's not the way I understood the request in the initial ticket.
To me the point with "visible" was to say that some components can be seen as "intermediate" components.

Some IDEs may decide to hide it, some IDEs may implement more advanced strategies.
But I think it may be worth having an attribute to say that a component is not absolutely mandatory for project representation to the end-user.
The criteria to define it would depend on the Pack provider.

A typical use-case to me would be this:
Application--requires-->Middleware--requires-->Wrapper--requires-->Driver

Then probably the Wrapper has no real added-value for the end-user (because maybe the wrapper is induced by the driver we select or there is only one wrapper available).
We may decide that it is not "mandatory" to show it (but mandatory to have it in cproject.yml).

The exact behavior would remain IDE dependent, it would only be an indication from the pack ddesigner that this component is not an essential element in a project.

fred-r, 2022-03-24:
> We need to move `<key>` element again It should be an attribute of GeneratorCommandType to move it closer to the semantic command element and avoid confusion:
>
> ```
>  <xs:complexType [...]

To me, I think we go too far.
The specification must remain OS-agnostic in my opinion, and we let the tool implementer decide the best implementation strategy.

So my position would be to say that "key" is a string used as an identifier by a "generator manager".
Then, this generator manager is in charge of deciding how to link the "key" to the actual way to trigger the generator:
- can be via a HKEY on Windows
- can be via an executable name combined with "whereis" on Linux
- can be a script on MacOS
- ...

I would not try to specify it in the OpenCMSIS-Pack-Spec.
I would just define the "key" format and mandate for the implementer to link it to the actual generator executable.

In the "vanilla tooling", it can be fully acceptable to me that the user is asked to provide the executable associated to the Key.
In Keil Studio, it can be a different strategy.
And so on..

edriouk, 2022-03-24:
Yes, we do because every generator requires a gpdsc file. Thus if no component refers to a generator, no corresponding  gpdsc is required  => it is not loaded

fred-r, 2022-03-24:
> Yes, we do because every generator requires a gpdsc file. Thus if no component refers to a generator, no corresponding gpdsc is required => it is not loaded

In fact my use-case is:

at t0, 3 components requires generatorX.
we generate and obtain a gpdsc with a set of files matching these 3 components.

at t1, the end-user removes one of the 3 components.
Only 2 components reference generatorX.

As the gpdsc is present, the tool does not know if the generator must be called.
But, as the number of components using the generator has chnaged, probably the gpdsc content is wrong.

So, as soon, as the list of components invoking generatorX is modified, we need to conclude that the generatorX must be called again.

tcsunhao, 2022-03-24:
@fred-r , in NXP cases, the "invisible" are used for components which 
1. not used by projects. Sometimes certain component is just used to store some docs or misc files of middleware, no need to show them.
2. intermediate and base component in the require chain like following, we will hide usb base, because it anyway will be selected, and no need to let customer to select from beginning to do multiple times selection. 
[image]

edriouk, 2022-03-24:
What is the reason to have the same generator for several components? Should such components require a sing one via conditions? 
Then, the generator needs to be run again for several reasons, such as:
- device change 
- device property (e.g. FPU usage change)
- components selection changes, in particular device-specific components  
- debug and trace setting changes
-  ...

In my opinion a client tool is responsible for triggering generator run, not the library

fred-r, 2022-03-24:
> What is the reason to have the same generator for several components? Should such components require a sing one via conditions? Then, the generator needs to be run again for several reasons, such [...]

This comes from this discussion:
https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack-Spec/issues/78

Several components can share the same generator.

edriouk, 2022-03-24:
As far as "key" value is concerned, its role is to launch the generator with out need to ship its "launcher" with a pack, beacue launcher applications and scripts can potentially be unsafe and not secure enough.

fred-r, 2022-03-28:
> CSolution generates *.cgenerator.yml file that provides project context and a list of user-selected components.

I think we must go down to the instances a component may contain.

fred-r, 2022-03-29:
Another point also is about the naming: to me .gpdsc is a bit misleading.

It is true that a .gpdsc refines a pdsc by adding files to a component and refining or adding conditions.

But, I think this can also refine the project by adding new files to an instance for example.
Maybe a generator may even add an instance under some circumstances.

So I wonder if .gpdsc is still the best name.

ReinhardKeil, 2022-03-29:
You are right with your observation about *.gdpsc.  We should review this.

fred-r, 2022-03-30:
> > @fred-r , in NXP cases, the "invisible" are used for components which
> > ```
> > 1. not used by projects. Sometimes certain component is just used to store some docs or misc files of middleware, no need to show them.
> > 
> > 2. intermediate and base component in the require chain like following, we will hide usb base, because it anyway will be selected, and no need to let customer to select from beginning to do multiple times selection.
> >    [image]
> > ```
> 
> I've got a question here: what if each of Usb-device, usb-host and Usb-base have two or more implementations (Cvariant or CsubGroup). The user needs to select the desired ones then and maybe later change the selection. How it would be possible with "hidden" components?

We can discuss it here:
https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack-Spec/issues/112

fred-r, 2022-05-06:
We would like to conclude on the support of this new generator concept, but at the moment the spec is not reflecting it.
The spec is still using parameters for a generator and the way gpdsc is described enables only one component to use a generator. The new file types are not present either.

Also, we did not conclude on the "generator key" concept.

Besides, I think we need to revisit the *.cgen.json:
- How to take instances of a component into account ?
https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack-Spec/issues/76
- Shouldn't we provide directly the required files in addition to the component/instance identifier?
- Need the file types to identify the templates:
https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack-Spec/issues/78

@iomint : I think this point is important for us in the coming weeks.

fred-r, 2022-05-17:
Regarding the cgenerator.yml, I am not sure we can go this way without putting too much intelligence in the generator.

A project may have 3 different generators integrated.
If each generator receives the same file containing a complete description of the project then each generator has to do some filtering to figure out why it has been called. This requires parsing the pdsc and maybe also the YML to some extent.

I'd prefer to keep a clear split of responsibilities:
- whatever is related to YML and PDSC is handled by the project manager (csolution)
- the generator is a "slave" that performs a processing based on an input file created by the project manager

This input file (cgenerator.json) can contain the entire description of the project if we think it can help.
But there should be a section listing the components and instances that the generator must explicitly process : the reason why it has been called.

Otherwise, each generator will parse the input file and the pdsc to filter what he needs to do: overhead of responsibility with project manager.

Now, the last point which remains and is a bit tricky is : should project manager provide the generator with a list of files ?
I see 3 possibilities:
A - provide only the components/instances ==> the generator must go back top the pdsc to find the files to be used for each component/isnatnce
B - provide the components/instances + all the files attached to them, even if not related to generation (doc...etc...) ==> the generator selects what it needs but it does not need to use a pdsc
C - provide a full filtering : based on the file categories, the project manager gives the component/instances + the listof files related to the generator processing

Option A can be a problem because the generator requires the pdsc...

Option C can be too extreme because from a generator to another, the input file categories might differ : or the generator needs to list the fime categories it is interested in in its definition....  

Option B can be a good trade-off : the generator keeps the responsibility of selecting the appropriate files but it does not need to parse the pdsc, the list of files is proivded in the input file.

@mdortel-stm
@nabilsafiSTM

fred-r, 2022-08-11:
Follow-up:

"The *.cgen.json file is passed to generator as argument."

==> I'd like to go down to file level granularity (path and filename) grouped by component.

We can use the lock file syntax proposed by @slhultgren

fred-r, 2022-08-30:
Dear all,

on STM side, we are currently progressing on our flow and usage of the generators.
From a user flow perspective, we are aligned with the proposal.

We confirm that we envision projects with:
- several components invoking the same generator
- 1 component invoking 1 generator

So:

__Components description__
1. The spec must allow us to associate the very same generator to several components.
2. The spec must allow us to provide a global unique identifier per generator: we do not need this to be a direct way to invoke an executable. It can be a unique identifier and then it is up to the project manager (csolution tool) to invoke it properly.

Then:

__Generator Invocation__
The project manager (csolution) must be able to invoke each __unique__ generator: no need to call N times the same generator if several components using it are in the project

__Generator Definition__
Now, regarding the way to define this generator: we can keep the existing arguments for backward compatibility reasons.
But, we also need one more argument to provide the generator input file (cgen.yml/json).

Our current thinking is to introduce a $G argument:
```
    <generator id="myUniqueId">
      <description>An example of Open CMSIS Generator</description>
      <exe>
        <command host="win">generator.bat</command>
        <!-- $S = Device Family Pack base folder -->
        <argument>$S</argument>
        <!-- D = Device (Dname/Dvariant as configured by environment) -->
        <argument>$D</argument>
        <!-- Project path and project name (as configured by environment) -->
        <argument>$P</argument>
        <argument>#P</argument>
        <!-- #name of the gpdsc template contained in this PACK and aligned with GPDSC name-->
        <argument>my_template.gpdsc</argument>
        <!-- #path of the generator input file -->
        <argument>$G</argument>
      </exe>
      <!-- workingDir></workingDir -->
      <!-- path is specified either absolute or relative to PDSC or GPDSC file. If not specified it is the project directory configured by the environment -->
      <gpdsc name="$P\my_gpdsc.gpdsc" />
    </generator>
```

For the format of this input file, we are in favor of "unifying it" with the lock file proposal from @slhultgren.
This file must describe the project "at file level" so that there is no need to go back to the pdsc to understand it.

__Executive Summary__
We need:
1. a spec update to allow multiple components to reference the same generator (more a clarification than a real update)
2. a spec update to introduce the unique generator id concept
3. a spec update to define the generator input file format : hopefully, this can be the same as the lock file format
4. a spec update to introduce the new $G argument in the \<generator\> element
5. a spec update to have new files categories
6. a csolution tool update to invoke only each __unique__ generator and not all generators
7. a csolution tool update to generate the generator input file and pass it with $G

__Points to be clarified__
The concept of user selected component passed to the generator is unclear to us.
I guess that if the project composition evolves, then the project manager (csolution) should reinvoke the unique generators "blindly". Then, up to each generator to determine if something needs to be done or not depending on what it does ?
The complexity is more when a file used by a generator gets updated.
If this file is not related to project composition, then the project manager (csolution) has no way to detect this change and determine that a generator must be run again.
Probably here we should not try to deal with it at project manager level.
The "entity" updating the file should be able to invoke the concerned generator, ideally via csolution so that csolution can decide if it runs only the requested generator or all of them.
This might get us to the point identified by Joachim : expressing dependencies between generators.
I am not sure we should shoot for this complexity from day one.
The dependencies could be kept at components level, then I am not sure we should try to have the project manager dealing with all dependencies coming from generators.
Open thoughts : if really we want to go this way, then probably a generator should also define his dependencies (file level) in its definition so that the project manager (csolution) can evaluate it. But I am afraid we will enter a too high complexity level here.
Probably the project manager (csolution) should care only for components.

fred-r, 2022-10-07:
Is this discussion superseded by: 
https://github.com/Open-CMSIS-Pack/devtools/blob/main/tools/projmgr/docs/Manual/Generator%20(Proposal).md

ReinhardKeil, 2022-10-19:
Yes, the https://github.com/Open-CMSIS-Pack/devtools/blob/main/tools/projmgr/docs/Manual/Generator%20(Proposal).md is the result of that discussion.

fred-r, 2022-11-16:
Hi,

in the proposal, this part is really fine in my opinion:
https://github.com/Open-CMSIS-Pack/devtools/blob/main/tools/projmgr/docs/Manual/Generator%20(Proposal).md#components-element

But, the workflow proposal is not fully matching what we had in mind I think:

1. csolution tool generates *.cbuild.yml file that provides project context and a list of user-selected components.
==> I would disagree here, to me cbuild.yml must provide the complete context  of the project (similar to lock file)

2. Generator/User generates a *.genlayer.yml file
==> I still have the concern of backward compatibility. Still unclear to me why .gpdsc disappears.
I agree .gpdsc was an incomplete name as it is not only a generated-pdsc but a generated-project-definition-refinement.

3. https://github.com/Open-CMSIS-Pack/devtools/blob/main/tools/projmgr/docs/Manual/Generator%20(Proposal).md#example-content-of-genlayeryml-in-this-case-stm32cubemxgenlayeryml
==> I still think that here we have only a result, and not what we need to understand where what comes from.
To be more specific : gpdsc provides conditions ==> how to do without it ?

Can we proceed as follows:

1.	We document the gap between genlayer and gpdsc
2.	we see what needs to be bridged or not
3.	we decide how to bridge it
4.	we decide if gpdsc is deprecated or not

slhultgren, 2022-11-28:
Indeed. One point we need to decide is 

`How do we know when we need to (re-)run the generator or not`
I believe it is not enough to check if some specific file exists or not, since a generator may require re-running even after the first run. So the only way here IMO is to be able to run each generator in a kind of `dry-run` mode, where we can basically ask the generator to only create the `gpdsc`, and then we can analyze this to determine if something has changed or not.

This would require a new `mode` to be added for generators. All generators need to be able to run in two modes:
* `dry-run` mode, which only creates a `gpdsc` on stdout (so we don't touch the filesystem)
* `normal` mode, which creates the `gpdsc` and any other potential files that the generator may want to create

slhultgren, 2022-12-12:
So to expand a bit on the `1, dependency file pattern list`:
```xml
  <generator>
    <additionalInputs>
      <filePattern>$P/.*/settings.json</filePattern>
```

Compare with `<HASH-METHOD>` stored in the `generator` section in the `cbuild.yml`.
If the HASH from the `<additionalInputs>` + `gpdsc` differs from stored HASH in `cbuild.yml` (or if the `gpdsc` is missing) then the generator needs to be re-run.

ReinhardKeil, 2023-01-09:
This is outdated an replaced with https://github.com/Open-CMSIS-Pack/devtools/blob/main/tools/projmgr/docs/Manual/Generator%20(Proposal).md

closed 2023-01-09 by ReinhardKeil