Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #107: Pack in csolution does not work

fred-r, 2022-03-22:
```
frq09468@LMECWL0888 MINGW64 /c/Cube20/stm32cube_fw_all/examples/hal/gpio/toggle/projects (main)
$ csolution -s gpio-toggle.csolution.yml convert
error csolution: board 'B-U585I-IOT02A' was not found
error csolution: processing context 'gpio-toggle.AC6+B-U585I-IOT02A' failed
```
with:
```
  packs:
    # specify the DFP pack to avoid selecting the U5 device from Keil DFP pack
    - pack: STMicroelectronics::STM32U5xx_DFP
  projects:
    - project: ./gpio-toggle.cproject.yml
```

though:
```
$ csolution list packs
STMicroelectronics::Example_HAL_GPIO_Toggle@0.0.1-alpha1
STMicroelectronics::HAL_CRC_UserDefinedPolynomial_example@0.1.0
STMicroelectronics::STM32U5xx_DFP@0.0.1
STMicroelectronics::STM32U5xx_DFP@0.1.0-alpha1
STMicroelectronics::STM32U5xx_Drivers@0.0.1-alpha1
STMicroelectronics::STM32U5xx_Drivers@0.1.0-alpha1
STMicroelectronics::board_if_codegen@0.0.1
STMicroelectronics::board_resources_B-U585I-IOT02A@0.0.1
STMicroelectronics::board_resources_STM32L562E-DK@0.0.14
```

ReinhardKeil, 2022-03-28:
With the `- pack:` definition you basically select just one pack.  I guess this pack does not contain the `board:` specification.  Could you add also the `- pack:` definition that contains the `board:`?

fred-r, 2022-03-28:
I see, you mean that the packs I list here are not an "additive" information to force a selection in the available packs.

```
<html>
<body>
<!--StartFragment-->

packs: | Content
-- | --
-filter: | Lists a selection for software packs (additive) with wildcards
-pack: | Explicit pack specification (additive), overrules filter: settings

<!--EndFragment-->
</body>
</html>packs:	Content
   -filter:	Lists a selection for software packs (additive) with wildcards
   [-pack:](https://github.com/Open-CMSIS-Pack/devtools/blob/main/tools/projmgr/docs/Manual/YML-Format.md#pack)	Explicit pack specification (additive), overrules filter: settings
```

By reading additive I thought it was "in addition to" but in fact it replaces the list of packs right ?
We select only the packs listed here, ignoring the content of CMSIS_PACK_ROOT and local packs ?

ReinhardKeil, 2022-04-05:
`filter:` has been combined with `pack:` as explained here https://github.com/Open-CMSIS-Pack/devtools/blob/main/tools/projmgr/docs/Manual/YML-Format.md#pack

I believe this was therefore a bug that is in the meantime already fixed.  Could you confirm?

jkrech, 2022-04-10:
Closing comment:
a) we decided not to make a distinction between including a specific pack and a set of packs (filter)
b) only the specified packs which are also installed in $CMSIS_PACK_ROOT or have local references are considered by csolution. 
c) in case no `packs:` are specified, all (!) installed packs in $CMSIS_PACK_ROOT are considered

closed 2022-04-10 by jkrech

fred-r, 2022-04-10:
Hi @jkrech,

your explanation is very clear but when it comes to the documentation:

"The pack: list allows to add specific software packs, optional with a version specification."

==> to me add is misleading because I understood it as "adding" to CMSIS_PACK_ROOT while in your explanation it is clear it replaces CMSIS_PACK_ROOT.

Also, we need to clarify in b) the meaning of local pack.
Is it "local packs" as installed with the pack installer ?

Or is it local if we specify the "path" attribute:

Also, for this attribute:
path: | Explicit path name that stores the software pack

It is not crystal clear to me if the "path" can be anywhere (disk or remote location or even github?) or must point to $CMSIS_PACK_ROOT or local packs paths or a path within the projects folders ?

jkrech, 2022-04-10:
I agree, that we need to get the documentation and implementation fully aligned. I agree that the term "add" is misleading unless you embrace the concept that 
- not specifying 'packs:' in csolution.yml means that all *.pdsc files from installed packs in $CMSIS_PACK_ROOT/<vendor>... and referenced in $CMSIS_PACK_ROOT/.Local/local_repository.pidx are processed by csolution.
- if you specify 'packs:' in csolution.yml means that from the available pool of pdsc files listed above (installed packs and local_repository.pidx referenced packs) only the listed packs are "added" for processing by csolution.

=> not specifying packs: is in fact equivalent to 
```
packs:
  - pack: *
```
(note: the current schema does not support to specify only '*'. I need to check why)
When it comes to specifying a pack with a path, the file reference is specified relative to the csolution.yml file. 
```
<csolution_dir>/<path>/<packvendor>.<packname>.pdsc
```
Note: The <packvendor>.<packname>.<packversion> shall take precedence over any version of this pack being found in the pool of packs even if enabled e.g. by
```-pack: <packvendor>::*```

We are not anticipating that the csolution project has any further information about the directory of the 'path', e.g. that it is a git submodule of a certain tagged version. This would be under the full control of the maintainer of the csolution project.