Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #108: Introduce a pack type ?

fred-r, 2022-03-24:
In another issue, we discussed that sometimes we can end up with several DFP packs in a project if the same device is published by 2 pack providers.

To help detecting such issues, can't we introduce a concept of pack type ?

For instance a pack can be of type DFP and contain devices definitions and startup components.
A pack can be of type Drivers and contain drivers components.

By doing this maybe we can reach several objectives:
1. Propose a granularity for packs because we said that huge pack would be a problem for the "local copy" approach
2. Be able to say that some components (Cclass="Device" Cgroup="Startup") can come only from some packs (DFP)
3. Be able to say that DFP must come from the same vendor as the device to avoid multiple devices definitions

We do not need to mandate to implement checks.
Some tools may decide not to check anything while other tools may decide to do some consistency checks based on such rules.

fred-r, 2022-03-29:
maybe we should also see it as a way to refine this possibility:

https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/element_dominate.html

This "dominate" sounds interesting but I think it makes more sense if we correlate it to a category of packs.
Some elements should come only from specific packs.
Then we can say that only 1 pack of a type is allowed.
This sounds better to me than a very wide "dominate" where anybody can override anybody : what if 2 packs declare the same dominate ?

mdortel-stm, 2022-03-30:
Hi @fred-r 
Indeed, there is an issue when dealing with the same information coming from several sources.
I don't know if the constraints you propose would solve the issue.
Concerning the dominate attribute, IMHO, it will never work by design: if every packs tell they are dominating others, in the end, the problem remains. Such mechanism can only work if the information is centralized and not provided by another pack.

LMESTM, 2022-03-30:
Shouldn't that be addressed in a generic way by raising a conflict to the user that there are 2 conflicting candidates for resolving the same condition, whether a component or a device and that user should select the one he wants ?

fred-r, 2022-03-30:
We can ask the user, but the intention is:
1. to limit such situations: even if there is no condition mismatch, then the tool can detect that it is abnormal to select 2 packs of some categories (1 DFP per project is the rule IMO, unless you have several targets)
2. by introducing the pack type, open the possibility to define the expected content of a pack. 
As we are trying to standardize, I think it is good if packs have "similar" content. 
3. the type is also here to introduce the possibility to control: if you are not ST and publish a ST device in a DFP then a warning can be raised telling the end-user : do you want to work with this ST device definition not provided by ST ?
I do not mean it should be forbidden, but a warning can be helpful.

ReinhardKeil, 2022-04-05:
We discussed that, and our view is:
  - the explicit selection via `pack:` allows to specify the packs that should be used for a project.
  - with an IDE, the `pack:` list should be always generated, for example via a select packs dialog (https://www.keil.com/support/man/docs/uv4/uv4_ca_packselect.htm)
  - the IDE could also manage a list of packs type and offer a way to select packs using this pack type.

We may consider an extension of the csolution tool at a future point in time, but currently we don't see a lack of features.

fred-r, 2022-04-27:
The definition you highlighted here:
https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack-Spec/issues/115#issuecomment-1109909234

is a good proposal.
I guess that if I need another type of (HW) pack then I can define another type of element at the same level as board and device.

If you agree with this view, then I guess we can close this ticket.