Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #115: Open the door to more HW description

fred-r, 2022-04-16:
Today in the spec we can describe boards (via BSP pack) and devices (via Software Pack).

I think we induce some pack types without really defining them, that's why I still promote this proposal to introduce pack types:
https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack-Spec/issues/108

Then, in the project YML file we have definitions like:
```
  target-types:
    - type: B-U585I-IOT02A
      board: B-U585I-IOT02A
```

The tool picks the board definition from the BSP.

I think we may envision the use of various pack types to describe hardware.
So I think it may make sense to explicitly tell the tool which packs he can use when defining this target.

We may have something like:
```
  target-types:
    - type: B-U585I-IOT02A
      board: B-U585I-IOT02A
      target_packs: 
         - pack: can be a BSP pack for a board definition
         - pack: can be a DFP pack for a device definition
         - pack: can be a pack dedicated to some hardware definitions (to be discussed)
```

The proposal is really to let the tool know about all the packs that relate to a target.
This should be restricted to some pack types (BSP, DFP, to be discussed).

As hardware is not as versatile as SW it probably makes sense to indicate the links to be considered.

fred-r, 2022-04-26:
As suggested by Evgueni : maybe we can deal with packs dependencies.

fred-r, 2022-04-27:
Ok, so based on your clarifications:
- If I need another HW pack type then I can create in the spec another element at the same level as board and device
- From the BSP pack I can create dependencies on these additional packs

ReinhardKeil, 2022-04-27:
@fred-r yes I suggest to close the ticket for now. I'm sure we need extension over time.

fred-r, 2022-06-10:
Reopening as the extension is needed now.

ReinhardKeil, 2022-06-10:
Can you explain the requirement next Tuesday?

ReinhardKeil, 2022-09-06:
Sorry it is unclear what is needed now.

fred-r, 2022-09-06:
Superseded by #144

closed 2022-09-06 by fred-r