Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #116: Incomplete handling for file attr="template"

fred-r, 2022-04-20:
In the "file" entry, the category is defined as "informative" : "Defines the purpose of the file."
While the attribute seems to be more related to actions: "Defines the special use and handling of a file"

But when we follow the attribute link we can read: 

<html>
<body>
<p style="font-family: &quot;Lucida Grande&quot;, Verdana, Geneva, Arial, sans-serif; font-size: 13px; line-height: 1.3; color: rgb(0, 0, 0); font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><b>Table: File Attributes</b></p><p style="font-family: &quot;Lucida Grande&quot;, Verdana, Geneva, Arial, sans-serif; font-size: 13px; line-height: 1.3; color: rgb(0, 0, 0); font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; white-space: normal; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; background-color: rgb(255, 255, 255); text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;">The file attribute defines the special handling in the project when being used as configuration, template, or interface file. The table lists the values available as a file attribute.</p>

attr= | Description
-- | --
config | The file is a configuration file of the component. It is expected that only configuration options are modified. The file is managed as part of the component, as a project-specific file typically copied into the component section of the project.s
template | The file is used as a source code template file. It is expected to be edited and extended by the software developer. The file can be copied into a user section of the project.

<br class="Apple-interchange-newline">
</body>
</html>

==> The main difference in the handling is mainly where it should be copied:

- config : "file typically copied into the component section of the project.s"
- template: "The file can be copied into a user section of the project."

But, when using the tools, only the config files are managed (copied in RTE folder).
Today, the template files are not copied at all.

I agree that the spec says "can be" but we have no way to tell the tool if we want it to be copied or not.

I see (at least 2 use-cases):
- optional template : no need to copy it, the end-user can decide if he needs it or not
- mandatory template: the component cannot work without it, then it should be copied "into the user section of the project"

This would be useful because to me these 2 kind of templates do exist:
- almost empty template files just providing a skeleton of what you can do
- templates providing "patterns" that you must implement

So I think we are missing something:
- in pdsc a way to indicate if the template MUST be copied or CAN be copied
- in the project a way to say, if the optional template is copied or not in the user section: today I am under the impression that it is the responsibility of the end-user to add it by copy. But to me, the idea with components is to "hide" the source of the component, no user copy should be needed. 
- both for config and template : where the component section and user section is so that the files are copied where the user wants.

If we do not enable this, the risk is that the attr="config" is systematically used to copy a file, even if it is not a configuration file.
Then the semantics becomes clumsy...

fred-r, 2022-04-25:
This topic also has a link with instantiable components (maxInstances) where a copy and renaming is done.

fred-r, 2022-04-26:
Basically, I think that to connect 2 components A and B where:
- A consumes a feature
- B provides a feature

Then we can have:
- a "config" file providing the glue : the logic is fully implemented and the end-user makes some configuration choices
- a "template" file providing the glue : here the logic is not (fully) provided, it is not only a matter of configuration but really filling a "template" or implementing a "pattern"

ReinhardKeil, 2022-04-26:
I suggest to use the term "interface" to drive this discussion forward.  I believe an "interface" could be also an "component".  If we add the attribute "auto" or "hide", such components could be automatically added when needed.  Let's explore the use case here further before we come to a decision.

fred-r, 2022-04-26:
Ok, I will try to collect all needs and see if it goes beyond interfaces (but in this case it should be an applicative pattern so matching the template definition I guess).

Nevertheless, I'd like to avoid creating too many components, even if it is possible to hide them.
There is already the concept of API to provide a header file.

fred-r, 2022-05-03:
It goes beyond interface:
- sometimes it is some logic that is mandatory for the project but must be offered to the user for him to tweak it 
- sometimes it is some logic that is mandatory for the project but that the user must complete
- sometimes it is indeed an interface

In fact we need a way to tell that a file must be copied in the user section of the project in a mandatory manner.
This is not the "template" purpose so I think we need another attr.

ReinhardKeil, 2022-09-13:
@fred-r The templates are used in MDK and the Eclipse version of the CMSIS-Pack system. Maybe you try it yourself.
See here https://developer.arm.com/documentation/101407/0537/Creating-Applications/Creating-Projects/Add-Source-Files-to-Project?lang=en

IMHO it does not make much sense to capture the version of a template, as it is basically a user stub for implementing a SW feature in their application.

If you have something to add, I suggest we discuss it in todays meeting.

ReinhardKeil, 2022-12-06:
No further feedback - hence closing it.

closed 2022-12-06 by ReinhardKeil