Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #119: Decision process in OpenCMSIS, especially for the spec

fred-r, 2022-04-28:
Dear all,

I hope I won't appear to be rude and won't offend anybody but at the moment I am not comfortable with the decision process in OpenCMSIS.

In the weekly forum, we address a list of topics, usually we call for comments but we do no set deadlines nor method to conclude.
And sometimes an issue can just "disappear from the radar" for some time.

Can we improve this way of working to reach conclusions in a more predictable way ?

What I imagine is that in the weekly forum we do a kind of CCB:
1. Review the points
2. If a point deserves more discussion and thinking then I think we should set a deadline for this discussion to happen and maybe also a owner who may summarize what has been discussed when the deadline is met
3. When the deadline is met, the CCB decides how to conclude the point
4. If the point is accepted, then we need to decide: is it clear enough and we do nothing more ? Does it deserve an ADR ?
5. In case an ADR is needed, somebody is appointed to write it within 2 weeks
6. The ADR is then reviewed and formally approved within 2 more weeks

These ADRs should be publicly available and very short, I guess md format would be perfect.

To be discussed...we can already experiment the decision making process with this proposal :-)

iomint, 2022-05-03:
I think A linaro representative involvement could be also interesting being a neutral party

ReinhardKeil, 2022-05-03:
Agreed.  Maybe Bill can help us how other projects organize that.

IMPORTANT NOTICE: The contents of this email and any attachments are confidential and may also be privileged. If you are not the intended recipient, please notify the sender immediately and do not disclose the contents to any other person, use it for any purpose, or store or copy the information in any medium. Thank you.

fred-r, 2022-05-04:
I think we can take some examples form here:
https://www.redhat.com/en/blog/understanding-open-source-governance-models

I am in favor of the election process.

My proposal is the following:
1. The CCB (or Technical Steering Committee) decides that an issue must be studied
2. A working team is selected and a spokesman presents the conclusions to the CCB/TSC after N weeks
3. Here 2 things can happen 
    ==> direct adoption of the conclusions
    ==> no consensus: a Linaro "neutral" representative (Bill?) tries to find a gentlemen's agreement.
    If no agreement is found then a vote must happen and I am thinking about a triumvirate:
   - 1 ST representative : can be @iomint
   - 1 NXP representative: David ?
   - 1 ARM representative : Reinhard ?
   Nobody can give a blank vote so there is always a decision (at least 2 versus 1)
4. If the change in the spec is accepted then a PR on the spec is issued by the submitter or somebody appointed to do so

I understand that this might be a problem for ARM who is contributing more for the vanilla tools.
But, here I am talking only about the spec.

When it comes to the code, I think a maintainer should be appointed for each tool and he makes the decision to implement or not if the change is optional. If the change is mandatory, the maintainer can decide when the change is to be implemented.
The change can be implemented by another contributor but the maintainer defines the period when he wants to deal with it.
If the maintainer does not want to accept a mandatory change, then I guess a CCB vote should be required?

For accepting the PRs on the code, the maintainer is the gate keeper.
I guess we may have 3 maintainers:
- cpackget : Charles ?
- devtools: Daniel ?
- cbuild : Daniel ?

This way, I think we can have:
- an Electoral approach for governing the spec
- a Doer/Founder approach for governing the vanilla tools

@jkrech : can we try to discuss this to elaborate what you want to present next week ?

jkrech, 2022-05-04:
@fred-r  thanks for the input. I am happy for you to present your proposal next week. What I had prepared for this week was rather focused on the tracking aspect of change requests to make this transparent for everyone. At the same time I will need a bit of time to review your proposal and provide feedback here.

jkrech, 2022-05-09:
@fred-r, while I think your proposal of gatekeepers per repository make sense, we must ensure that the specification and reference implementation do stay in sync somehow. Therefore it maybe necessary for the Steering Group to resolve resourcing/contributions to the reference implementation as part of change requests.

fred-r, 2022-05-09:
That's the point where I disagree :-)
To me resources are always a bottleneck and this should not hinder the capacity to specify :-)

If you look at 3GPP (past experience), the spec is usually far ahead of implementation.

If we take a look at something more software oriented : the JavaScript specification evolves regardless of its adoption by the web browsers.

So, I am in favor of decorrelating both aspects.

VGRSTM, 2022-05-09:
I'm voting same as @fred-r
I've already shared my mind about Thanks couple of comments (as example [[devtools] Promote standalone tools being orchestrated by consumer itself](https://github.com/Open-CMSIS-Pack/devtools/issues/85) )
To me primary goal on OpenCMSIS initiative is to define / agree on some standard allowing industry to converge on. It will ensure all adopters to enjoy some do facto interoperability. Such is already awesome !
We come then to `devtools`. I and guess we are all happy to have them. But such are extra helping us to support proof of concept, discussion running real cases, .... `devtools` are not first guess outcome expectation to me out of OpenCMSIS.
Adding `devtools` as part of specification approval cycle is leading to slow down the whole process what's a bit a pain.
As extra I would say I'm a bit worried about `devtools` because at a point they may fulfill some specification lacks promoting some behavior Thanks their implementation becoming under the hood do facto part of the specification. My fear is such is going to lead to issues because my guess is not all OpenCMSIS adopters are going to rely on `devtools` ...

iomint, 2022-05-10:
Couple of comments:
-@fred-r:  Considering how Linaro segment groups (incl Lite that is hosting Open-CMSIS) works, I think you have to consider Linaro as a voting member. I let Bill F comments
-I do think it comes as a surprise: I am also of the opinion of my colleagues to decorrelate the spec from the ref tools implementation. After we can discuss some boundary conditions rules like if something is no supported by any vendors after a given amount of time than it is deprecated (assuming no side effect etc ...)
-The option to have maintainers identified for each tools makes sense.  I think it is a (almost) universal practice for open source projects

fred-r, 2022-05-13:
I can see that in the project tab the point is tagged as "Accepted" but I did not see anything about how to handle the decision making when no consensus is reached.
So I am not sure we are ready to go to implementation/deployment.
Is there a document explaining all this ? (I can't find it from the project pane)

jkrech, 2022-05-30:
Proposed Change Board:
https://github.com/orgs/Open-CMSIS-Pack/projects/2/views/1
Agreed Decision Committee Charter:
https://linaro.atlassian.net/wiki/spaces/CMSIS/pages/28717383715/Open-CMSIS-Pack+Decision+Committee+Charter

closed 2022-11-08 by jkrech