Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #124: Formally support board specific memory in board support

tcsunhao, 2022-05-24:
In existing CMSIS pack spec, memory definitions are formally supported under device.

We should be aware of that many boards have dedicated memory chips onside whose memories are mapping into device memory map. One example is NXP RT series, nearly all boards have board memories. These board memories are missing in existing CMSIS packs. 

Open-CMSIS-Pack-Spec should support set board memory in schema to hold board memory data.

fred-r, 2022-05-24:
I think this can be addressed with the rzone format but we need to determine where to store what:

https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack-Spec/issues/120#issuecomment-1116195927

fred-r, 2022-06-07:
To me, it seems that a direction has been taken when creating rzone ?
I think it is clearer to have a dedicated format in a separate file.

jkrech, 2022-06-07:
Slightly related to the above, I believe we need to consider extending the board description with flash programming algorithms required to program on-board flash memory in the CMSIS-Pack specification.

fred-r, 2022-06-21:
WOuld be fine for me, what do you think @tarek-bochkati ?

pmoSTM, 2022-07-08:
Could you please explicit the memory information you want to expose? Provide some examples? thank you.

jkrech, 2022-07-08:
@pmoSTM I think the original idea was to extend the [\<board> section](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_boards_pg.html#element_board) with [\<memory> child elements](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_family_pg.html#element_memory) identical to the format used in \<devices>.
I had suggested that we should also add the [\<algorithm>](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_family_pg.html#element_algorithm) at the same time to specify the flash programming algorithm for external flash memory chips mounted on the board.
For example:
```
<board vendor="STMicroelectronics" name="STM32G474E-EVAL" revision="Rev.A" salesContact="http://www.st.com/stonline/contactus/contacts/index.php">
  <description>STMicroelectronics STM32G474E-EVAL</description>
  <image small="MDK/Boards/ST/STM32G474E-EVAL/Documentation/stm32g474e-eval_small.jpg"
         large="MDK/Boards/ST/STM32G474E-EVAL/Documentation/stm32g474e-eval_large.jpg"/>
  <book category="overview"  name="https://www.st.com/content/st_com/en/products/evaluation-tools/product-evaluation-tools/mcu-mpu-eval-tools/stm32-mcu-mpu-eval-tools/stm32-eval-boards/stm32g474e-eval.html" title="Web Page"/>
  <book category="schematic" name="MDK/Boards/ST/STM32G474E-EVAL/Documentation/?.pdf" title="Schematics"/>
  <book category="manual"    name="MDK/Boards/ST/STM32G474E-EVAL/Documentation/?.pdf" title="User Manual"/>
  <mountedDevice    deviceIndex="0" Dvendor="STMicroelectronics:13" Dname="STM32G474QETx"/>
  <compatibleDevice deviceIndex="0" Dvendor="STMicroelectronics:13" Dname="STM32G474*"/>
  <debugInterface adapter="JTAG/SW" connector="20-pin JTAG (0.1 inch connector)"/>
  <debugInterface adapter="JTAG/SW" connector="20-pin Cortex debug + ETM Trace (0.05 inch connector)"/>
  <debugInterface adapter="ST-Link" connector="Micro-USB"/>
  <debugProbe connector="Micro-USB" debugClock="10000000" debugLink="swd" name="ST-Link"/>
  <memory name="Ext_Flash" access="rx" start="0x90000000" size="0x08000000" startup="0" default="0"/>
  <algorithm name="CMSIS/Flash/MT25QL512ABB_STM32G474E-EVAL.FLM"    start="0x90000000" size="0x08000000" RAMstart="0x20000000" RAMsize="0xA0000" default="0" />
</board>

The external on board memory is a resource we require in an rzone file. The question is, whether we agreee to duplicate the information here with the risk of inconsistent information.
```
One benefit of duplicating it that we had identified is the fact that this information can be "extracted" and e.g. used for rendering on a web page or a product selector, without the need to download the whole pack in order to read this information from the associated rzone file within the pack.

ReinhardKeil, 2022-07-19:
Added <memory> and <algorithm> child element to board element in Open-CMSIS-Spec

closed 2022-07-19 by JonatanAntoni