Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #125: Support processor(core) specific memory data in pdsc

tcsunhao, 2022-05-27:
In NXP RT1170, each core has its own TCM: 512KB RAM shared with M7 TCM and 256KB RAM
shared with M4 TCM, such memory data should be presented in Open-CMSIS-Pack pdsc.

Core special memories should be a standard for pdsc, not in some kind extension point.

Related to [board specific memory](https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack-Spec/issues/124)

jkrech, 2022-05-30:
@tcsunhao, could you please be a bit more specific which features you are lacking?
In the specification the memory element is described here: https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_family_pg.html#element_memory
Note that in case of a multi-core device the attribute `Pname` is used to describe that a memory belongs to a certain processor instance.
If a memory is specified without `Pname`, the resource belongs to all processor instances.
There is also an attribute `alias` to indicate that a <memory> element shares the actual physical resource with access via a different address range.
What is the meta information we are missing and shall add to the specification?

tcsunhao, 2022-05-30:
Thanks @jkrech, existing Open-CMSIS-Pack can meet our needs.

closed 2022-05-30 by tcsunhao