Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #126: Pack Schema does not strictly enforce Semantic Versioning for pack versions

jkrech, 2022-05-31:
The current version of the Pack Schema (PACK.xsd) does not enforce semantic versioning for the attribute `version` in the `<release>` element. Since this version needs to match the version part of the pack filename `<vendor>.<name>.<version>.pack` and `packchk` requiring the latest version/first `<release>` element in `<releases>` section

The current version of `cpackget` works under the assumption that the filename always has a semantic version compliant version part.

Is this a realistic assumption and do we tighten the PACK.xsd or do we need to relax this requirement and allow deviations.

A typical deviation is to allow leading zeros.

ReinhardKeil, 2022-06-02:
There are at least two projects that are using a `year.month` concept for top-level version numbers.
 - AWS FreeRTOS LTS has been mapped into the CMSIS version schema by using the schema `yyyy.mm.patch`
 - TFL as provided by the ML team in Arm by using the schema `major.yy.mm`
The `mm` encoding has the issue that it may start with a leading `0` followed by a numeric digit, for example `02` encodes February. Strictly speaking this violates [sematic versioning rules](https://semver.org/) where version numbers cannot start with a leading `0` digit.

To support the `year.month` concept of some software projects, it is therefore proposed to relax the sematic versioning check and allow also a leading `0` for MAJOR, MINOR, PATCH fields.  Currently this is enforced in some tools, i.e. `cpackget` whereas other tools seem to accept that format.

mdortel-stm, 2022-06-02:
Indeed leading 0 is not accepted by semantic versioning but it does not break the meaning of the version items. So the deviation looks obvious for me and not breaking anything, AFAIK.
What I would expect by not bringing semantic versioning into the pack schema is clarifying what is the consequence of not using semantic version the processing flows Open-CMSIS ecosystem will support.

fred-r, 2022-06-02:
A trade-off may be to introduce a --strict-semver option in the cpackget and packchecker tools ?

Or the other way round:
- by default it is strict
- we add a --loose-versioning option

It is not perfect but it may be a way to "educate" pack designers without preventing them from publishing their packs ?

As pointed out by Maxime the key point is : do we plan to leverage semver in the lifecycle management ?

ReinhardKeil, 2022-06-03:
I agree with to aspects of that posts:
  - documenting the consequences of not using semver rules
  - adding a option to packchk that relaxes semver rules (i.e. --relax-semver), by default semver rules should be enforced

I would propose to relax the rules in all other tools.  Of course we could add an option -strict-semver, but given the fact that this would make some packs unusable, I'm not sure about the value (in my view it just adds complexity for the user/tool).

semver is leverage in the lifecycle management, we could also add an option to the PDSC file that indicates that semver rules are relaxed in a pack. That would indicate that every version mismatch is reported as potentially incompatible.  However it should be noted that this is more at the component level, not at the pack level itself. The AWS packs itself use semver for components, but not for the pack itself (where the version indicates somewhat the release date).

fred-r, 2022-06-03:
And also when it comes to semver, sometimes it is not easy when dealing with version with major being 0.
A common practice is to start a dev with 0.1.0.
But then, I guess a minor update could imply any kind of change...

For instance cpackget moved from 0.4.x to 0.5.0 and the API has changed (not backward compatible).
I think semver says nothing here.
It could have been 0.4.x+1 , but I guess that here @chaws decided to do 0.4+1.0 to idnciate the API break as the major is not used yet ?

Is there a rule explaining this ? How should the check behave here ?

LMESTM, 2022-06-03:
I think this is answered here:
https://semver.org/#spec-item-4

fred-r, 2022-06-03:
Ok, so as soon as major is 0 then there is nothing to check

fred-r, 2022-06-03:
SO can't this be a solution ?
People who want to "escape" semver check start with 0 ?

SO version 06.03.2022 could be 0.0603.22 for instance ?

ReinhardKeil, 2022-06-03:
@fred-r the issue is that `0` is allowed as single digit element of the version number, but `06` is not allowed.  My proposal is to accept also `06` and not flag this as error (as it is currently done by cpackget - where the proposed [regex](https://semver.org/#is-there-a-suggested-regular-expression-regex-to-check-a-semver-string) of semver has been used to validate the format

fred-r, 2022-06-03:
I was just thinking about something to avoid the strict parameter:

* check the major
* IF major > 0 : apply semver regex
* IF major == 0: do not check anything

ReinhardKeil, 2022-06-03:
I believe the best solution is the proposal from @thorstendb-ARM to use the definition in PACK.XSD.  It allows us to a relaxed PACK.XSD or a more strict version of the PACK.XSD file going forward and the format control is not in the tool, but in an external schema file.

fred-r, 2022-06-03:
Any solution for me, as long as we have flexibility :-)
What is the idea to switch from one xsd to the other ?

thorstendb-ARM, 2022-06-03:
In the first round the RegEx for "PackVersionType" will be hard coded into PackChk (and remain as default), but in a later version PackChk could be extended to scan a schema file for this attribute.

jkrech, 2022-06-07:
a) we only look at pack versions here
b) the request is to retain SemVer 2.0 format, but relax the format allowing leading '0' in major, minor, patch
c) I propose to ignore leading '0', therefore `05.07.00` == `5.7.0` in version comparison, so the code itself does not have to change.  The Pack vendor is responsible for ensuring unique pack versions!
d) Pack installer use the version to name the directory for the pack's installation directory. (`vendor/name/version`). Assuming c) it is safe to retain the version string as specified without the risk of installing duplicate pack versions.
e) from looking at the pack version the user will make an assumption that a major version change indicates an incompatible change. If the pack vendor chooses to encode the year in 'major', then a release in a new year will appear as incompatible. I don't know of any tools today, that implement functionality based on different major pack versions, so it may just need 'educating users'.

Is the above missing important aspects?

fred-r, 2022-06-07:
No strong requirement from my side (even if I support SemVer adoption), I think we need flexibility to ease the adoption of OpenCMSIS.
As we are dealing with packs, I think  probably we should not use this information in the PLMN then?
Better to use components versioning only.

ReinhardKeil, 2022-06-13:
I believe we should have some common sense.  https://semver.org/ defines
```
<valid semver> ::= <version core>
                 | <version core> "-" <pre-release>
                 | <version core> "+" <build>
                 | <version core> "-" <pre-release> "+" <build>
```
This gives already flexibility, for example:
```
vendor.CMSIS-FreeRTOS.10.4.6+XAlloc     # to indicate a extension in the memory allocation
vendor.FatFs.0.14.2-beta1               # to indicate a pre-release
```

Also packs have a release date, which takes away one argument mentioned in the video.  

We are using sematic versioning in MDK since ~10 years and it helps a great deal to indicated and keep compatibility.

jkrech, 2022-07-11:
Proposal: 
Modify Open-CMSIS-Pack Schema (PACK.xsd) such that SemanticVersioning is checked strictly, except allow leading '0' for major, minor and patch.

closed 2023-02-07 by jkrech