Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #131: How to specify a component dependency on board

tcsunhao, 2022-06-02:
From [condition](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_conditions_pg.html#element_condition), we could know following condition attributes are supported. 
[image]

Question is how to specify a component dependency on board? There is no board related attribute there.

ReinhardKeil, 2022-06-02:
@tcsunhao this is a feature gap that we should IMHO add.

**PROPOSAL**
Add to [conditions element](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_conditions_pg.html#element_condition) a way to filter for boards with:

Attribute  | Description
:-----------|:--------------------
Bvendor   | Specifies board vendor
Bname     | Specifies board name
Brevision  | Specifies board revision

fred-r, 2022-06-02:
We may even support the extension points in the conditions ?
@jkrech : what do you think ?

jkrech, 2022-06-07:
@fred-r, I think this needs to be reviewed as part of specifying: https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack-Spec/issues/27 but I do agree that we need to support the extension of attributes used to `filter` components for a given context.

jkrech, 2022-06-07:
example:
```
  <require Bvendor="ARM" Bname="V2M-MPS2-SSE-200" Brevision="B"
```

ReinhardKeil, 2022-07-19:
added 'Bvendor', 'Bname', 'Brevision' attribute to conditions element to allow board specific filtering to Open-CMSIS-Spec