Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #132: What are the rules to submit a PR on the spec ?

fred-r, 2022-06-02:
It is unclear to me what is expected from people submitting a PR on the spec : xsd and/or doxygen.

I can see some PR like this one:
https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack-Spec/pull/118

With no update of the versioning and the history, guthub action tests failing, and it is approved and merged.

What are the criteria to update a spec version ?
What are the criteria for a PR to be accepted ?
The tests are optional ?

JonatanAntoni, 2022-06-02:
This can be debated and we didn't strictly follow any specific rule in the past. People "just" adding new vendor ids tend to not increase the schema version and date.

jkrech, 2022-06-02:
Is it fair to say it is a warning rather than an error then?

JonatanAntoni, 2022-06-02:
Yes, I considered this a warning so far. Unfortunately, GitHub doesn't distinguish between warnings and errors. Checks can either pass or fail.

fred-r, 2022-06-02:
The check is one thing (important), but more important to me is the versioning rules.
Currently, I am under the impression that we can end-up with different contents under the same version.
This can lead to misunderstandings.

ReinhardKeil, 2022-09-13:
I believe this highlights an issue that we should have:

1. a concept of releases and a view to the "release quality" specification.
2. a workflow for contributing changes to the specification, my proposal is below.

**Workflow**

1. change the verbal documentation in the doxygen
2. change the schema to reflect the changes
3. ensure that history in the documentation reflects the changes
4. release a new version of the "specification" once all changes are accepted.

Also during this workflow there should be a way to view the work-in-progress specification.

ReinhardKeil, 2024-01-30:
Is there anything that is not working today?  Should we add more process given the limited resources that we have.?

closed 2024-02-12 by jkrech