Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #134: Add <csolution> element to *.PDSC format

ReinhardKeil, 2022-06-07:
This is WiP to provide more details for https://github.com/ReinhardKeil/IoT_App/blob/main/ProposedStructure/Proposals.md#changes-to-the-pdsc-file

It replaces to an extend #122

-----

For a project that is based on the `csolution.yml` file it should be possible to add content to a software pack that is copied literally to the csolution/cproject directory.  This has multiple use-cases and therefore the <csolution> element could have the following child elements:

<cproject> a generic example projects
<cimage> a pre-build image that should be added to a csolution
<clayer> a software layer

tcsunhao, 2022-06-28:
Record the question here: how does Open-CMSIS-Pack expect vendor's tool/IDE to use such <csolution> element data in pdsc, present them in the menu like component to customers? I don't see clear granularity or restrictions in clayer data for tools analysis(Please correct me if I have a wrong understanding, sorry in advance).

Another comment is that the way of usage for clayers/project/image here implicitly need decouple between csolution(target-type/build-type) and clayers/project/image.  Associated with https://github.com/Open-CMSIS-Pack/devtools/issues/356

jkrech, 2022-07-05:
Since the build- and target-type is defined by the csolution.yml file, I do agree that the referenced elements `cproject` and `clayer` must not rely on for-type/not-for-type unless a common naming scheme and associated settings have been somehow agreed.
Discussion regarding https://github.com/Open-CMSIS-Pack/devtools/issues/356 is still not concluded.

iomint, 2022-07-05:
Finally  I am not sure to catch the final goal of this proposal allowing to add a "csolution" element in a pdsc file.
Doing one step back, can't we stick to the "example" element in the pdsc file to point to a project ?

jkrech, 2022-07-05:
While the <example> element would reference and specify a self-contained csolution project (copy folder and build), the <csolution> section contains "pre-configured" or "pre-build" elements (image, layer or "generic" project) to be used to be added to a solution that a user constructs.

jkrech, 2022-07-07:
Request from @iomint: Consider alternative name instead of <csolution> to avoid confusion that this section would describe a csolution where in fact this section lists readily available `images`, `layers` and "generic" `projects`.

options
a) split up into 3 sections
```
<cprojects>
  <cproject ...>
  </cproject>
</cprojects>
<cimages> ... </cimages>
<clayers> ... </clayers>
```
b) find an alternative name instead of <csolution> one of:
```
  <csolutionParts>
  <csolutionElements>
  <csolutionItems>
```

iomint, 2022-07-11:
Joachim, all
we have had further discussions on this proposal with some colleagues and despite Reinhard initial post and the additional information you provided we still have a questionmark on the way it is going to work and the associated benefit bring by this proposal in particular for the "cproject" part. Would it be possible for you to share a few examples on the way to use this proposal e.g illustrating the way you have in mind for "pre-configuration" ?

ReinhardKeil, 2022-07-11:
A cproject is useful for csolution's that require multiple projects to become useful.  For example, it could be that an application always requires "mcuboot" or "psa" to be part of the csolution. Such projects might be 
 - (a) device or board specific and therefore come from a BSP or DFP or 
 - (b) being available in multiple variants (i.e. in case of PSA as implementations based on TFM or WolfSSL) which come then from a different software pack.

It appears that the element name "csolution" causes some discussion, we could use "BuildingBlocks" instead, however since these building blocks are copied to a csolution, I'm personally in favor of the element name "csolution".

ReinhardKeil, 2022-07-19:
https://github.com/mdk-packs/TrustZone shows a project that requires the <csolution> element to drive forward. It uses:

- Multiple images, some a pre-build (could be also delivered as projects)
- Layers that are board-specific

fred-r, 2022-08-25:
I guess that https://github.com/mdk-packs/TrustZone is used to illustarate a case with an aplication project (described by csolution.yml) requiring several "building blocks":
- 1 application project with 2 sw projects : AWS apps
- 1 prebuilt bootloader (binary image)
- 1 prebuilt TFM secure image (signed binary image)

But, I do not find a pdsc describing how to deliver these elements.

I like Joachim's proposal to have sub-categories per "building block" type:
- cimage for a prebuilt binary image (signed or not)
- clayer for a pre-composed/pre-configured set of components : this goes in a cproject.yml, not a csolution.yml
- cproject for a complete software project

Then, I guess the application project (csolution.yml) project assembling these building blocks could be delivered with the existing <project> element ?

```
      <project>
        <environment name="opencmsis:projmgr" load="my_project.csolution.yml"/>
      </project>
```

Regarding the container, <csolution> alone sounds a bit misleading to me, as we may think this is a complete project.
Why not <applicationElt> standing for application's elements ?

closed 2023-03-07 by jkrech

ReinhardKeil, 2023-03-07:
Genau, aber Fred war sowieso nicht am call.  Passt also