Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #135: Restrict Cclass values to pre-defined list

willlordarm, 2022-06-10:
Cclass values should offer a useful way to categorise components and support software discovery through an IDE UI, web interface or API. For example, a user could reasonably be expected to want to view a list of all the available `Security` middleware or all available `RTOS`. There is a pre-defined list of _suggested_ values for Cclass - https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/element_taxonomy.html#CclassType - but this is not a fixed taxonomy. This means that vendors have used Cclass in a variety of ways:

| Predefined values  | Actual values from packs |
| ------------- | ------------- |
| Audio |	 ADC |
| Board Support |	 AMP |
| Board Part |	 AWS IoT |
| Compiler |	 BLE |
| CMSIS |	 BSP |
| CMSIS Driver |	 Bluetooth |
| Device |	 Board Support |
| Data Exchange |	 Board |
| Extension Board |	 CMSIS Driver Validation |
| File System |	 CMSIS Driver |
| Graphics |	 CMSIS RTOS Validation |
| IoT Client |	 CMSIS |
| IoT Utility |	 ChipInit_Wizard |
| Network |	 Communications |
| RTOS |	 Compiler |
| Security |	 CycloneACME |
| USB |	 CycloneCRYPTO |
| Utility |	 CycloneCommon |
|  |	 CycloneSSH |
|  |	 CycloneSSL |
|  |	 CycloneSTP |
|  |	 CycloneTCP |
|  |	 Data Exchange |
|  |	 Data Processing |
|  |	 Device |
|  |	 Driver |
|  |	 Drivers |
|  |	 Examples |
|  |	 FAT32 |
|  |	 File System |
|  |	 Flex Software |
|  |	 FreeRTOS |
|  |	 Graphics Display |
|  |	 Graphics |
|  |	 HAL |
|  |	 IEC60730_Support |
|  |	 IRQ_Handler |
|  |	 IoT Client |
|  |	 IoT Utility |
|  |	 LVGL |
|  |	 Library |
|  |	 MAXSDHC |
|  |	 MAXUSB |
|  |	 MAXUSB_HS |
|  |	 Machine Learning |
|  |	 Middleware |
|  |	 Multicore |
|  |	 MyClass |
|  |	 Native Driver |
|  |	 Network |
|  |	 None |
|  |	 Operating System |
|  |	 PSA |
|  |	 Peripherals |
|  |	 Project |
|  |	 Qualcomm |
|  |	 RTOS |
|  |	 RXF |
|  |	 SMQ |
|  |	 Security |
|  |	 Sensors |
|  |	 SharkSSL |
|  |	 Simulation |
|  |	 Sino32StdPeripherals |
|  |	 TFM Platform |
|  |	 TFM |
|  |	 TencentOS tiny |
|  |	 Test |
|  |	 Third Parties |
|  |	 Trust Protection Unit |
|  |	 USB |
|  |	 Wi-Fi |
|  |	 Wireless |
|  |	 mbed |
|  |	 nRF_ANT |
|  |	 nRF_BLE |
|  |	 nRF_Drivers |
|  |	 nRF_Drivers_External |
|  |	 nRF_IoT |
|  |	 nRF_Libraries |
|  |	 nRF_NFC |
|  |	 nRF_Properitary_RF |
|  |	 nRF_Serialization |
|  |	 nRF_SoftDevice |
|  |	 wolfSSL |

You can see how this confusing use of Cclass manifests itself in uVision RTE today:

[image]

I think a user should expect to see the Cyclone components categorised under `Security`. 

My proposal is that `CclassType` is changed to an enumerated value, with an agreed (probably expanded) set of Classes available for use.

ReinhardKeil, 2022-06-13:
@willlordarm not sure what you are suggesting here. 
The pre-defined Cclass list should become a standardized set of interfaces. However it should be always possible to add new Cclass values. Limiting the scope of Cclass values would otherwise prevent innovation.

willlordarm, 2022-06-13:
@ReinhardKeil I'm suggesting that we make the taxonomy fixed to aid discoverability. I don't think this would stifle innovation - I'm not suggesting we limit what can be _functionally_ be delivered as a component. I think we can help developers find the right components by requiring a bit more discipline on naming by vendors. If, for example, we wanted to generate a web page or search that lists all of the available RTOSs we couldn't do that reliably today because some vendors will have categorised them using a CClass of `RTOS` but others might have categorised using something like `MyVendorRTOS`. This means they can't be logically grouped and can only be indexed based on text (i.e. we look for all instances of 'RTOS' and group on this basis).

DavidJurajdaNXP, 2022-06-14:
Related issue: 
Taxonomic rank shift https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack-Spec/issues/22

I am supporting open approach "it should be always possible to add new Cclass values". Tools should be data driven.
Pure data driven approach is important also because developers will use alternative names during creation process for testing.

On the other hand we should motivate users to use standard classification. 
First question is why someone feel need to create new category. 

1. motivation - RTE manager
In my opinion could be motivated by attempt to reach specific appearance in RTE manager. In other words data are not created by set of rules, but by appearance in RTE manager. 

2. motivation - custom API
Another reason is difference in API. If some component is not compliant with CMSIS Pack API framework, there is natural need to indicate difference by new name. Maybe only its wrapper is fitting to standard classification.

willlordarm, 2022-06-15:
Perhaps I'm misunderstanding something but https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack-Spec/issues/23 looks like it will enable even more categories to be added? Will a predfined list of key/value pairs be enforced as a means of categorisation so that, for example, the whole pack ecosystem could be queried for everything categorised as an 'RTOS'? I think the central point - that categorisation and identification are two separate concepts - is right but I'm not sure how the current proposal will play out in real life.

jkrech, 2022-06-28:
@willlordarm, so we are aligned on separating identification from categorization.
#23 will allow both standardized and custom categorization.
Let's take the example of the key 'OCP-Classification' in #23.
The tools reading the components from the pdsc files will collect a unique list of values associated with the key 'OCP-Classification'. In the Component dialog people can select 'OCP-Classification' and all components will get grouped by the value, one of which ought to be 'RTOS'.
Does that clarify how this shall play out in real life?

willlordarm, 2022-06-28:
Thanks @jkrech. I think that clarifies everything for me.

closed 2022-06-28 by willlordarm