Open-CMSIS-Pack/Open-CMSIS-Pack-Spec #156: Clarification required about variants

fred-r, 2022-09-15:
In the spec today, I think there is this section:
https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/cp_PackTutorial.html#cp_PackSteps
that explains that a variant is part of the component identification.

Based on this, we can conclude that only one variant of a component can be selected at a time in a project (as this is one component).

But here:
https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/pdsc_components_pg.html#element_component

Nothing clearly states that variants are exclusive.
They are an optional part of the component identification, so does it mean that when used they create 2 distoinct identities so maybe 2 distinct components we may select at the same time ?

I mean : components are unique in a project (they can have instances only), but as they admit several variants, one may think that we can have:
- comp A variant A
- comp A variant B
at the same time.

This is not the intention of the spec IMO.

Besides, Keil uvision really implemtns it as exclusive : select comp A with variant A or variant B.

fred-r, 2022-09-15:
@slhultgren @lmestm

VGRSTM, 2022-09-15:
Grey area mixing specs & tools implementation. We may have couple of time same discussions considering `*.yml` stuff & `devtools` implementation ... a schema cannot fully describe a workflow / under the hood logic. By the end only the pair is self contain. Claiming standard definition such is an issue definitively but is solving so easy ?

Back to component variants myself I would be a bit afraid to mix various variant of a same component as part of a common project .... I like so Keil uvision implementation.

brondani, 2022-09-15:
It seems `csolution` currently allows the selection of multiple variants.
But I agree with you, from the nature of "variant" classification I would rather interpret them as mutually exclusive.

jkrech, 2024-01-30:
raise defect in csolution
raise defect against Open-CMSIS-Pack-Spec

jkrech, 2024-02-05:
PR created and issue against csolution raised.

closed 2024-02-05 by jkrech